esphome/issues #5708: interrupt_pin not working with CST820 Touch Screen Controller

regystro, 2024-04-17:
### The problem

When using `interrupt_pin` with a CST820 sensor , the touches do not trigger `on_touch` action.

Polling works as expected (removing `interrupt_pin`, and setting `update_interval: 50ms` as suggested by the [documentation](https://esphome.io/components/touchscreen/#configuration-variables))

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

cst816 Touch Screen Controller

### Example YAML snippet

```yaml
esp32:
  board: esp32dev

logger:

spi:
 clk_pin: GPIO14
 mosi_pin: GPIO13
 miso_pin: GPIO12

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: GPIO15
    dc_pin: GPIO2
    id: my_display

i2c:
  sda: GPIO33
  scl: GPIO32

touchscreen:
  platform: cst816
  id: my_touchscreen
  display: my_display
  interrupt_pin: GPIO21
  reset_pin: GPIO25
  update_interval: never
  on_touch:
      - lambda: |-
            ESP_LOGI("touch (single)", "x=%d, y=%d",
                touch.x,
                touch.y
                );
```

### Additional information

The board is an ESP32-2432S024C (2.4'' display with CST820 capacitive touchscreen). The display is an ILI9341.
The board is detected by ESPHome like this:
```
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

The interrupt pin is GPIO21. Platformio definitions: https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/esp32-2432S024C.json

I'm forcing `update_interval: never` because even when the [documentation](https://esphome.io/components/touchscreen/#configuration-variables) says it should default to _never_, `esphome config` shows a value of 50ms when interrupt_pin is set. Also if not set to _never_, the logs suggest you to: `[W][touchscreen:032]: Touch Polling Stopped. You can safely remove the 'update_interval:' variable from the YAML file.`

nielsnl68, 2024-04-18:
Hello @regystro , could you do me a favor and hook the interrupt pin to a `bineary_sensor` and check what the response is in the log or in HA? And report back with the patron you are seeing when you press the display and move it around and lift it up again.

regystro, 2024-04-19:
Hi @nielsnl68 . I just tested adding this:

```
binary_sensor:
  - platform: touchscreen
    name: Top Left Touch Button
    x_min: 0
    x_max: 100
    y_min: 0
    y_max: 100
    on_press:
      - lambda: |-
            ESP_LOGI("binary_sensor", "press");
    on_release:
      - lambda: |-
            ESP_LOGI("binary_sensor", "release");
    on_state:
      - lambda: |-
            ESP_LOGI("binary_sensor", "state");
    on_click:
      - lambda: |-
            ESP_LOGI("binary_sensor", "click");
    on_double_click:
      - lambda: |-
            ESP_LOGI("binary_sensor", "double_click");
```

Using `interrupt_pin: GPIO21`, nothing is detected.
Disabling `interrupt_pin` and changing `update_interval: 50ms` (= polling) detects and logs all possible states.

percy10442, 2024-05-21:
I encountered the same problem.

raineth, 2025-05-21:
I don't have one of these touchscreens, but I believe esphome 2025.5 (specifically esphome/esphome#8632) may have also fixed interrupt handling on the CST820.  Could someone test interrupt mode with 2025.5?