esphome/issues #5709: DOOYA RF remote_base - DC1600

Minims, 2024-04-17:
### The problem

Hello,

I'm trying the new Dooya RF and i'm unable to read codes from my remote DC1600

https://github.com/esphome/esphome/pull/6488

What remote has been tested in this PR ? How can i debug ? 
The remote is seen on my RFLINK.
My ESP32 can received nexa code / rcswitch code without issue.

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32-IDF

### Board

ESP32 WROOM 32

### Component causing the issue

Dooya

### Example YAML snippet

```yaml
remote_receiver:
  - id: RF_IN
    pin:
      number: 22
    dump:
      - rc_switch
      - nexa
      - dooya
```
```

Minims, 2024-04-19:
> I had to add a couple of extra parameters to get this to work:
> 
> ```
> remote_receiver:
>   pin: GPIO15
>   dump: dooya
>   # Add the following:
>   tolerance: 50%
>   filter: 250us
>   idle: 7ms
> ```

Your right It works with this config in the receiver.
many thanks !

closed 2024-04-19 by Minims

nagyrobi, 2024-07-19:
A quick note to the ones brought here by a search on `dooya`:

KinCony boards, at least, may print in the log after boot: 
`Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' in the pin schema!`

But, this is a false alarm, because if you really set `inverted: true` in the `remote_receiver`, you may not be able to decode `dooya` at all!

On a KinCony KC868-A16, this is the correct config to receive and send `dooya` codes:
```yaml
remote_receiver:
  pin:
    number: GPIO2
    ignore_strapping_warning: true
  dump: 
    - dooya
  tolerance: 50%
  filter: 250us
  idle: 7ms

remote_transmitter:
  pin:
    number: GPIO15
    ignore_strapping_warning: true
  carrier_duty_percent: 100%
```
It works, despite the warning presented at boot.