esphome/issues #5710: sht3xd component breaks in 2024.4.0, works in 2024.3.2

asolochek, 2024-04-18:
### The problem

SHT31 module fails to initialize after updating to 2024.4.0. I reverted to 2024.3.2 and verified it works there. 

### Which version of ESPHome has the issue?

24.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32

### Board

Adafruit Feather ESP32-S2

### Component causing the issue

sht3xd via i2c

### Example YAML snippet

```yaml
sensor:
  - platform: sht3xd
    i2c_id: bus_a
    address: 0x44
    update_interval: 2s
    temperature:
      name: "Temperature"
      id: temperature
      filters:
        - exponential_moving_average:
            alpha: 0.1
            send_every: 1
    humidity:
      name: "Humidity"
      id: humidity
      filters:
        - exponential_moving_average:
            alpha: 0.1
            send_every: 1
```

### Anything in the logs that might be useful for us?

```txt
This is the error I'm getting with 2024.4.0

[21:30:36][E][sht3xd:040]:   Communication with SHT3xD failed!
[21:30:36][E][component:082]:   Component sht3xd.sensor is marked FAILED

I also narrowed the problem to the module itself by copying the 2024.3.2 version locally and building it as an external component. That older version, built against 2024.4.0 everything else, works.
```

sbrk-r, 2024-04-18:
There was a recent change to add support for SHT85, maybe missed something in testing

https://github.com/esphome/esphome/pull/6415 

@mrtoy-me

mrtoy-me, 2024-04-18:
@sbrk-r 

weird - tested on arduino and esp-idf on SHT31D ADA2857 and just retested now working fine for me
Change was ID register as per 
https://sensirion.com/media/documents/E5762713/63D103C2/Sensirion_electronic_identification_code_SHT3x.pdf

What sensor are you using ?

asolochek, 2024-04-18:
It's the SHT31 module from adafruit. https://www.adafruit.com/product/2857

If you have any changes you want me to test just let me know.

asolochek, 2024-04-18:
Oh, I just noticed you used the exact same module to test.

I have another module and can test again on a couple different esp32 boards tomorrow.

asolochek, 2024-04-18:
Mine looks identical to the image on the adafruit site.

mrtoy-me, 2024-04-18:
Mine is blue board - newer pcb and maybe different chip model

mrtoy-me, 2024-04-18:
@sbrk-r  
Do yo have time to test this on your sensor... before I submit PR to fix
Use following yaml to reference my proposed fix:

external_components:
  - source:
      type: git
      url: https://github.com/mrtoy-me/esphome-my-components
      ref: main
    components: [ sht3xd ]

mrtoy-me, 2024-04-19:
Hi @asolochek 

Aaron , any chance you would have time to try my proposed change on your sensor in the next day - if ok I will submit PR 

Thanks a lot Brett

asolochek, 2024-04-19:
Yes, I'll test it first thing tomorrow (in roughly 8 hours).

kixtarter, 2024-04-19:
I think I can confirm this problem.
I am using two different SHT3x modules connected to a Wemos Lolin S3 mini (ESP32-S2) with the Arduino framework.
These SHT3x modules are:

- [DFRobot SHT31 Weather-proof Temperature & Humidity Sensor](https://wiki.dfrobot.com/SHT31_Temperature_Humidity_Sensor_Weatherproof_SKU_SEN0385)
- [Wemos SHT30 Temperature and Humidity Shield for D1 Mini](https://www.wemos.cc/en/latest/d1_mini_shield/sht30.html)

After upgrading from ESPHome version 2024.3.0 to 2024.4.0 both modules were failing with the error:

```
[11:16:17][C][sht3xd:038]: SHT3xD:
[11:16:17][E][sht3xd:040]:   Communication with SHT3xD failed!
[11:16:17][E][component:082]:   Component sht3xd.sensor is marked FAILED
[11:16:17][C][sht3xd:038]: SHT3xD:
[11:16:17][E][sht3xd:040]:   Communication with SHT3xD failed!
[11:16:17][E][component:082]:   Component sht3xd.sensor is marked FAILED
```
In this case several reboots of the ESP32-S2 device did not make any difference.
However, after changing the I2C frequency from 200kHz to 100kHz both modules came back online again.

I am also using other SHT3x modules on other ESP devices, and up to now with all upgrades of these ESP devices from ESPHome version 2024.3.0 to 2024.4.0 I initially got the same failed errors for the SHT3x modules.
However, for all these devices the SHT3x modules came back online again after a reboot of the ESP devices (without changing the I2C frequency).
So I am not completely sure that the cause of the problem is clear now…

mrtoy-me, 2024-04-19:
@kixtarter 

If you able to test my proposed fix usage the following yaml that would be great

external_components:
  - source:
      type: git
      url: https://github.com/mrtoy-me/esphome-my-components
      ref: main
    components: [ sht3xd ]

asolochek, 2024-04-19:
@mrtoy-me  your fix works for me. Thanks!

mrtoy-me, 2024-04-19:
Thanks a lot for testing !

WarC0zes, 2024-04-20:
Hi @mrtoy-me 
your fix works for me too. Thanks!

mrtoy-me, 2024-04-20:
@kixtarter 
Really appreciate your testing - my fix under this issue I have submitted as a PR to address the change I made which was in the most recent release 2024.4.0 I am not sure if there were changes to any other components after 2024.3.0 which might affect i2c frequency issue on this component.
I don't have much time over the next few weeks so I wanted to address this issu (#5710) as soon as I could. The issue you are seeing with i2c frequency seems before my change in 2024.4.0 and I did not have time to look at that. However, I am looking to see if I can track that issue down as well. I am working through that at https://github.com/esphome/issues/issues/5303 where this issue was raised last December.
If you go over to issue#5303, I have gone further today and have something else you can try. I have just worked on it today and I got it working on my devices but I will not have time to refine it and do another PR for several weeks, but hopefully it will tell me if I am on the right track.
Hope that helps...

anatoly-savchenkov, 2024-04-21:
Hi @mrtoy-me,

I faced the same problem. My setup is Lolin D1 mini (esp8266) and SHT31, 6 devices in total. Sensors' serial numbers differ by last 3 hex digits only, I assume all should be of the same revision. After upgrading to release 2024.4.0, 2 out of 6 started to report the error described in this issue.

Proposed fix works ok. Thanks!

mattlward, 2024-04-22:
I have been having the same problem with an external sht30 probe on ESP32.  I was able to resolve it to a degree by installing the following code after a discussion on reddit as it felt that it was an initialization problem.

`on_boot:
    priority: 200
    then:
      - lambda: |-        
          delay(500);`

I will try the extension and see if that fixes my issue and report back.
My original thread:
[https://www.reddit.com/r/Esphome/comments/1c9zroa/strange_issue_with_esphome_and_sht30_probe/](url)

mattlward, 2024-04-22:
I did try adding the extension and removing the delay and through 2 reboots the issue is gone!

closed 2024-04-22 by jesserockz