esphome/issues #5711: Binary file created with 2024.4.0 will not connect to WiFi

SteveRMann, 2024-04-19:
### The problem

    Here’s my problem. I upgraded ESPHome to 2024.4.0 and upgraded a few of my devices as well. But one (a Sonoff Basic) would not come back online again. Fortunately, this Sonoff Basic that was easy to reach. I plugged in a UART to manually flash the latest .bin file. When I restarted the device, the WiFi would not connect. I tried flashing the same .bin file to a Wemos D1 Mini and it connected to WiFi just fine. I flashed an Arduino sketch to the Sonoff Basic, and it too connectd to WiFi.

I also made a new device, using a new Sonoff Basic, new ESPHome code and it also would not connect to WiFi.

To summarize:
Sonoff Basic, ESPHome .bin file, no WiFi.
The same .bin file flashed to a new Sonoff Basic- no WiFi.
Wemos D1 Mini, same .bin file, WiFi connects OK.
Sonoff Basic, flashed with an Arduino Sketch, WiFi connects OK.

So, it looks like everything I tried connects the ESP to WiFi except the .bin file from ESPHome 2024.4.0

Reverting to ESPHome 2024.3.2 fixed the issue.

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2024.4.3

### What platform are you using?

ESP8266

### Board

Sonoff Basic

lucasimons, 2024-04-19:
I have the same issue only with a sonoff basic, all the other esp8266 (ex D1 mini) or esp32 are updated and are connected to wifi

ssieb, 2024-04-19:
This is a known issue with basics.  Wifi sometimes doesn't work and randomly changing the config, log levels, components, etc. will get it to work again.  There are already several issues about this and there's nothing that we can do about it.

closed 2024-04-19 by ssieb