esphome/issues #5714: "Error: The OTA partition on the ESP is too small" after updating to latest esphome version

digaus, 2024-04-19:
### The problem

After updating esphome version I can no longer flash my ESP32-C3 which the current config.

Compiling indicates that it will take 98.8% of the available flash space. This should be enough for OTA?

Any way to get it working again without having to reduce features? Currently disabled logger to be able to upload.

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32-IDF

### Board

Custom ESP32-C3

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.4.0
INFO Reading configuration /config/esphome/esphome-eqiva-fingerprint.yaml...
INFO Detected timezone 'Europe/Berlin'
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-eqiva-fingerprint (board: seeed_xiao_esp32c3; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40406.240122 (4.4.6) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/esphome-eqiva-fingerprint/src/main.o
Linking .pioenvs/esphome-eqiva-fingerprint/firmware.elf
RAM:   [==        ]  16.0% (used 52560 bytes from 327680 bytes)
Flash: [==========]  98.8% (used 1813794 bytes from 1835008 bytes)
Building .pioenvs/esphome-eqiva-fingerprint/firmware.bin
Creating esp32c3 image...
Successfully created esp32c3 image.
esp32_create_combined_bin([".pioenvs/esphome-eqiva-fingerprint/firmware.bin"], [".pioenvs/esphome-eqiva-fingerprint/firmware.elf"])
Wrote 0x1d8ec0 bytes to file /data/build/esphome-eqiva-fingerprint/.pioenvs/esphome-eqiva-fingerprint/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 18.83 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.3.47
INFO Uploading /data/build/esphome-eqiva-fingerprint/.pioenvs/esphome-eqiva-fingerprint/firmware.bin (1871552 bytes)
ERROR Error binary size: Error: The OTA partition on the ESP is too small. ESPHome needs to resize this partition, please flash over
```

HarvsG, 2024-07-06:
I get he same on 2024.6.6 with the atom echo when I try and adopt it with the voice assistant config (unedited)
When I do a USB instal (COM3 through the browser) I then get this:
```
[01:34:52]E (823) esp_image: Image length 1835744 doesn't fit in partition length 1835008
[01:34:52]E (823) boot: OTA app partition slot 0 is not bootable
[01:34:52]E (825) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[01:34:52]E (833) boot: OTA app partition slot 1 is not bootable
[01:34:52]E (839) boot: No bootable app partitions in the partition table
[01:34:52]ets Jun  8 2016 00:22:57
[01:34:52]
[01:34:52]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[01:34:52]configsip: 188777542, SPIWP:0xee
[01:34:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:34:52]mode:DIO, clock div:2
[01:34:52]load:0x3fff0030,len:6652
[01:34:52]ho 0 tail 12 room 4
[01:34:52]load:0x40078000,len:15052
[01:34:52]load:0x40080400,len:3836
[01:34:52]entry 0x4008069c
[01:34:52]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[01:34:52]I (57) boot: compile time 19:22:31
[01:34:52]I (58) boot: Multicore bootloader
[01:34:52]I (61) boot: chip revision: v1.1
[01:34:52]I (65) boot.esp32: SPI Speed      : 40MHz
[01:34:52]I (70) boot.esp32: SPI Mode       : DIO
[01:34:52]I (74) boot.esp32: SPI Flash Size : 4MB
[01:34:52]I (79) boot: Enabling RNG early entropy source...
[01:34:52]I (84) boot: Partition Table:
[01:34:52]I (88) boot: ## Label            Usage          Type ST Offset   Length
[01:34:52]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[01:34:52]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[01:34:52]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[01:34:52]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[01:34:52]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[01:34:52]I (133) boot: End of partition table
[01:34:52]I (137) boot: No factory image, trying OTA 0
[01:34:52]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=65914h (416020) map
[01:34:52]I (301) esp_image: segment 1: paddr=0007593c vaddr=3ffbdb60 size=051bch ( 20924) load
[01:34:52]I (310) esp_image: segment 2: paddr=0007ab00 vaddr=40080000 size=05518h ( 21784) load
[01:34:52]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=136820h (1271840) map
[01:34:53]I (779) esp_image: segment 4: paddr=001b6848 vaddr=40085518 size=19a6ch (105068) load
```

mrtnkhl, 2024-07-06:
I see the same behavior as @HarvsG does but after trying to adopt an Atom Echo once it has been flashed from esphome.io with the Atom Echo voice assistant config. Tried flashing via USB on the HA Yellow after wireless OTA did not work since a partition resize was necessary. Erased the Atom Echo before with a blank ESPWeb config and retried this several times.

Frederick-G764, 2024-07-06:
same issue

HarvsG, 2024-07-06:
Workaround that worked for me: https://github.com/esphome/firmware/issues/227#issuecomment-2211758801

steveof2620, 2024-07-07:
I have the same issue. I ended up installing via [these instructions by Mostly Chris](https://www.mostlychris.com/more-voice-more-voice-more-voice/).

orywig, 2024-07-18:
I am seeing this with my M5Stack Atom Echo when I tried updating it to 2024.7

HarvsG, 2024-07-18:
> I am seeing this with my M5Stack Atom Echo when I tried updating it to 2024.7

Please post your environment details in: https://github.com/esphome/firmware/issues/227

orywig, 2024-07-19:
> > I am seeing this with my M5Stack Atom Echo when I tried updating it to 2024.7
> 
> Please post your environment details in: [esphome/firmware#227](https://github.com/esphome/firmware/issues/227)

Done.. https://github.com/esphome/firmware/issues/227#issuecomment-2240060942

lmcancu1, 2025-01-23:
Same issue as well. Logs attached

[logs_screek-human-sensor-2a-2228ec_run.txt](https://github.com/user-attachments/files/18521107/logs_screek-human-sensor-2a-2228ec_run.txt)

digaus, 2025-03-11:
https://github.com/esphome/feature-requests/issues/2506

Would be cool if we could just add two phase OTA to allow 55% bigger size

StanPolozov, 2026-01-31:
@digaus , this is exactly the case our tool RACSS was designed for.
It lets you keep large static firmware data compressed in flash and read it directly without decompression, so binary size drops while CPU/RAM overhead stays near zero. The decoder is ~200 lines of C.
https://velixlab.com/racss.html
If you share a small sample of your firmware data (strings/tables), we can run a quick test and show the real size reduction on your case.