esphome/issues #5715: Error compiling- error trying to exec 'as': execvp: No such file or directory

gmawdsley1, 2024-04-20:
### The problem

Compiling .pioenvs/esphome-web-camera/src/esphome/components/api/api_pb2_service.cpp.o
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory
*** [.pioenvs/esphome-web-camera/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec 'as': execvp: No such file or directory

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.4.2

### What platform are you using?

ESP32

### Board

ESP32-cam

### Example YAML snippet

```yaml
substitutions:
  name: esphome-web-camera
  friendly_name: esp-camera

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Example configuration entry
esp32_camera:
  name: my-camera
  external_clock:
    pin: GPIO27
    frequency: 20MHz
  i2c_pins:
    sda: GPIO25
    scl: GPIO23
  data_pins: [GPIO17, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15
  resolution: 640x480
  jpeg_quality: 10

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:
```

### Anything in the logs that might be useful for us?

```txt
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.4.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.12.2",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Toronto",
    "os_name": "Linux",
    "os_version": "6.1.73-haos-raspi",
    "supervisor": "2024.04.0",
    "host_os": "Home Assistant OS 12.2",
    "docker_version": "25.0.5",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "infinitude_beyond": {
      "version": "2024.3.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.34.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "smartthinq_sensors": {
      "version": "0.39.0",
      "requirements": [
        "pycountry>=23.0.0",
        "xmltodict>=0.13.0",
        "charset_normalizer>=3.2.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "esphome",
    "name": "ESPHome",
    "after_dependencies": [
      "zeroconf",
      "tag"
    ],
    "codeowners": [
      "@OttoWinter",
      "@jesserockz",
      "@kbx81",
      "@bdraco"
    ],
    "config_flow": true,
    "dependencies": [
      "assist_pipeline",
      "bluetooth"
    ],
    "dhcp": [
      {
        "registered_devices": true
      }
    ],
    "documentation": "https://www.home-assistant.io/integrations/esphome",
    "integration_type": "device",
    "iot_class": "local_push",
    "loggers": [
      "aioesphomeapi",
      "noiseprotocol",
      "bleak_esphome"
    ],
    "requirements": [
      "aioesphomeapi==23.2.0",
      "esphome-dashboard-api==1.2.3",
      "bleak-esphome==1.0.0"
    ],
    "zeroconf": [
      "_esphomelib._tcp.local."
    ],
    "is_built_in": true
  },
  "data": {
    "config": {
      "entry_id": "23e6b498f281561289fe1c914339a0d8",
      "version": 1,
      "minor_version": 1,
      "domain": "esphome",
      "title": "ESPHome Web 540d80",
      "data": {
        "host": "192.168.2.241",
        "port": 6053,
        "password": "",
        "noise_psk": "",
        "device_name": "esphome-web-540d80"
      },
      "options": {
        "allow_service_calls": true
      },
      "pref_disable_new_entities": false,
      "pref_disable_polling": false,
      "source": "zeroconf",
      "unique_id": "a0:b7:65:54:0d:80",
      "disabled_by": null
    },
    "storage_data": {
      "device_info": {
        "uses_password": false,
        "name": "esphome-web-540d80",
        "friendly_name": "ESPHome Web 540d80",
        "mac_address": "**REDACTED**",
        "compilation_time": "Mar 26 2024, 07:25:13",
        "model": "esp32dev",
        "manufacturer": "Espressif",
        "has_deep_sleep": false,
        "esphome_version": "2024.3.0",
        "project_name": "esphome.web",
        "project_version": "1.0",
        "webserver_port": 80,
        "voice_assistant_version": 0,
        "legacy_bluetooth_proxy_version": 0,
        "bluetooth_proxy_feature_flags": 0,
        "suggested_area": ""
      },
      "services": [],
      "api_version": {
        "major": 1,
        "minor": 9
      }
    },
    "dashboard": "5c53de3b_esphome"
  }
}
```

### Additional information

Is "as" in the PATH? This looks like https://community.home-assistant.io/t/esp32-wont-compile-xtensa-esp32-elf-g-internal-compiler-error-killed-program-cc1plus/490814

ssieb, 2024-04-20:
It's not the same as that.  Try re-creating the container.

gmawdsley1, 2024-04-24:
I have updated core and Supervisor. I tried loading the bare template, and had the same fail as above.
I then teried the MHz19 install:

INFO ESPHome 2024.4.1
INFO Reading configuration /config/esphome/esphome-web-540d80.yaml...
INFO Updating https://github.com/fornellas/esphome@mhz19_patches
INFO Generating C++ source...
INFO Compiling app...
Processing environmental (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/environmental/src/esphome/components/api/api_connection.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/environmental/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
Compiling .pioenvs/environmental/src/esphome/components/api/api_pb2.cpp.o
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/environmental/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/environmental/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/environmental/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/environmental/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: riscv32-esp-elf-g++: not found
*** [.pioenvs/environmental/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 34.97 seconds =========================

sol3uk, 2024-05-31:
I have the same issue with an ever so slightly different error message.
I'm on HA 2024.5.5
Addon version: 2024.5.4 
The error message is:
riscv32-esp-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory

From doing a bit of reading online it seems like the addon is maybe missing a package gcc-c++ / g++? 
https://stackoverflow.com/questions/8878676/compile-error-g-error-trying-to-exec-cc1plus-execvp-no-such-file-or-dir

I'm not 100% sure, but I would rather not uninstall the addon and reinstall it, since I'm not sure what would happen to my existing devices. If anyone can help clarify what would happen on a reinstall and if my devices would need to be setup again in HA, then that would help me make a decision on where to go next or if to wait for a fix.

ssieb, 2024-05-31:
Nothing happens if you do a re-install.  The yaml files are in the HA common config area.  And it has nothing to do with the devices being integrated in HA.  You should of course be backing them up anyway.

dekiesel, 2025-07-03:
I'm seeing the same on 2025.6.3. I also had it on 2025.6.1.

Log:
```
INFO ESPHome 2025.6.3
INFO Reading configuration /config/bluetooth_node_test.yaml...
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing bluetoothtest (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - broken
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/config/.esphome/build/bluetoothtest/.pioenvs/bluetoothtest/CMakeFiles/CMakeScratch/TryCompile-HBg7om'
    
    Run Build Command(s): /config/.esphome/platformio/packages/tool-ninja/ninja -v cmTC_c345a
    [1/2] /config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_c345a.dir/testCCompiler.c.obj -c /config/.esphome/build/bluetoothtest/.pioenvs/bluetoothtest/CMakeFiles/CMakeScratch/TryCompile-HBg7om/testCCompiler.c
    FAILED: CMakeFiles/cmTC_c345a.dir/testCCompiler.c.obj 
    /config/.esphome/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_c345a.dir/testCCompiler.c.obj -c /config/.esphome/build/bluetoothtest/.pioenvs/bluetoothtest/CMakeFiles/CMakeScratch/TryCompile-HBg7om/testCCompiler.c
    xtensa-esp-elf-gcc: fatal error: cannot execute 'as': execvp: No such file or directory
    compilation terminated.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:564 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 2.41 seconds ==========================
```

Yaml:

```
esphome:
  name: "bluetoothtest"
  friendly_name: "Bluetoothtest"
  area: "Wohnzimmer"

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y

wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Web Fallback Hotspot"
    password: !secret wifi_fallback_password

logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_api_key

ota:
  - platform: esphome
    password: !secret esphome_ota_password
time:
  - platform: homeassistant

captive_portal:

esp32_ble_tracker:

bluetooth_proxy:
  active: true

```

I'd really appreciate any hints!

Ten0, 2025-08-12:
Same with everything fresh (ESPHome 2025.7.5, freshly flashed ESPHome from the web flasher), I get this error when I try to make my first install.

<details>

<summary>
logs
</summary>

```
INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/esphome-web-f05d7c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-f05d7c (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - broken
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/data/build/esphome-web-f05d7c/.pioenvs/esphome-web-f05d7c/CMakeFiles/CMakeScratch/TryCompile-ByTDCW'
    
    Run Build Command(s): /data/cache/platformio/packages/tool-ninja/ninja -v cmTC_b3e46
    [1/2] /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_b3e46.dir/testCCompiler.c.obj -c /data/build/esphome-web-f05d7c/.pioenvs/esphome-web-f05d7c/CMakeFiles/CMakeScratch/TryCompile-ByTDCW/testCCompiler.c
    FAILED: CMakeFiles/cmTC_b3e46.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_b3e46.dir/testCCompiler.c.obj -c /data/build/esphome-web-f05d7c/.pioenvs/esphome-web-f05d7c/CMakeFiles/CMakeScratch/TryCompile-ByTDCW/testCCompiler.c
    xtensa-esp-elf-gcc: fatal error: cannot execute 'as': execvp: No such file or directory
    compilation terminated.
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:564 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 6.76 seconds ==========================
```
</details>

Running the addon on HA on a ~RPI w. 2G Ram & 16G eMMC.

Ten0, 2025-08-12:
Update: Deleting the addon and its data and reinstalling, then taking care to run the build process only once by fixing the config (adding wifi config) before the first attempt to take control of a device solved the issue.

That makes me suspect that suspect that the issue came from that I might have started concurrent attempts to install at some point and they raced on installing stuff and ended up with missing files but thinking they didn't need a reinstall.

I could compile things this time, but I suspect something still went wrong: my first attempt to install gave the following incomplete log:

<details>
<summary>
First install attempt logs
</summary>

```
INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/esphome-web-f05d7c.yaml...
INFO Generating C++ source...
INFO Core config, version changed, cleaning build files...
INFO Compiling app...
Processing esphome-web-f05d7c (board: esp32dev; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
INFO Installing https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@53.3.13 has been installed!
INFO espressif32@53.3.13 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.3.2/esp-idf-v5.3.2.zip
INFO Installing https://github.com/pioarduino/esp-idf/releases/download/v5.3.2/esp-idf-v5.3.2.zip
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.50302.0 has been installed!
INFO framework-espidf@3.50302.0 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530
INFO Installing platformio/toolchain-xtensa-esp-elf @ 13.2.0+20240530
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
INFO toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
Tool Manager: Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
INFO Installing espressif/toolchain-esp32ulp @ 2.35.0-20220830
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-esp32ulp@2.35.0-20220830 has been installed!
INFO toolchain-esp32ulp@2.35.0-20220830 has been installed!
Tool Manager: Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
INFO Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
INFO tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
INFO Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
INFO tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v4.8.6/esptool.zip
INFO Installing https://github.com/pioarduino/esptool/releases/download/v4.8.6/esptool.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@4.8.6 has been installed!
INFO tool-esptoolpy@4.8.6 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
INFO tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.30.2
INFO Installing platformio/tool-cmake @ ~3.30.2
Downloading  [####################################]  100%
[... 140 lines cut ...]
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/logger/task_log_buffer.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/ota/ota_backend.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/safe_mode/safe_mode.cpp.o
[... cut ...]
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/core/application.cpp.o
Compiling .pioenvs/esphome-web-f05d7c/src/esphome/core/color.cpp.o
```
</details>

Note how it just cuts in the middle, and then gives the "Retry" button at the bottom right of the log stream.

Only the second install actually succeeded.

I'm not sure when that will come and bite me...
For this particular issue I suspect low resources on my device might be one of the necessary factors for the issue to appear.

dekiesel, 2025-08-13:
> Update 2: Solved the issue by removing the addon and doing things locally using the docker image. That works much faster and is a better dev UX anyway.

Hmm, I am already running esphome in a docker image. Did you change anything else? Thanks!

ssieb, 2025-08-13:
Try clearing the platformio cache.