esphome/issues #5721: Dooya RF Remote possible off-by-one

pedrovanzella, 2024-04-22:
### The problem

When receiving an RF Remote with the Dooya protocol, ESPHome detects it with ID `0x00b2c5a9`. However, both a Flipper Zero and a HackRF Portapack with the Mayhem firmware read the same remote as ID `0x004d3a56`. If you add `0x00b2c5a9` and `0x004d3a56`, you get `0x00ffffff`, so this is probably an off-by-one error somewhere.

The Button, Check and Channel codes also look weird. Channel 0 reads as 127, and channel 1 as 126, and so on. Button Up, short press reads as Code 14, check 1, and long press as Code 14 check 14.

If I try to send the decoded content back through an RF Transmitter with the Dooya protocol and all the same data (ID, Channel, Button, Check etc), my blinds don't respond, and I'm not able to decode the signal in the Flipper or the HackRF, even though I do see some signal being output, indicating the transmitter is indeed trying to send something.

If I plug in the data from the Flipper dump, I'm not able to get the blinds to respond either. Note that I can control the blinds from the Flipper or the HackRF with those dumps.

Reference: https://github.com/esphome/esphome/pull/6488

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

ms40724, 2024-05-05:
I have the same problem, sending the same decoded content back with the Dooya protocol does not work as expected.

deejaybeam, 2024-06-18:
Same problem here - sending captured values does not work.

jesserockz, 2024-06-20:
It is not off by one, the 1s and 0s are flipped completely.

```py
>>> correct = 0x004d3a56
>>> bin(correct)
'0b010011010011101001010110'
>>> wrong = 0x00b2c5a9
>>> bin(wrong)
'0b101100101100010110101001'
```

jesserockz, 2024-06-20:
Try the linked PR:
```yaml
external_components:
  - source: github://pr#6947
    components: remote_base
```

ssieb, 2024-06-20:
Do you see a message in the logs about inverting the receive pin?

deejaybeam, 2024-06-20:
No - does not work.
It works if I send the captured RAW data, but does not work if I try to re-send the (new/changed) Dooya data.
Old Dooya data was:
```[I][remote.dooya:116]: Received Dooya: id=0x00BAF9E5, channel=174, button=10, check=10```
New Dooya data is:
```[I][remote.dooya:116]: Received Dooya: id=0x0045061A, channel=81, button=5, check=5```
RAW-Data I send is:
```code: [4759, -1556, 306, -764, 670, -402, 304, -764, 306, -765, 306, -763, 671, -403, 303, -764, 671, -405, 307, -760, 310, -760, 310, -762, 308, -763, 307, -759, 676, -397, 674, -397, 309, -763, 313, -758, 312, -758, 312, -760, 675, -393, 677, -395, 312, -753, 682, -392, 314, -760, 316, -754, 680, -391, 316, -754, 681, -389, 317, -755, 315, -754, 317, -751, 682, -394, 319, -754, 681, -390, 316, -753, 682, -390, 317, -749, 685, -387, 319, -750, 685, -404]```

closed 2024-06-21 by jesserockz

ssieb, 2024-06-21:
@deejaybeam what do you receive when you transmit?  Maybe come to the esphome discord where it's easier to interactively discuss it.