esphome/issues #5722: HLW8012 causes RTL8710BN (WR3E) boot loop

Jaco1990, 2024-04-22:
### The problem

When ever code is compiled for the RTL8710BN, and the HLW8012 (BL0937) component is included, the firmware will automatically reset the device when the HLW8012 component is setup. Once the component is disabled the device works correctly. 

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

RTL87XX

### Board

RTL8710BN (TUYA WR3E)

### Component causing the issue

HLW8012

### Example YAML snippet

```yaml
- platform: hlw8012
    model: BL0937
    current_resistor: ${current_resistor}
    voltage_divider: ${voltage_divider}
    change_mode_every: "never"
    sel_pin:
      number: PA19
    cf_pin: PA15
    cf1_pin: PA14
    current:
      name: "HLW8012 Current"
    voltage:
      name: "HLW8012 Voltage"
    power:
      name: "HLW8012 Power"
    energy:
      name: "HLW8012 Energy"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d] 
calibration_result:[2:13:7][1:d] 
calibration_result:[3:1:1][1:1] 
calibration_ok:[2:13:7] 
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x8086470:4280:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x8040b79]
System_Init1
System_Init2
I [      0.000] LibreTiny v1.5.1 on wr3e, compiled at Apr 22 2024 17:51:59, GCC 10.3.1 (-Os)
[C][status_led:014]: Setting up Status LED...
[C][ota:483]: There have been 9 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][switch.gpio:011]: Setting up GPIO Switch 'Isolator Relay'...
[D][switch:016]: 'Isolator Relay' Turning OFF.
[D][switch:055]: 'Isolator Relay': Sending state OFF
[D][light:036]: 'indicator_light' Setting:
[D][light:041]:   Color mode: 
[D][switch:016]: 'Isolator Relay' Turning OFF.
[D][binary_sensor:034]: 'button': Sending initial state ON
[C][light:035]: Setting up light 'indicator_light'...
[D][light:036]: 'indicator_light' Setting:
[C][hlw8012:014]: Setting up HLW8012...
W [      0.099] CHANGE interrupts not supported
```

szupi-ipuzs, 2024-04-23:
See https://github.com/libretiny-eu/libretiny/issues/155#issuecomment-1925712517

Jaco1990, 2024-04-25:
Thank you for the link, however on HAOS I cannot access the data files. i have found it within the docker file, but cant open it in an editor.

Jaco1990, 2024-04-30:
Issue resolved.

closed 2024-04-30 by Jaco1990