esphome/issues #5724: R503-s Grow Fingerprint Sensor [component:082]:  Component fingerprint_grow is marked FAILED

Marcos-kp, 2024-04-22:
### The problem

I just got a R503-s Grow Fingerprint Sensor and I wired everything and got it working with HA very well. Suddenly, the sensor stopped working. I guess it occured right after I updated ESPHome to the latest version.
To make sure nothing changed, I looked at the wiring and did some soldering again but nothing changed. Everytime I placed my finger at the sensor, nothing happened, no led indicator, nothing.

Than, I tried to add a fingerprint by calling the enrol service and to my surprise, I got a purple LED as configured, but the service did not proceed with the enrolment and the sensor is stuck in purple.

I also was able to delete specific ids (fingerprints) as the log shows.

What I got that I couldn't find any information about is this:  [component:082]:  Component fingerprint_grow is marked FAILED

Any idea what might be happening?

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

[component:082]:   Component fingerprint_grow

### Example YAML snippet

```yaml
esphome:
  name: sensor-digital
  friendly_name: Sensor_Digital

esp8266:
  board: nodemcuv2
# Enable logging
logger:

ota:
  password: "password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sensor-Digital Fallback Hotspot"
    password: "password"

captive_portal:
    

api:
  encryption:
    key: "apikey"
  services:
  - service: enroll
    variables:
[... 274 lines cut ...]
[13:30:46][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:30:52][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:30:58][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:02][I][fingerprint_grow:295]: Deleting fingerprint in slot 3
[13:31:02][I][fingerprint_grow:299]: Deleted fingerprint
[13:31:02][D][fingerprint_grow:285]: Getting fingerprint count
[13:31:02][D][fingerprint_grow:288]: Got fingerprint count
[13:31:02][D][sensor:093]: 'Fingerprint Count': Sending state 7.00000  with 0 decimals of accuracy
[13:31:04][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:10][I][fingerprint_grow:295]: Deleting fingerprint in slot 2
[13:31:10][I][fingerprint_grow:299]: Deleted fingerprint
[13:31:10][D][fingerprint_grow:285]: Getting fingerprint count
[13:31:10][D][fingerprint_grow:288]: Got fingerprint count
[13:31:10][D][sensor:093]: 'Fingerprint Count': Sending state 6.00000  with 0 decimals of accuracy
[13:31:10][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:16][I][fingerprint_grow:097]: Starting enrollment in slot 2
[13:31:16][D][fingerprint_grow:348]: Setting Aura LED
[13:31:16][D][fingerprint_grow:352]: Aura LED set
[13:31:16][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:22][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:28][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:34][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:40][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[13:31:46][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
```

ssieb, 2024-04-22:
serial logs from boot might provide more info.

Marcos-kp, 2024-04-22:
I updated it with VERY_VERBOSE log level and here it is the logs. I tried to delete and add another fingerprint just to generate some more information.

I guess this is the boot you mentioned, right @ssieb?

`Uploading: [============================================================] 100% Done...

INFO Upload took 4.25 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.15.74 using esphome API
INFO Successfully connected to sensor-digital @ 192.168.15.74 in 19.557s
INFO Successful handshake with sensor-digital @ 192.168.15.74 in 2.927s
[19:30:12][I][app:100]: ESPHome version 2024.4.0 compiled on Apr 22 2024, 19:28:23
[19:30:12][C][wifi:580]: WiFi:
[19:30:12][C][wifi:408]:   Local MAC: 48:5E:DA:44:5C:3D
[19:30:12][C][wifi:413]:   SSID: ''[redacted]
[19:30:12][C][wifi:416]:   IP Address: 192.168.15.74
[19:30:12][C][wifi:419]:   BSSID: ""[redacted]
[19:30:12][C][wifi:421]:   Hostname: 'sensor-digital'
[19:30:12][C][wifi:423]:   Signal strength: -76 dB ▂▄▆█
[19:30:12][V][wifi:425]:   Priority: 0.0
[19:30:12][C][wifi:427]:   Channel: 11
[19:30:12][C][wifi:428]:   Subnet: 255.255.255.0
[19:30:12][C][wifi:429]:   Gateway: 192.168.15.1
[19:30:12][C][wifi:430]:   DNS1: 192.168.15.34
[19:30:12][C][wifi:431]:   DNS2: 8.8.4.4
[19:30:12][C][logger:166]: Logger:
[19:30:12][C][logger:167]:   Level: VERY_VERBOSE
[19:30:12][C][logger:169]:   Log Baud Rate: 115200
[19:30:12][C][logger:170]:   Hardware UART: UART0
[19:30:12][C][uart.arduino_esp8266:118]: UART Bus:
[19:30:12][C][uart.arduino_esp8266:119]:   TX Pin: GPIO4
[19:30:12][C][uart.arduino_esp8266:120]:   RX Pin: GPIO5
[19:30:12][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[19:30:12][C][uart.arduino_esp8266:124]:   Baud Rate: 57600 baud
[19:30:12][C][uart.arduino_esp8266:125]:   Data Bits: 8
[19:30:12][C][uart.arduino_esp8266:126]:   Parity: NONE
[19:30:12][C][uart.arduino_esp8266:127]:   Stop bits: 1
[19:30:12][C][uart.arduino_esp8266:131]:   Using software serial
[19:30:12][C][template.text_sensor:020]: Template Sensor 'Fingerprint State'
[19:30:12][C][fingerprint_grow:537]: GROW_FINGERPRINT_READER:
[19:30:12][C][fingerprint_grow:538]:   System Identifier Code: 0x0000
[19:30:12][C][fingerprint_grow:539]:   Touch Sensing Pin: GPIO12
[19:30:12][C][fingerprint_grow:541]:   Sensor Power Pin: None
[19:30:12][C][fingerprint_grow:546]:   Idle Period to Sleep: Never
[19:30:12][C][fingerprint_grow:548]:   Update Interval: 0.5s
[19:30:12][C][fingerprint_grow:550]:   Fingerprint Count 'Fingerprint Count'
[19:30:12][C][fingerprint_grow:550]:     State Class: ''
[19:30:12][C][fingerprint_grow:550]:     Unit of Measurement: ''
[19:30:12][C][fingerprint_grow:550]:     Accuracy Decimals: 0
[19:30:12][C][fingerprint_grow:550]:     Icon: 'mdi:fingerprint'
[19:30:12][C][fingerprint_grow:551]:     Current Value: 65535
[19:30:12][C][fingerprint_grow:554]:   Status 'Fingerprint Status'
[19:30:12][C][fingerprint_grow:554]:     State Class: ''
[19:30:12][C][fingerprint_grow:554]:     Unit of Measurement: ''
[19:30:12][C][fingerprint_grow:554]:     Accuracy Decimals: 0
[19:30:12][C][fingerprint_grow:555]:     Current Value: 255
[19:30:13][C][fingerprint_grow:558]:   Capacity 'Fingerprint Capacity'
[19:30:13][C][fingerprint_grow:558]:     State Class: ''
[19:30:13][C][fingerprint_grow:558]:     Unit of Measurement: ''
[19:30:13][C][fingerprint_grow:558]:     Accuracy Decimals: 0
[19:30:13][C][fingerprint_grow:558]:     Icon: 'mdi:database'
[19:30:13][C][fingerprint_grow:559]:     Current Value: 65535
[19:30:13][C][fingerprint_grow:562]:   Security Level 'Fingerprint Security Level'
[19:30:13][C][fingerprint_grow:562]:     State Class: ''
[19:30:13][C][fingerprint_grow:562]:     Unit of Measurement: ''
[19:30:13][C][fingerprint_grow:562]:     Accuracy Decimals: 0
[19:30:13][C][fingerprint_grow:562]:     Icon: 'mdi:security'
[19:30:13][C][fingerprint_grow:563]:     Current Value: 255
[19:30:13][C][fingerprint_grow:566]:   Last Finger ID 'Fingerprint Last Finger ID'
[... 463 lines cut ...]
[19:32:12][VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2216116077
  state: '-'
  missing_state: NO
}
[19:32:12][VV][api.connection:132]: Sending keepalive PING...
[19:32:12][VV][api.service:037]: send_ping_request: PingRequest {}
[19:32:12][VV][api.service:611]: on_ping_response: PingResponse {}
[19:32:18][VV][scheduler:225]: Running interval 'update' with interval=6000 last_execution=143532 (now=149537)
[19:32:18][V][text_sensor:013]: 'Fingerprint State': Received new state -
[19:32:18][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[19:32:18][VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2216116077
  state: '-'
  missing_state: NO
}
[19:32:24][VV][scheduler:225]: Running interval 'update' with interval=6000 last_execution=149532 (now=155532)
[19:32:24][V][text_sensor:013]: 'Fingerprint State': Received new state -
[19:32:24][D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[19:32:24][VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 2216116077
  state: '-'
  missing_state: NO
}
`

Marcos-kp, 2024-04-23:
Taking a further look at my entities in HA, my sensor became unknown at the exact same time my update.sensor firmware was updated.
[image]
[image]

swoboda1337, 2024-04-25:
So if you look at the log:

[19:30:12][C][fingerprint_grow:541]:   Sensor Power Pin: None

In the code:
  // Immediately return if there is no power pin or the sensor is already on
  if ((!this->has_power_pin_) || (this->is_sensor_awake_))
    return;

This module wont do anything unless you set a power pin

You can see this PR went in a month ago and it now requires a power pin: https://github.com/esphome/esphome/pull/6116

If you provide a power pin it will probably work. 

@jesserockz  the component should probably throw an error during yaml validation if there is a sense pin but no power pin.

Marcos-kp, 2024-04-25:
Hey @swoboda1337 ! Thanks for the info!

Unfortunately I am still getting the same error. I tried pin D0, D5 and D6 on my nodemcu board.

Here is the log:

`[15:20:21][C][logger:170]:   Hardware UART: UART0
[15:20:21][C][uart.arduino_esp8266:118]: UART Bus:
[15:20:21][C][uart.arduino_esp8266:119]:   TX Pin: GPIO4
[15:20:21][C][uart.arduino_esp8266:120]:   RX Pin: GPIO5
[15:20:21][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[15:20:21][C][uart.arduino_esp8266:124]:   Baud Rate: 57600 baud
[15:20:21][C][uart.arduino_esp8266:125]:   Data Bits: 8
[15:20:21][C][uart.arduino_esp8266:126]:   Parity: NONE
[15:20:21][C][uart.arduino_esp8266:127]:   Stop bits: 1
[15:20:21][C][uart.arduino_esp8266:131]:   Using software serial
[15:20:21][C][template.text_sensor:020]: Template Sensor 'Fingerprint State'
[15:20:21][C][fingerprint_grow:537]: GROW_FINGERPRINT_READER:
[15:20:21][C][fingerprint_grow:538]:   System Identifier Code: 0x0000
[15:20:21][C][fingerprint_grow:539]:   Touch Sensing Pin: GPIO12
[15:20:21][C][fingerprint_grow:541]:   Sensor Power Pin: GPIO16
[15:20:21][C][fingerprint_grow:544]:   Idle Period to Sleep: 5000 ms
[15:20:21][C][fingerprint_grow:548]:   Update Interval: 0.5s
[15:20:21][C][fingerprint_grow:550]:   Fingerprint Count 'Fingerprint Count'
[15:20:21][C][fingerprint_grow:550]:     State Class: ''
[15:20:21][C][fingerprint_grow:550]:     Unit of Measurement: ''
[15:20:21][C][fingerprint_grow:550]:     Accuracy Decimals: 0
[15:20:21][C][fingerprint_grow:550]:     Icon: 'mdi:fingerprint'
[15:20:21][C][fingerprint_grow:551]:     Current Value: 65535
[15:20:21][C][fingerprint_grow:554]:   Status 'Fingerprint Status'
[15:20:21][C][fingerprint_grow:554]:     State Class: ''
[15:20:21][C][fingerprint_grow:554]:     Unit of Measurement: ''
[15:20:21][C][fingerprint_grow:554]:     Accuracy Decimals: 0
[15:20:21][C][fingerprint_grow:555]:     Current Value: 255
[15:20:21][C][fingerprint_grow:558]:   Capacity 'Fingerprint Capacity'
[15:20:21][C][fingerprint_grow:558]:     State Class: ''
[15:20:21][C][fingerprint_grow:558]:     Unit of Measurement: ''
[15:20:21][C][fingerprint_grow:558]:     Accuracy Decimals: 0
[15:20:21][C][fingerprint_grow:558]:     Icon: 'mdi:database'
[15:20:21][C][fingerprint_grow:559]:     Current Value: 65535
[15:20:21][C][fingerprint_grow:562]:   Security Level 'Fingerprint Security Level'
[15:20:21][C][fingerprint_grow:562]:     State Class: ''
[15:20:21][C][fingerprint_grow:562]:     Unit of Measurement: ''
[15:20:21][C][fingerprint_grow:562]:     Accuracy Decimals: 0
[15:20:21][C][fingerprint_grow:562]:     Icon: 'mdi:security'
[15:20:22][C][fingerprint_grow:563]:     Current Value: 255
[15:20:22][C][fingerprint_grow:566]:   Last Finger ID 'Fingerprint Last Finger ID'
[15:20:22][C][fingerprint_grow:566]:     State Class: ''
[15:20:22][C][fingerprint_grow:566]:     Unit of Measurement: ''
[15:20:22][C][fingerprint_grow:566]:     Accuracy Decimals: 0
[15:20:22][C][fingerprint_grow:566]:     Icon: 'mdi:account'
[15:20:22][C][fingerprint_grow:567]:     Current Value: -1
[15:20:22][C][fingerprint_grow:570]:   Last Confidence 'Fingerprint Last Confidence'
[15:20:22][C][fingerprint_grow:570]:     State Class: ''
[15:20:22][C][fingerprint_grow:570]:     Unit of Measurement: ''
[15:20:22][C][fingerprint_grow:570]:     Accuracy Decimals: 0
[15:20:22][C][fingerprint_grow:570]:     Icon: 'mdi:account-check'
[15:20:22][C][fingerprint_grow:571]:     Current Value: -1
[15:20:22][E][component:082]:   Component fingerprint_grow is marked FAILED
[15:20:22][C][captive_portal:088]: Captive Portal:
[15:20:22][C][mdns:115]: mDNS:
[15:20:22][C][mdns:116]:   Hostname: sensor-digital
[15:20:22][C][ota:096]: Over-The-Air Updates:
[15:20:22][C][ota:097]:   Address: sensor-digital.local:8266
[15:20:22][C][ota:100]:   Using Password.
[15:20:22][C][ota:103]:   OTA version: 2.
[15:20:22][C][api:139]: API Server:
[15:20:22][C][api:140]:   Address: sensor-digital.local:6053
[15:20:22][C][api:142]:   Using noise encryption: YES `

swoboda1337, 2024-04-25:
Sorry I misread the code, it doesnt need the power pin. Not sure what caused it to stop working. I can take another look in an hour or so.

Marcos-kp, 2024-04-25:
Thanks! I will greatly appreciate!

tergontor, 2024-04-25:
In my case, after more than a month of running it, today it stopped working with the same message: [E][component:082]: Component fingerprint_grow is marked FAILED.
I updated esphome to version 2024.4.1 two days ago so I didn't think that was the problem.

Reviewing my code I had commented the sensing pin
fingerprint_grow:
#  sensing_pin: D6  #gpio12
on_finger_scan_matched:
- homeassistant.event:
....

What I have done is remove the comment
fingerprint_grow:
sensing_pin: D6 #gpio12
on_finger_scan_matched:
- homeassistant.event:

Once compiled again it started working without problems.

Marcos-kp, 2024-04-25:
nope, didn't work for me. Now I am not getting any contact with the sensor :
[17:14:33][I][fingerprint_grow:295]: Deleting fingerprint in slot 3
[17:14:34][E][fingerprint_grow:467]: No response received from reader
 :(

tergontor, 2024-04-25:
In case it helps, these are my connections

ESP8266					fingerprint			power
3V3   ------------------------     (red and white)
D7  (RX)-------------------	 (yelow-TX)
D6  -------------------------     (blue)
D5  (TX)----------------------    (purple-RX)
GND  -------------------------   (black )------------------   5v -
5V    ----------------------------------------------------      5v +

Marcos-kp, 2024-04-25:
Thanks! Will try it like that. I was using D1 and D2, also tried D7 and D8 for tx and rx.
Do you mind sharing the rest of your configuration?

Marcos-kp, 2024-04-25:
Ok, I had some progress. After intalling wirelessly and looking at the usb logs, I got a wrong password message! That is so strange, since I did the whole process of updating the password and my sensor was working even after disconnecting it from the power. The firmware update kinda reverted or give me some issues. I tried without a password and I do not get an error as before, but the sensor is still not working.

Here are some logs:

`����n��r��n|�l�$`c��|s�l�g��o�l ��;�l�d��
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
[I][logger:034]: Log initialized
[C][ota:483]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][scheduler:063]: set_interval(name='', interval=60000, offset=9504)
[C][uart.arduino_esp8266:059]: Setting up UART bus...
[VV][scheduler:063]: set_interval(name='update', interval=6000, offset=2256)
[C][fingerprint_grow:060]: Setting up Grow Fingerprint Reader...
[D][fingerprint_grow:232]: Checking password
[E][fingerprint_grow:240]: Wrong password
[E][component:119]: Component fingerprint_grow was marked as failed.
[E][component:164]: Component fingerprint_grow set Error flag: unspecified
[VV][scheduler:063]: set_interval(name='update', interval=500, offset=196)
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 
[V][wifi_esp8266:061]: Enabling STA.
mode : sta()
add if0
[V][wifi_esp8266:556]: Event: Changed Mode old=OFF new=STA
wifi evt: 8
nul mode, fpm auto sleep set:enalbe
sleep disable
[D][wifi:462]: Starting scan...
[VV][scheduler:225]: Running interval '' with interval=60000 last_execution=4294905026 (now=7340)
[VV][scheduler:225]: Running interval 'update' with interval=6000 last_execution=4294966287 (now=7340)
[V][text_sensor:013]: 'Fingerprint State': Received new state -
[D][text_sensor:064]: 'Fingerprint State': Sending state '-'
[W][component:157]: Component wifi set Warning flag: unspecified
[VV][scheduler:225]: Running interval 'update' with interval=6000 last_execution=4991 (now=10991)
[V][text_sensor:013]: 'Fingerprint State': Received new state -
[D][text_sensor:064]: 'Fingerprint State': Sending state '-'
scandone
[D][wifi:477]: Found networks:
[I][wifi:520]: - '' [redacted]▂▄▆█
[D][wifi:522]:     Channel: 3
[D][wifi:523]:     RSSI: -54 dB
[I][wifi:520]: - '' [redacted]▂▄▆█
[D][wifi:522]:     Channel: 1
[D][wifi:523]:     RSSI: -70 dB
[D][wifi:525]: - [redacted] [redacted]▂▄▆█
[D][wifi:525]: - [redacted] [redacted]▂▄▆█
[I][wifi:303]: WiFi Connecting to ''...
[V][wifi:305]: Connection Params:
[V][wifi:306]:   SSID: ''
[V][wifi:309]:   BSSID: 
[V][wifi:329]:   Password: [redacted]
[V][wifi:334]:   Channel: 3
[V][wifi:340]:   Manual IP: Static IP=192.168.15.74 Gateway=192.168.15.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[V][wifi:345]:   Hidden: NO
[V][wifi_esp8266:514]: Event: Changed AuthMode old=OPEN new=WPA2 PSK
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 35
cnt 

connected with , channel 3
ip:192.168.15.74,mask:255.255.255.0,gw:192.168.15.1
ip:192.168.15.74,mask:255.255.255.0,gw:192.168.15.1
[V][wifi_esp8266:490]: Event: Connected ssid='' bssid= channel=3
[... 336 lines cut ...]
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=14827 (now=15354)
[V][fingerprint_grow:021]: No touch sensing
[C][ota:096]: Over-The-Air Updates:
[C][ota:097]:   Address: 192.168.15.74:8266
[C][ota:100]:   Using Password.
[C][ota:103]:   OTA version: 2.
[C][api:139]: API Server:
[C][api:140]:   Address: 192.168.15.74:6053
[C][api:142]:   Using noise encryption: YES
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=15327 (now=15833)
[V][fingerprint_grow:021]: No touch sensing
[D][api:102]: Accepted 192.168.15.34
[VV][api.socket:696]: 192.168.15.34: Handshake complete!
[W][component:170]: Component api cleared Warning flag
[W][component:237]: Component api took a long time for an operation (293 ms).
[W][component:238]: Components should block for at most 30 ms.
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=15827 (now=16425)
[V][fingerprint_grow:021]: No touch sensing
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=16327 (now=16827)
[V][fingerprint_grow:021]: No touch sensing
[VV][scheduler:225]: Running interval 'update' with interval=500 last_execution=16827 (now=17331)
[V][fingerprint_grow:021]: No touch sensing
[VV][scheduler:225]: Running interval 'update' with interval=6000 last_execution=11609 (now=17609)
[V][text_sensor:013]: 'Fingerprint State': Received new state -
[D][text_sensor:064]: 'Fingerprint State': Sending state '-'`

Marcos-kp, 2024-04-25:
Now I got it working again!

I reverted to my initial configuration:
```
uart:
  rx_pin: D1
  tx_pin: D2
  baud_rate: 57600
  

fingerprint_grow:
  sensing_pin: D6
  ```
  
  Apparently something in the update messed up my password and I wasn't getting any logs/feedback regarding this. Not until I installed a code and followed the log through the usb.
  
  Thanks all for the help! Now I will try to see if setting a password will work. If not, I might have to stay without one.

drublin, 2024-10-26:
i have the same issue. it randomly stopped working, now is always dead. The sensor does NOT go to sleep after timeout, though other ESP32 pins work, API works. During Enroll the sensor LED glows (with wrong color) but doesn't enroll.

@Marcos-kp , do you have any updates?