esphome/issues #5727: WebUI sensor history, min and max

geotibi, 2024-04-23:
### The problem

In the last webUI interface I can see some "data" that is not very visible. 1st I was thinking that is because of my old monitor but than I was observing the same on the laptop. Where I have the sensor value on the up and down side of the current value I can see numbers which I guess is referring to max and min value, numbers which are not very visible
[image]
Also between the sensor name and the value I can see somthing like a graph, also not very visible like in the picture 
[image]

My question is: can I do something to change the style of these numbers and the graph to be more visible?

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP8266

sbrk-r, 2024-04-25:
It looks like it is supposed to be faint until the cell of the chart is selected, then it should expand and become less faint. [src](https://github.com/esphome/esphome-webserver/blob/4673cdaf3d02eda6987185c24872e5da420c5aed/packages/v3/src/esp-entity-chart.ts#L63)

It doesn't happen for me on firefox 115.9.1esr, the cell expands but the graph does not, and opacity does not change either.

firefox:
[image]

konqueror:
[image]

I think probably it wouldn't hurt to dial up both opacity a little bit...

geotibi, 2024-04-25:
I thought that it may be a setting that can be changed in the configuration. Maybe will be added in the future an option like transparency.

vermut, 2024-09-30:
Same for me. Works fine in Edge, broken in FF 130.0.1 (64-bit)