esphome/issues #5728: Unable to use `!extend` syntax with LD2410

Pimentoso, 2024-04-23:
### The problem

In my house I am using many LD2410 presence sensors, so I wanted to use !extend to add some actions to the binary sensors. The yml is failing to compile.

main yml file
```
substitutions:
  device_name: device-test-01

packages:
  common: !include modules/common.yml
  motion_sensor: !include modules/motion_sensor.yml
  light: !include modules/light.yml

  # this contains the extend code
  motion_status_leds: !include automations/motion_status_leds.yml
```

motion_sensor.yml
```
[...]
ld2410:

binary_sensor:
  - platform: ld2410
    has_target:
      name: "${device_name}-presence"
      id: motion_sensor
    has_moving_target:
      name: "${device_name}-moving-target"
      id: moving_target_sensor
    has_still_target:
      name: "${device_name}-still-target"
      id: still_target_sensor
```

motion_status_leds.yml
```
binary_sensor:
- platform: ld2410
  has_moving_target:
    id: !extend moving_target_sensor
    on_press:
      then:
        - light.addressable_set:
            id: rgb_leds
            [...turn on a led]
    on_release: 
      then:
        - light.addressable_set:
            id: rgb_leds
            [...turn off a led]
```

I receive the following error
```
Failed config

binary_sensor.ld2410: [source automations/motion_status_leds.yml:2]
  platform: ld2410
  has_moving_target: 
    
    Source for extension of ID 'moving_target_sensor' was not found.
    id: !extend moving_target_sensor
```

I suspect it's because the `!extend` syntax cannot handle nested IDs like in the LD2410 case? Or am I doing anything wrong? Is there any workaround?

cc @quentinmit

### Which version of ESPHome has the issue?

2024.3.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

ssieb, 2024-04-24:
I think extend can only be used in the main yaml.

Pimentoso, 2024-05-03:
@ssieb does not work even by moving the contents of `motion_status_leds.yml` inside `main.yml` (after the packages declaration). Gives the same error.

ssieb, 2024-05-03:
And you can't extend sub-components like that.  You need to move it up one level.  I think that should work.

Pimentoso, 2024-05-03:
But that's literally what the extend function should do. From the original PR description #3555

>which means that a user can override a field in a package's component by referencing the same id in their config with !extend id.

However! I might have found a workaround, which is using the [Copy integration](https://esphome.io/components/copy.html#copy-binary-sensor)

Basically I made `moving_target_sensor` internal, and exposed a copy of it. Now it does not have the `has_moving_target:` section, the yml compiles. I haven't tested it yet though.

So `motion_sensor.yml` became
```
binary_sensor:
  - platform: ld2410
    has_target:
      id: motion_sensor_internal
      internal: true
    has_moving_target:
      id: moving_target_sensor_internal
      internal: true
    has_still_target:
      id: still_target_sensor_internal
      internal: true
  - platform: copy
    source_id: motion_sensor_internal
    id: motion_sensor
    name: "${device_name}-presence"
  - platform: copy
    source_id: moving_target_sensor_internal
    id: moving_target_sensor
    name: "${device_name}-moving-target"
  - platform: copy
    source_id: still_target_sensor_internal
    id: still_target_sensor
    name: "${device_name}-still-target"
```

ssieb, 2024-05-03:
You didn't want to do:
```yaml
binary_sensor:
- platform: ld2410
  id: !extend my_ld2410
  has_moving_target:
    on_press:
      then:
        - light.addressable_set:
            id: rgb_leds
            [...turn on a led]
    on_release: 
      then:
        - light.addressable_set:
            id: rgb_leds
            [...turn off a led]
```

Pimentoso, 2024-05-06:
That's not working because:
```
[id] is an invalid option for [binary_sensor.ld2410].
```
Thanks for your suggestions however. I will test my workround soon and report back.

Pimentoso, 2024-05-22:
I can confirm my workaround is working. I hope this helps anyone stumbling here. 
Final yaml files:

```
// main.yml
substitutions:
  device_name: device-test-01

packages:
  common: !include modules/common.yml // this has wifi config, i2c config, etc
  motion_sensor: !include modules/motion_sensor.yml
  leds: !include modules/leds.yml

  # this contains the extend code
  motion_status_leds: !include automations/motion_status_leds.yml
```

```
// motion_sensor,yml
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:

binary_sensor:
  - platform: ld2410
    has_target:
      id: motion_sensor_internal
      internal: true
    has_moving_target:
      id: moving_target_sensor_internal
      internal: true
    has_still_target:
      id: still_target_sensor_internal
      internal: true
  - platform: copy
    source_id: motion_sensor_internal
    id: motion_sensor
    name: "${device_name}-presence"
  - platform: copy
    source_id: moving_target_sensor_internal
    id: moving_target_sensor
    name: "${device_name}-moving-target"
  - platform: copy
    source_id: still_target_sensor_internal
    id: still_target_sensor
    name: "${device_name}-still-target"

sensor:
  - platform: ld2410
    moving_distance:
      name : "${device_name}-moving-distance"
    still_distance:
      name: "${device_name}-still-distance"
    moving_energy:
      name: "${device_name}-move-energy"
    still_energy:
      name: "${device_name}-still-energy"
    detection_distance:
      name: "${device_name}-detection-distance"

switch:
  - platform: ld2410
    bluetooth:
      name: "${device_name}-ld2410-bluetooth"

button:
  - platform: ld2410
    factory_reset:
      name: "${device_name}-ld2410-reset"
```

```
// leds.yml
light:
- platform: neopixelbus
  pin: D5
  name: "${device_name}-leds"
  id: rgb_leds
  variant: WS2811
  num_leds: 2
```

```
// motion_status_leds.yml
binary_sensor:
- id: !extend moving_target_sensor
  on_press:
    then:
      - light.addressable_set:
          id: rgb_leds
          range_from: 0
          range_to: 0
          red: 100%
          green: 0%
          blue: 0%
          color_brightness: 20%
  on_release: 
    then:
      - light.addressable_set:
          id: rgb_leds
          range_from: 0
          range_to: 0
          red: 0%
          green: 0%
          blue: 0%

```