esphome/issues #5730: ESP32 (Wroom) doent work with Dallas DS18B20 Sensor

StrSebastian, 2024-04-23:
### The problem

I tested 300 Versions of these kind of yaml. 
And all kinds of configuratiosn nothing worked. 
Sensor not found or jost nothing in the log.
I also uses all kind of resostors. 
Also multible or single sensors.
But what makes me sure about the issu: it worked first try with Arduino IDE. Got the right temprature with example code.
My guess is, the bug is somewhere in the lib ob HA build pipline.

### Which version of ESPHome has the issue?

2024.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32

### Board

Wroom 32

### Component causing the issue

Dallas Sensor DS18B20 

### Example YAML snippet

```yaml
I tested 300 Versions of these kind of yaml. 
And all kinds of configuratiosn nothing worked. 
Sensor not found or jost nothing in the log.
I also uses all kinds of resostors. 
Also multible or single sensors.
But what makes me sure about the issu: it worked first try with Arduino IDE. Got the right temprature with example code.
My guess is, the bug is somewhere in the lib ob HA.
dallas:
  pin:
    number: GPIO26<Optional tested>
    mode:
      input: true <Optional tested> 
      pullup: true<Optional tested>
  update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
nop.
```

### Additional information

https://community.simon42.com/t/probleme-mit-esp32-und-dallas-temperatur-sensor/4522/15

sbrk-r, 2024-04-24:
Try commenting out mode/input/pullup and have a hardware pullup, see if it works. Maybe mode is overriding something crucial. 

I see open feature request asking for advanced pin options, so must not be supported currently..? Maybe it was added without closing FR [FR 2499 ](https://github.com/esphome/feature-requests/issues/2499)

I use configuration like this on many boards with no issues:

```
dallas:
  - pin: GPIO16
    id: ow0
    update_interval: 10s

sensor:
  - platform: dallas
    #address: 0x6c0000052b7f0828
    dallas_id: ow0
    index: 0
    name: "ow0_t0"
    accuracy_decimals: 2
```

StrSebastian, 2024-04-25:
thx for the answer. 
Unfortunatly i tryed all versions. 
specially with and without these lines of code.
I testet your config but it didnt work again :/
console:
[19:59:27][C][dallas.sensor:075]: DallasComponent:
[19:59:27][W][dallas.sensor:080]:   Found no sensors!
[19:59:27][C][dallas.sensor:089]:   Device 'ow0_t0'
[19:59:27][C][dallas.sensor:089]:     Device Class: 'temperature'
[19:59:27][C][dallas.sensor:089]:     State Class: 'measurement'
[19:59:27][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[19:59:27][C][dallas.sensor:089]:     Accuracy Decimals: 2
[19:59:27][C][dallas.sensor:091]:     Index 0

 here my code:
substitutions:
  name: esphome-web-xxx
  friendly_name: ESPHome Web xxxx

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
#web_server:

dallas:
  - pin: GPIO16
    id: ow0
    update_interval: 10s

sensor:
  - platform: dallas
    #address: 0x6c0000052b7f0828
    dallas_id: ow0
    index: 0
    name: "ow0_t0"
    accuracy_decimals: 2

swoboda1337, 2024-04-30:
Did you pull the usb/serial logs. Should see "Setting up DallasComponent..."

closed 2024-05-05 by StrSebastian

danila1994, 2024-06-25:
I don't know if I am late or not but still.
ESPHome: 2024.6.2

//Example configuration entry

one_wire:

    platform: gpio
    pin: 13
    id: dallas_1
    platform: gpio
    pin: 12
    id: dallas_2

sensor:

    platform: dallas_temp
    address: 0xff03109794599728
    name: "Water Temperature from CV"
    one_wire_id: dallas_1 # Explicitly specify the one_wire_id
    platform: dallas_temp
    address: 0xf203029794031128
    name: "Water Temperature to CV"
    one_wire_id: dallas_2 # Explicitly specify the one_wire_id