esphome/issues #5732: INA219 component not returning signed power

tomlut, 2024-04-24:
### The problem

This is a bi-directional device. 

A negative current and a positive voltage should produce a negative power (P = I x V).

However the power reported by ESPHome is always positive for the INA219. This makes it impossible to track battery state of charge without a work-around.

[image]
[image]
[image]

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2024.4.4

### What platform are you using?

ESP32

### Board

QuinLED ESP32

### Component causing the issue

INA219

### Example YAML snippet

```yaml
sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "Shed Battery Current"
      filters:
        - multiply: -1
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
    power:
      name: "Shed Battery Power"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
    bus_voltage:
      name: "Shed Battery Voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
    shunt_voltage:
      name: "Shed Shunt Voltage"
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
    max_voltage: 15.0V
    max_current: 3A
    update_interval: 6s
```

ssieb, 2024-04-24:
What value do you get when it should be negative?

tomlut, 2024-04-24:
You can see it in the history graphs I posted. It is `value|abs`.  i.e. the positive magnitude.

tomlut, 2024-04-24:
Look at the values from midnight to 7am. The power is positive, but the current is negative. Voltage is always positive.

tomlut, 2024-05-01:
Currently working around this issue with the following config:
```
sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "Shed Battery Current"
      id: battery_current
      filters:
        - multiply: -1
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
      on_value:
        then:
          - component.update: battery_power
    # power:
    #   name: "Shed Battery Power"
    #   filters:
    #     - sliding_window_moving_average:
    #         window_size: 10
    #         send_every: 10
    #         send_first_at: 1
    bus_voltage:
      name: "Shed Battery Voltage"
      id: battery_voltage
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
      on_value:
        then:
          - component.update: battery_power
    max_voltage: 15.0V
    max_current: 3A
    update_interval: 6s

  - platform: template
    name: "Shed Battery Power"
    id: battery_power
    unit_of_measurement: 'W'
    state_class: measurement
    device_class: power
    lambda: |-
      if ( (id(battery_current).state) && (id(battery_voltage).state) ){
        return (id(battery_current).state * id(battery_voltage).state);
      } else {
        return NAN;
      }

tomlut, 2024-05-27:
And the [INA260](https://esphome.io/components/sensor/ina260.html) does the same thing. Positive power no matter if the current is positive or negative.

neel-m, 2024-12-23:
The INA219 (and the others in the family) have this as a feature, likely since they use 16-bit integer registers and when you multiply a 12-bit unsigned number (bus voltage) by a 15-bit (+sign) number (current) and need the results to fit into 16 bits you have to do some magic and that appears to include throwing out the sign bit to get an extra bit of resolution.

Tasmota deals with this by ignoring the power register and calculating power (signed) in the ESP, since it is pretty easy math and not done that frequently.