esphome/issues #5733: ESP restarts when I send long command via api.service

Vizualni, 2024-04-24:
### The problem

ESP restarts when I send long command via api.service, but it doesn't when I send it directly as remote_transmitter.transmit_raw. Here is the config I am using:

```yaml
esphome:
  name: klima_dnevni
  friendly_name: KlimaDnevni

esp8266:
  board: nodemcu

# Enable logging
logger:
  level: NONE

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: "foobar"
  services:
    - service: send_raw_command
      variables:
        command: int[]
      then:
        - remote_transmitter.transmit_raw:
            carrier_frequency: 38kHz              
            code: !lambda 'return command;'
switch:
  - platform: template
    name: "Send payload"
    turn_on_action:
      remote_transmitter.transmit_raw:
        carrier_frequency: 38kHz  
        code: [305, -33377, 122, -41051, 426, -426, 426, -426, 396, -426, 457, -396, 426, -426, 426, -24789, 3380, -1705, 396, -1279, 396, -457, 396, -457, 396, -457, 426, -1249, 426, -426, 426, -426, 426, -426, 426, -396, 457, -1249, 426, -426, 426, -1279, 396, -1310, 396, -457, 396, -1279, 396, -1310, 396, -1279, 396, -1310, 426, -1249, 457, -396, 426, -426, 426, -1279, 426, -396, 426, -426, 426, -426, 426, -426, 426, -426, 426, -426, 396, -457, 396, -457, 396, -457, 396, -426, 426, -1279, 396, -457, 396, -1279, 426, -426, 426, -426, 426, -426, 426, -1249, 426, -1279, 426, -426, 426, -396, 426, -426, 426, -426, 426, -457, 396, -426, 426, -426, 396, -457, 396, -457, 396, -457, 396, -426, 396, -457, 396, -457, 396, -457, 396, -426, 457, -396, 426, -1279, 426, -1249, 426, -1279, 426, -426, 426, -1249, 426, -457, 396, -1310, 396, -1279, 396, -34169, 3319, -1736, 396, -1310, 396, -426, 396, -457, 396, -457, 396, -1279, 396, -457, 396, -457, 396, -457, 426, -396, 457, -1249, 426, -426, 426, -1249, 457, -1249, 426, -426, 426, -1279, 396, -1310, 396, -1310, 396, -1279, 396, -1310, 396, -426, 396, -457, 396, -1310, 426, -396, 457, -396, 426, -426, 426, -426, 426, -426, 426, -396, 426, -426, 426, -426, 426, -426, 426, -457, 396, -426, 396, -1310, 396, -457, 396, -426, 396, -457, 396, -457, 396, -1310, 396, -426, 396, -457, 426, -426, 426, -426, 426, -396, 457, -396, 426, -426, 426, -426, 426, -426, 426, -396, 426, -426, 426, -426, 426, -457, 396, -457, 396, -426, 396, -457, 396, -457, 396, -457, 396, -426, 426, -1279, 396, -457, 396, -1310, 426, -396, 426, -1279, 426, -396, 426, -34169, 3319, -1705, 426, -1279, 426, -396, 457, -396, 426, -426, 426, -1279, 426, -396, 426, -426, 426, -457, 396, -457, 396, -1279, 396, -457, 396, -1310, 396, -1279, 396, -457, 396, -1310, 426, -1249, 426, -1279, 426, -1249, 426, -1279, 426, -426, 426, -396, 426, -1310, 396, -457, 365, -487, 365, -457, 365, -487, 365, -487, 396, -457, 365, -457, 396, -457, 365, -487, 365, -487, 365, -487, 396, -426, 396, -457, 396, -457, 396, -457, 396, -426, 426, -426, 396, -457, 396, -1310, 396, -457, 396, -457, 365, -1340, 365, -487, 365, -457, 365, -487, 365, -487, 365, -487, 365, -457, 365, -1340, 396, -457, 396, -457, 396, -1279, 396, -457, 396, -457, 396, -426, 426, -426, 396, -457, 396, -457, 396, -487, 365, -457, 365, -487, 396, -457, 365, -487, 365, -487, 365, -487, 365, -457, 365, -487, 365, -1310, 365, -487, 396, -1310, 396, -457, 396, -426, 396, -457, 396, -457, 396, -457, 396, -426, 396, -457, 396, -457, 396, -487, 365, -487, 365, -457, 365, -487, 365, -487, 365, -487, 365, -457, 365, -487, 365, -487, 365, -1340, 365, -1340, 365, -487, 365, -457, 396, -457, 396, -426, 396, -487, 365, -487, 365, -457, 396, -457, 365, -518, 335, -518, 335, -1340, 365, -1340, 335, -518, 335, -487, 365, -487, 335, -518, 335, -518, 335, -487, 365, -487, 396, -457, 365, -487, 365, -487, 365, -457, 396, -457, 396, -457, 365, -487, 365, -457, 396, -457, 396, -487, 335, -1370, 335, -487, 365, -487, 365, -487, 335, -518, 335, -518, 335, -1340, 335, -1370, 335, -487, 365, -487, 365, -487, 365, -487, 365, -487, 365, -457, 396, -457, 365, -1340, 365, -487, 365, -487, 365, -487, 335, -518, 335, -518, 335, -487, 365, -487, 335, -518, 335, -518, 335, -1340, 365, -1340, 335, -518, 365, -457, 396, -457, 396, -457, 365, -1340, 365, -101502]

ota:
  password: "foobar"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Klima Fallback Hotspot"
    password: "foobar"

remote_transmitter:
  pin: 5
  carrier_duty_percent: 50%

captive_portal:
```

When I take the same payload that's in the yaml provided and send it via developer tools > services like this:

[image]

my esp restarts. When I send it by calling the "Send Payload" action, it works. 

### Which version of ESPHome has the issue?

2024.2.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.4

### What platform are you using?

ESP8266

### Board

nodemcu

closed 2024-04-24 by ssieb

Vizualni, 2024-04-25:
Why was this closed without providing any meaningful help?

> This isn't something for filing an issue

Of course it is. There obviously is a bug and it's very easy to reproduce. I wrote all the steps on how to reproduce it.

Vizualni, 2024-04-25:
Here is the loom recording: https://www.loom.com/share/2a69b22688cb46539c2e3d08cdf65cf9?sid=6701a867-f18a-40b6-879a-98752164d3ec

And here is the output from the serial:

```
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffb90 end: 3fffffd0 offset: 0000
3ffffb90:  3fff4c12 00000000 3fff5c9c 40100e41
3ffffba0:  000000fe 00000000 00000000 00000000
3ffffbb0:  00000000 00000000 00000000 3ffffd00
3ffffbc0:  3fffefd0 000000fe 00000002 3fff0818
3ffffbd0:  00000000 00000001 3fffac38 3ffffc70
3ffffbe0:  3ffffc70 4021e13b 0000092c 4022cdbe
3ffffbf0:  38079138 3ffe8614 00000938 4022cddd
3ffffc00:  0000ad60 4021e13b 0000092c 4022c41c
3ffffc10:  3ffffc10 0000092c 3ffffc3c 4021e13b
3ffffc20:  3ffffcb0 3fff1a2c 3fff17cc 4021e759
3ffffc30:  40100c00 3ffe8614 00000938 00000000
3ffffc40:  00000000 00000000 0000092c 4022c407
3ffffc50:  00000000 0000092c 3ffffc70 00000000
3ffffc60:  3ffffcb0 3fff17cc 3fff1a24 4021e6c3
3ffffc70:  3fffb584 3fffbeb0 3fffbeb0 401012b4
3ffffc80:  3fff1a2c 4021e718 0000092c 4022c407
3ffffc90:  3ffffcec 0000092c 3ffffcb0 3fff4610
3ffffca0:  4021e684 3ffffce0 3fff17cc 4021e464
3ffffcb0:  3fffac44 3fffb570 3fffb570 4022c407
3ffffcc0:  00003d88 0000092c 3ffffce0 4021e14c
3ffffcd0:  4021e15c 4021e43c 3fff17cc 4021e19c
3ffffce0:  3fffa304 3fffac30 3fffac30 3fff99c4
3ffffcf0:  3fffa2f0 3fffa2f0 3fff9084 3fff99b0
3ffffd00:  3fff99b0 3fff8744 3fff9070 3fff9070
3ffffd10:  4021e15c 3fff1734 3fff45ac 4021e3f9
3ffffd20:  3fff7e04 3fff8730 3fff8730 00000000
3ffffd30:  00000000 000006e6 3fffc228 00000002
3ffffd40:  3fff4598 3ffffda0 00000008 40211fdb
3ffffd50:  00000001 00000000 000006e6 00000000
3ffffd60:  3fff5598 000006e6 00000002 3fff1710
3ffffd70:  3ffffda0 00000000 3fff1710 4020cae8
3ffffd80:  3fff3e74 00000000 00000000 3fff1634
3ffffd90:  3fff0c24 4023151c 3fff34f4 402115ae
3ffffda0:  4023151c acddc8e2 3fff4534 3fff4598
3ffffdb0:  3fff4598 00000000 00000000 4020efce
3ffffdc0:  3fff558c 000006f2 00000702 00000000
3ffffdd0:  00000000 00030c8e 83958106 038684c8
3ffffde0:  3fff0c24 3fff0c20 3fff35b4 3fff1634
3ffffdf0:  3fff0c24 00000000 3fff34f4 4020d3d8
3ffffe00:  ffffffd2 0bc7a1dd 401039a6 00000100
3ffffe10:  3ffeacb8 7fffffff 00002200 00000001
3ffffe20:  00000001 00000080 4000050c 3fffc278
3ffffe30:  3ffeacb8 3fffc200 00000022 0bc7a1dd
3ffffe40:  3ffeacc4 3fff558c 3fff5c8e 3fff5c8e
3ffffe50:  3fff002a 00000004 000006ee 00000bb1
3ffffe60:  40217467 00000030 00000010 3fff1634
3ffffe70:  3fff0c24 3fff0c20 3fff1634 40211d05
3ffffe80:  00004bc6 00000000 3ffefba4 fffffffe
3ffffe90:  00000000 3fffc6fc 00000000 3fff128c
3ffffea0:  00030402 3fff138c 3fff128c 00000030
3ffffeb0:  00000002 3fff0e31 3fff2b74 402134f8
3ffffec0:  00000000 3ffe8f62 00000002 00000000
3ffffed0:  3fff2b64 00000000 3ffe8ccc 402133d0
3ffffe80:  00004bc6 00000000 3ffefba4 fffffffe
3ffffe90:  00000000 3fffc6fc 00000000 3fff128c
3ffffea0:  00030402 3fff138c 3fff128c 00000030
3ffffeb0:  00000002 3fff0e31 3fff2b74 402134f8
3ffffec0:  00000000 3ffe8f62 00000002 00000000
3ffffed0:  3fff2b64 00000000 3ffe8ccc 402133d0
3ffffee0:  00000000 0000001c c147ae14 3ffefba0
3ffffef0:  00000000 00030c8a 028f5c28 03867f92
3fffff00:  3fffdad0 3ffefba0 3ffefba0 00000001
3fffff10:  3fff2894 3ffefd48 3fff28c8 40225318
3fffff20:  40263d96 3fff128c 0000001c 00000000
3fffff30:  00000000 00030c8a 14bc6a7e 03867fa7
3fffff40:  00000000 00030c8a 15c28f5c 03867fa9
3fffff50:  3fff2bcc 00000002 3fffff80 3fff1634
3fffff60:  3fff2bd0 3fff1634 3fffff80 3fff1634
3fffff70:  3fff2bd0 00000002 3ffefba0 4021aa62
3fffff80:  00030c8a 3fff1634 3fff1b00 3ffeff98
3fffff90:  3fff2bd4 3fff195c 3ffefbd4 3ffeff98
3fffffa0:  3fffdad0 00000000 3ffeff84 3ffeff98
3fffffb0:  3fffdad0 00000000 3ffeff84 4022c8e8
3fffffc0:  feefeffe feefeffe 3ffe8610 40100449
<<<stack<<<

last failed alloc call: 4021E13B(2348)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4021e13b

 ets Jan  8 2013,rst cause:2, boot mode:(3,5)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007d1d0
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[I][logger:359]: Log initialized
[C][ota:483]: There have been 3 suspected unsuccessful boot attempts.

```

swoboda1337, 2024-04-25:
Exception is an out of memory error: Unhandled C++ exception: OOM
I tried and this works fine on esp32 which has more memory. Most likely out of heap on esp8266.