esphome/issues #5736: Template Switch State Problem

Salvora, 2024-04-26:
### The problem

I have two issues with template switches after updating to 2024.4.1. I don't remember which version I was before since I did not check it before but it was at least 9 months old. After updating
Case 1:
Every time I reboot ESP device, it engages the template switch
There is a GPIO binary sensor to get the TV state and an IR transmitter to control the TV. There is a template switch where all of that comes together to create a switch for TV power.
```
# Monitor TV State
binary_sensor:
  - platform: gpio
    name: "$roomname TV Durumu"
    internal: True
    entity_category: "diagnostic"
    device_class: "power"
    id: emre_tv_power
    pin:
      number: 34 # Input only pin without internal pull-up or pull-down resistors
```
```
  # TV Switches
  - platform: template
    name: "$roomname TV Güç"
    id: emre_tv_pswitch
    icon: mdi:power
    device_class: "switch"
    lambda: |-
      if (id(emre_tv_power).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - remote_transmitter.transmit_lg:
          data: 0x20DF10EF
          nbits: 32
    turn_off_action:
      - remote_transmitter.transmit_lg:
          data: 0x20DF10EF
          nbits: 32
```
For some reason, every time the device reboots, the action is engaged. I did not have this issue pre-update.

Case 2:
Global variable states are not preserved and do not survive the reboots. I highly suspect that this is related to Case 1. I have another template switch that controls the screen of a tm1637 which gets the screen state from a global variable. This global variable is set to restore_value but it is not working.

```
switch:
  # On/Off Switch for Display
  - platform: template
    name: "$devicename Ekran"
    icon: mdi:monitor
    device_class: "switch"
    entity_category: "config"
    lambda: !lambda |-
      if (id(display_status) == true) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - globals.set:
          id: display_status
          value: "true"
      - component.update: tm1637_display
    turn_off_action:
      - globals.set:
          id: display_status
          value: "false"
      - component.update: tm1637_display
```
[... 185 lines cut ...]
[06:34:38][C][debug:072]:     Icon: 'mdi:counter'
[06:34:38][D][debug:079]: ESPHome version 2024.4.1
[06:34:38][D][debug:083]: Free Heap Size: 197264 bytes
[06:34:38][D][debug:113]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[06:34:38][D][debug:162]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[06:34:38][D][debug:170]: ESP-IDF Version: v4.4.2
[06:34:38][D][debug:175]: EFuse MAC: 9C:9C:1F:C7:00:D4
[06:34:38][D][debug:275]: Reset Reason: Software Reset CPU
[06:34:38][D][debug:320]: Wakeup Reason: Unknown
[06:34:38][D][switch:012]: 'Emre Oda TV Güç' Turning ON.
[06:34:38][W][component:237]: Component api took a long time for an operation (72 ms).
[06:34:38][W][component:238]: Components should block for at most 30 ms.
[06:34:39][D][dht:048]: Got Temperature=28.5°C Humidity=32.6%
[06:34:39][W][component:170]: Component dht.sensor cleared Warning flag
[06:34:39][D][binary_sensor:036]: 'Emre Oda TV Durumu': Sending state ON
```

### Additional information

Looking at the logs, for both of the template switches I am seeing an "always OFF" mode. I tried to modify this by checking these docs
https://esphome.io/components/switch/#base-switch-configuration

and tried multiple combinations but, it is not changing. I tried "RESTORE_DEFAULT_OFF", "RESTORE_DEFAULT_ON", "ALWAYS_ON", and "DISABLED". For some reason, the logs show this as OFF all the time no matter what config you use.

I suspect that during shutdown or boot up or in both at the same time, the board turns off the switches and the lambda is ignored.

Masterz69, 2024-04-26:
Just guess, try to handle case when `emre_tv_power` has no state:

```yaml
    lambda: |-
      if (id(emre_tv_power).has_state()) {
        if (id(emre_tv_power).state) {
          return true;
        } else {
          return false;
        }
      } else {
        return {};
      }
```

Salvora, 2024-04-27:
> Just guess, try to handle case when `emre_tv_power` has no state:
> 
> ```yaml
>     lambda: |-
>       if (id(emre_tv_power).has_state()) {
>         if (id(emre_tv_power).state) {
>           return true;
>         } else {
>           return false;
>         }
>       } else {
>         return {};
>       }
> ```

Nope, does not work. Still the same behaviour. The binary sensor should always have some state in any case.

Salvora, 2024-04-29:
Tried an USB flash instead of OTA in case some configuration file is not getting a proper update. Still the same behaviour.

Salvora, 2024-04-30:
issue closed.
RESTORE_DEFAULT_OFF", "RESTORE_DEFAULT_ON", "ALWAYS_ON", and "DISABLED, works now as it was my mistake and I was editing wrong config files.

closed 2024-04-30 by Salvora