esphome/issues #5739: BOX3 still makes popping sounds when playing audio and on startup

tannisroot, 2024-04-26:
### The problem

Hi, I'm running both ESPHome and Home Assistant versions that support new Native API for audio.
While it solved the issue of audio itself being glitchy, audio component is still problematic.
Specifically, when the unit starts up, right before it begins to play audio or if I yank the power out of it, it makes a very unpleasant and irritating popping sound. I've made a recording of it, and in it you can hear the pop before it plays the response.
[Voice 100-1.zip](https://github.com/esphome/issues/files/15132525/Voice.100-1.zip)
To check whether I might have a faulty Box3B unit, I flashed back to factory demo firmware,  I did not notice this problem 
 neither on startup nor when playing audio demos.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.5.0b0

### What platform are you using?

ESP32

### Board

ESP32-S3-BOX3B

### Example YAML snippet

```yaml
substitutions:
  name: "esp32-s3-box-3b"
  friendly_name: esp32-s3-box-3b
packages:
  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: MY_API_KEY

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Log output that was running while I reproduced the problem:

INFO ESPHome 2024.4.1
INFO Reading configuration /config/esp32-s3-box-3b.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.1.12 using esphome API
INFO Successfully connected to esp32-s3-box-3b @ 192.168.1.12 in 0.130s
INFO Successful handshake with esp32-s3-box-3b @ 192.168.1.12 in 0.259s
[18:01:52][I][app:100]: ESPHome version 2024.4.1 compiled on Apr 26 2024, 17:47:28
[18:01:52][I][app:102]: Project esphome.voice-assistant version 1.0
[18:01:52][C][wifi:580]: WiFi:
[18:01:52][C][wifi:408]:   Local MAC: 30:30:F9:5A:BF:DC
[18:01:52][C][wifi:413]:   SSID: [redacted]
[18:01:52][C][wifi:416]:   IP Address: 192.168.1.12
[... 208 lines cut ...]
[18:03:30][D][voice_assistant:439]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[18:03:30][D][voice_assistant:180]: Waiting for speech...
[18:03:30][D][voice_assistant:439]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[18:03:32][D][voice_assistant:193]: VAD detected speech
[18:03:32][D][voice_assistant:439]: State changed from WAITING_FOR_VAD to START_PIPELINE
[18:03:32][D][voice_assistant:445]: Desired state set to STREAMING_MICROPHONE
[18:03:32][D][voice_assistant:210]: Requesting start...
[18:03:32][D][voice_assistant:439]: State changed from START_PIPELINE to STARTING_PIPELINE
[18:03:32][D][voice_assistant:460]: Client started, streaming microphone
[18:03:32][D][voice_assistant:439]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[18:03:32][D][voice_assistant:445]: Desired state set to STREAMING_MICROPHONE
[18:03:32][W][component:237]: Component voice_assistant took a long time for an operation (66 ms).
[18:03:32][W][component:238]: Components should block for at most 30 ms.
[18:03:32][D][voice_assistant:563]: Event Type: 1
[18:03:32][D][voice_assistant:566]: Assist Pipeline running
[18:03:32][D][voice_assistant:563]: Event Type: 9
[18:03:37][D][voice_assistant:563]: Event Type: 0
[18:03:37][D][voice_assistant:563]: Event Type: 2
[18:03:37][D][voice_assistant:653]: Assist Pipeline ended
[18:03:37][D][voice_assistant:439]: State changed from STREAMING_MICROPHONE to WAIT_FOR_VAD
[18:03:37][D][voice_assistant:445]: Desired state set to WAITING_FOR_VAD
[18:03:37][D][voice_assistant:180]: Waiting for speech...
[18:03:37][D][voice_assistant:439]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
```
```

tannisroot, 2024-05-02:
It's an issue in 2024.5.0 too.

Findarato, 2024-05-16:
Glad I am not the only one having an issue with this. I am not even getting sound.

tannisroot, 2024-05-18:
> Glad I am not the only one having an issue with this. I am not even getting sound.
> 
> Looks like there are a lot of other people #5791 complaining as well.

It should be noted this issue is specifically for the popping sound, which for me occures even on 2024.4.2 version of ESPHome has working sound.

ChristophCaina, 2024-06-03:
for me, this was "always" the case - since I have my box3 and it also happens on my older box.
You can reproduce the case in the following ways:

Box3 and Box:
Set the box to mute - and unmute them.
The "pop" sound will be louder on the Box than on the Box3

Box3:
if you have enabled the radar and the device is going to 'sleep' - and then also be muted - and wake it up, you will also hear the "pop"

tannisroot, 2024-06-03:
> for me, this was "always" the case - since I have my box3 and it also happens on my older box. You can reproduce the case in the following ways:
>
> Box3 and Box: Set the box to mute - and unmute them. [...]

Yep, same experience as you, it was like this since I got the Box3.

b-col, 2024-08-12:
Same issue for me with the Box-3

genehand, 2024-08-30:
FWIW, the `media_player` setup from https://github.com/gnumpi/esphome_audio as used in https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome has fixed this for me.  Volume is controllable and goes noticeably louder, plus the touchscreen works with it as well. 

Lots of differences there but I suspect `keep_pipeline_alive: true` prevents the re-initialization and popping sound.

tannisroot, 2024-08-30:
> FWIW, the `media_player` setup from https://github.com/gnumpi/esphome_audio as used in https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome has fixed this for me. Volume is controllable and goes noticeably louder, plus the touchscreen works with it as well.
> 
> Lots of differences there but I suspect `keep_pipeline_alive: true` prevents the re-initialization and popping sound.
> 
> Edit: it does still pop on startup, but then it's good.

Would you mind sharing how you adapter it into the normal box3 config? I tried myself but seem to be getting compile errors

genehand, 2024-08-31:
> Would you mind sharing how you adapted it into the normal box3 config? I tried myself but seem to be getting compile errors

Sure, I have a sprawling config including converting to lvgl so I'll link the relevant parts:

- Replace the esp_adf external_component with [esphome_audio](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/316c40d5725e4690e43987d06a7b27a34d8e5103/s3b.yaml#L76C1-L82C18)
- Replace the `esp_adf`, `microphone`, and `speaker` sections with [these](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/316c40d5725e4690e43987d06a7b27a34d8e5103/s3b.yaml#L388-L450)
- In the `voice_assistant` section, replace `speaker` with [media_player](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/316c40d5725e4690e43987d06a7b27a34d8e5103/s3b.yaml#L462)
- The [touchscreen config](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/316c40d5725e4690e43987d06a7b27a34d8e5103/s3b.yaml#L1289-L1293) is optional, there's also a sensor for the [red circle](https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/316c40d5725e4690e43987d06a7b27a34d8e5103/s3b.yaml#L1313-L1315) at the bottom

If you're not on esphome 2024.8.1 yet, be sure to run Clean Build Files before compiling. Hope that helps!

tannisroot, 2024-08-31:
> Sure, I have a sprawling config including converting to lvgl so I'll link the relevant parts:
>
>     * Replace the esp_adf external_component with [...]

Yeah I did all of that by myself by looking at Bobba's config and it either doesn't work at all or doesn't compile.
Would you mind sharing your entire config? Maybe I made some mistake or missed something when swapping stuff out. I would really appreciate it!

genehand, 2024-09-04:
What are the compiler errors you're getting?