esphome/issues #5742: CustomAPIDevice state subscription issues

olicooper, 2024-04-26:
### The problem

Two issues I have spotted when using the `CustomAPIDevice::subscribe_homeassistant_state` function..

1. I am getting no updates when subscribing to `timer` attribute updates i.e. `this->subscribe_homeassistant_state(&MyComponent::on_entity_attr_finishes_at_update_, "timer.test_timer", "finishes_at");`. It is working it fine for lots of other entity types, just not this one. It could be a Home Assistant issue though?

2. When subscribing to attribute updates which contain arrays of items, I get back the python 'to_string' equivalent. i.e. calling `this->subscribe_homeassistant_state(&MyComponent::on_entity_attr_supported_color_modes_update_, "light.my_light", "supported_color_modes");` I get the response `[<ColorMode.COLOR_TEMP: 'color_temp'>, <ColorMode.HS: 'hs'>, <ColorMode.XY: 'xy'>]` instead of `color_temp,hs,xy`. I also noticed that I get the return value `None` for certain entities too. I think the returned value should not be blindly to_string'd and instead be converted to an acceptable c++ equivalent?

### Which version of ESPHome has the issue?

2023.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.3.3

### What platform are you using?

ESP32

### Board

NSPanel

### Component causing the issue

CustomAPIDevice

### Example YAML snippet

```yaml
N/A
```

olicooper, 2024-04-26:
I have found this which may be related: https://github.com/home-assistant/core/issues/112798

ssieb, 2024-04-26:
Are you aware that custom components are deprecated and shouldn't be used?
Also, your esphome version is getting old.

olicooper, 2024-04-26:
> Are you aware that custom components are deprecated and shouldn't be used? Also, your esphome version is getting old.

What is the alternative to custom components?

ssieb, 2024-04-26:
It's already deprecated.  We just haven't decided on when to remove it.
I suggest you come ask on discord about what you're trying to do.  It's very unlikely that you need to do this.

closed 2024-05-02 by olicooper