esphome/issues #5745: BLE_CLIENT seams not to trigger on_connect to publish state to a text_sensor

designerferro, 2024-04-28:
### The problem

I've setup a ESPhome BLE proxy on a ESP32 to remotely press a button on a SwitchBot. Everything seams to work fine as the button is pressed by the SwitchBot, but the action on_connect and on_disconnect don't trigger the action i've configured on my yaml.

My yaml seams ok, but, although the connection is estabelished, state of the text_sensor doesn't change.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.4

### What platform are you using?

ESP32

### Board

Mini D1 ESp32

### Component causing the issue

BLE_CLIENT

### Example YAML snippet

```yaml
esphome:
    name: haproxy
    area: Basement

substitutions:
  esphome_name: haproxy

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifissid
  password: !secret wifipass
  fast_connect: True
captive_portal:
logger:
  level: INFO
web_server:
  port: 80
  ota: off
api:
  encryption:
    key: !secret apikey
  reboot_timeout: 0s
ota:
  safe_mode: True
  password: !secret otapass

esp32_ble_tracker:

ble_client:
  - mac_address: DF:60:8E:88:0D:F4
    id: SwitchBot0DF4_ble_client
    name: "SwitchBot0DF4 ble_client"
    auto_connect: false
    on_connect:
      then:
        - text_sensor.template.publish:
            id: SwitchBot0DF4_Connection_sensor
            state: "Connected"
    on_disconnect:
      then:
        - text_sensor.template.publish:
            id: SwitchBot0DF4_Connection_sensor
            state: "Disconnected"

bluetooth_proxy:
  active: true
time:
  - platform: homeassistant
sensor:
  - platform: uptime
    name: "${esphome_name} - Uptime Sensor"
  - platform: wifi_signal
    name: "${esphome_name} - WiFi Signal"
    update_interval: 60s
  - platform: ble_rssi
    name: "SwitchBot0DF4 BLE RSSI"
    id: SwitchBot0DF4_rssi
    mac_address: DF:60:8E:88:0D:F4
text_sensor:
  - platform: template
    icon: "mdi:connection"
    id: SwitchBot0DF4_Connection_sensor
    name: "SwitchBot0DF4 template Connection sensor"
```

### Anything in the logs that might be useful for us?

```txt
[02:03:43][I][esp32_ble_client:084]: [1] [DF:60:8E:88:0D:F4] Disconnecting.
[02:03:43][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_CLOSE_EVT
[02:03:43][D][esp32_ble_client:172]: [0] [DF:60:8E:88:0D:F4] ESP_GATTC_DISCONNECT_EVT, reason 22
[02:03:43][D][esp32_ble_client:172]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_DISCONNECT_EVT, reason 22
[02:03:50][D][sensor:094]: 'SwitchBot0DF4 BLE RSSI': Sending state -56.00000 dBm with 0 decimals of accuracy
[02:03:50][D][sensor:094]: 'haproxy - Uptime Sensor': Sending state 50.18200 s with 0 decimals of accuracy
[02:03:51][D][sensor:094]: 'SwitchBot0DF4 BLE RSSI': Sending state -43.00000 dBm with 0 decimals of accuracy
[02:04:00][I][bluetooth_proxy:278]: [1] [DF:60:8E:88:0D:F4] Connecting v3 with cache
[02:04:00][D][esp32_ble_tracker:215]: Pausing scan to make connection...
[02:04:00][I][esp32_ble_client:067]: [1] [DF:60:8E:88:0D:F4] 0x01 Attempting BLE connection
[02:04:00][D][esp32_ble_client:110]: [0] [DF:60:8E:88:0D:F4] ESP_GATTC_CONNECT_EVT
[02:04:00][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_CONNECT_EVT
[02:04:00][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_OPEN_EVT
[02:04:01][I][esp32_ble_client:154]: [1] [DF:60:8E:88:0D:F4] Connected
[02:04:01][D][esp32_ble_tracker:266]: Starting scan...
[02:04:01][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_REG_FOR_NOTIFY_EVT
[02:04:01][D][esp32_ble_client:306]: [1] [DF:60:8E:88:0D:F4] Event 46
[02:04:01][D][esp32_ble_client:188]: [1] [DF:60:8E:88:0D:F4] cfg_mtu status 0, mtu 23
[02:04:01][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_WRITE_DESCR_EVT
[02:04:01][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_WRITE_CHAR_EVT
[02:04:01][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_NOTIFY_EVT
[02:04:01][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_WRITE_CHAR_EVT
[02:04:02][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_NOTIFY_EVT
[02:04:10][I][esp32_ble_client:084]: [1] [DF:60:8E:88:0D:F4] Disconnecting.
[02:04:10][D][esp32_ble_client:110]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_CLOSE_EVT
[02:04:10][D][esp32_ble_client:172]: [0] [DF:60:8E:88:0D:F4] ESP_GATTC_DISCONNECT_EVT, reason 22
[02:04:10][D][esp32_ble_client:172]: [1] [DF:60:8E:88:0D:F4] ESP_GATTC_DISCONNECT_EVT, reason 22
```

ssieb, 2024-04-28:
Things happening through the proxy won't trigger anything in your config.

closed 2024-04-28 by ssieb

designerferro, 2024-04-28:
Thanks, @ssieb. Just to confirm there is no way that ble_client on_connect will be trigered since the bluetooth_proxy has no action (on_...), right?

ssieb, 2024-04-28:
The proxy has no actions.  It doesn't even look at the data.