esphome/issues #5746: compile error with ESP32-S3 using ESP-IDF with esp32_rmt_led_strip

gargamelonly, 2024-04-28:
### The problem

compile error with ESP32-S3 using ESP-IDF with platform: esp32_rmt_led_strip
configuration and log below.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3 dev

### Component causing the issue

esp32_rmt_led_strip

### Example YAML snippet

```yaml
esphome:
  name: voyager
  friendly_name: voyager
  platformio_options:
    board_build.flash_mode: dio
#    board_build.arduino.memory_type: qio_opi
#    board_upload.maximum_ram_size: 524288

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
#    type: arduino
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32_S3_BOX_BOARD: "y"
      
psram:
  mode: octal
  speed: 80MHz
  
# Enable logging
logger:
  level: DEBUG

light:
  - platform: esp32_rmt_led_strip
    name: LED_RGB
    id: led
    rgb_order: RGB
    pin: GPIO48
    num_leds: 1
    rmt_channel: 0
    chipset: WS2812
```

### Anything in the logs that might be useful for us?

```txt
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection13light_commandERKNS0_19LightCommandRequestE+0x8): undefined reference to `esphome::light::LightState::make_call()'
[... 71 lines cut ...]
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.literal._Z5setupv+0x11c): undefined reference to `esphome::light::LightState::add_effects(std::vector<esphome::light::LightEffect*, std::allocator<esphome::light::LightEffect*> > const&)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.literal._Z5setupv+0x120): undefined reference to `esphome::esp32_rmt_led_strip::ESP32RMTLEDStripLightOutput::set_led_params(unsigned int, unsigned int, unsigned int, unsigned int)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o: in function `setup()':
/config/.esphome/build/voyager/src/main.cpp:196: undefined reference to `esphome::light::LightState::LightState(esphome::light::LightOutput*)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /config/.esphome/build/voyager/src/main.cpp:225: undefined reference to `esphome::light::LightState::set_restore_mode(esphome::light::LightRestoreMode)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /config/.esphome/build/voyager/src/main.cpp:225: undefined reference to `esphome::light::LightState::set_default_transition_length(unsigned int)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o: in function `esphome::Component::Component()':
/config/.esphome/build/voyager/src/esphome/core/component.h:68: undefined reference to `esphome::light::LightState::set_flash_transition_length(unsigned int)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o: in function `esphome::psram::PsramComponent* esphome::Application::register_component<esphome::psram::PsramComponent>(esphome::psram::PsramComponent*)':
/config/.esphome/build/voyager/src/esphome/core/application.h:163: undefined reference to `esphome::light::LightState::set_gamma_correct(float)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o: in function `esphome::Color::Color(unsigned char, unsigned char, unsigned char, unsigned char)':
/config/.esphome/build/voyager/src/esphome/core/color.h:40: undefined reference to `esphome::light::LightState::add_effects(std::vector<esphome::light::LightEffect*, std::allocator<esphome::light::LightEffect*> > const&)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o: in function `setup()':
/config/.esphome/build/voyager/src/main.cpp:247: undefined reference to `esphome::esp32_rmt_led_strip::ESP32RMTLEDStripLightOutput::set_led_params(unsigned int, unsigned int, unsigned int, unsigned int)'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0xc): undefined reference to `esphome::light::LightState::setup()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x10): undefined reference to `esphome::light::LightState::dump_config()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x14): undefined reference to `esphome::light::LightState::loop()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x18): undefined reference to `esphome::light::LightState::get_setup_priority() const'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x24): undefined reference to `non-virtual thunk to esphome::light::LightState::setup()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x28): undefined reference to `non-virtual thunk to esphome::light::LightState::loop()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x2c): undefined reference to `non-virtual thunk to esphome::light::LightState::dump_config()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: .pioenvs/voyager/src/main.o:(.rodata._ZTVN7esphome5light21AddressableLightStateE[vtable for esphome::light::AddressableLightState]+0x30): undefined reference to `non-virtual thunk to esphome::light::LightState::get_setup_priority() const'
collect2: error: ld returned 1 exit status
*** [.pioenvs/voyager/firmware.elf] Error 1
```

gargamelonly, 2024-04-28:
Doing a clean build solved it.

closed 2024-04-28 by gargamelonly

carryonrewardless, 2024-07-15:
Clean build worked for me too with this error. Thanks for sharing.