esphome/issues #5747: issue with fastled_clocklesss using WS2812 colors

deanfourie1, 2024-04-28:
### The problem

using the fastled_clockless integration with the WS2812 LEDs,
When specifying colors such as

green: 100% 

does not show green, it shows a light blue instead.

Stating red: 100% shows purple.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.4

### What platform are you using?

ESP32

### Board

esp32minikit

### Component causing the issue

fastled_clockless

### Example YAML snippet

```yaml
on_boot:
    priority: -1000
    then: 
      - if:
          condition:
            - sun.is_above_horizon:
          then:
              light.turn_on:
                id: led
                brightness: 100%
                green: 100%

          else:
             - light.turn_on:
                id: led
                brightness: 20%
                green: 0%
                red: 100%

light:
  - platform: fastled_clockless
    id: "led"
    name: "LED"
    pin: 23
    chipset: WS2812
    num_leds: 1
    rgb_order: GRB
```

### Anything in the logs that might be useful for us?

```txt
Nothing
```

jesserockz, 2024-04-28:
Try using the newer [esp32_rmt_led_strip](https://esphome.io/components/light/esp32_rmt_led_strip) platform instead.

deanfourie1, 2024-04-29:
> Try using the newer [esp32_rmt_led_strip](https://esphome.io/components/light/esp32_rmt_led_strip) platform instead.

Thanks, but it seems I have the same issue with this integration.

Am I doing something wrong.

green: 100% should = green, not while....?

deanfourie1, 2024-04-29:
> Try using the newer [esp32_rmt_led_strip](https://esphome.io/components/light/esp32_rmt_led_strip) platform instead.

Thanks, that worked.

But I still have another issue,

My led is setting brightness based on the sun, 

- when the sun is below horizon, then brightness is 20%
- when sun is above horizon, then brightness is 100%

however, the issue I have is when ever I call a Effect such as Fast Pulse, it is just defaulting to 100% regardless.

Is there a better way to do this or is this possible to set the effect flash brightness based on the sun being above or below horizon?

Thanks