esphome/issues #5754: Using `framework: type: esp-idf` frequently disconnects from WiFi / HA unless using `wifi: power_save_mode: none` or `wifi: fast_connect: true`

ptr727, 2024-05-01:
### The problem

TL;DR:  
When using `framework: type: esp-idf` the ESP32 keeps disconnecting from HA unless using `wifi: fast_connect: true`.

I discovered this problem when I wanted to use my first BT proxy.  
I created a BT proxy using standard ESPHome devices BT proxy on a Screek BP1 ESP32.  
The device kept on disconnecting / reconnecting from WiFi and / or HA.

I created a minimal configuration, no BT proxy, and through experimentation I found that:

- Using `framework: type: arduino`, or not specifying `framework`, the device works fine.
- Using `framework: type: esp-idf` the ESP32 keeps disconnecting from HA.
- Using `framework: type: esp-idf` and `wifi: power_save_mode: none`, the device works fine.
  - But `wifi: power_save_mode: none` is [incompatible](https://github.com/esphome/esphome/pull/1950) with BT proxies.
- Using `framework: type: esp-idf` and `wifi: fast_connect: true`, the device works fine.
  - But `wifi: fast_connect: true` is undesirable as I have several AP's.

Note: I run Ubiquity UniFi network equipment, including several Unifi v6 AP's and lots of devices, including several ESPHome devices that use the default Arduino framework, without any issues.

I verified that using `framework: type: esp-idf` and `wifi: fast_connect: true` and `esp32_ble_tracker` and `bluetooth_proxy` works.

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.4.4

### What platform are you using?

ESP32-IDF

### Board

Screek Bluetooth Proxy BP1

### Component causing the issue

esp-idf

### Example YAML snippet

```yaml
substitutions:
  device_name: esp32-test
  device_comment: "ESP32 test Device"

esphome:
  name: ${device_name}
  comment: ${device_comment}

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  reboot_timeout: 5min
  # power_save_mode: none
  fast_connect: true

logger:
  level: debug

api:
  id: api_server
  reboot_timeout: 5min

ota:
  password: !secret ota_password

time:
  - platform: homeassistant
    id: homeassistant_time

debug:
  update_interval: 60s

# esp32_ble_tracker:
#   scan_parameters:
#     active: true

# bluetooth_proxy:
#   active: true

button:
  - platform: restart
    name: ${device_name}_restart

binary_sensor:
  - platform: status
    name: ${device_name}_status

text_sensor:
  - platform: version
    name: ${device_name}_version
  - platform: wifi_info
    ip_address:
      name: ${device_name}_wifi_ip
      id: wifi_ip
      update_interval: 60s
    ssid:
      name: ${device_name}_wifi_ssid
      update_interval: 60s
    bssid:
      name: ${device_name}_wifi_bssid
      update_interval: 60s
  - platform: debug
    device:
      name: ${device_name}_device_info
    reset_reason:
      name: ${device_name}_reset_reason

sensor:
  - platform: wifi_signal
    name: ${device_name}_wifi_signal
    update_interval: 60s
  - platform: uptime
    name: ${device_name}_uptime
    unit_of_measurement: min
    filters:
      - lambda: return x / 60.0;
  - platform: debug
    free:
      name: ${device_name}_heap_free
```

### Anything in the logs that might be useful for us?

```txt
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.041s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.082s
[20:04:01][D][api:102]: Accepted 192.168.1.26
[20:04:01][D][api.connection:1210]: Home Assistant 2024.4.4 (192.168.1.26): Connected successfully
[20:04:01][D][time:049]: Synchronized time: 2024-04-30 20:04:01
[20:04:25][D][sensor:094]: 'esp32-test_uptime': Sending state 0.78760 min with 0 decimals of accuracy
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.026s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.050s
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.052s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.054s
[20:05:42][D][api:102]: Accepted 192.168.1.26
[20:05:42][D][api.connection:1210]: Home Assistant 2024.4.4 (192.168.1.26): Connected successfully
[20:05:42][D][time:049]: Synchronized time: 2024-04-30 20:05:42
[20:06:14][D][sensor:094]: 'esp32-test_uptime': Sending state 0.86745 min with 0 decimals of accuracy
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.004s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.040s
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.042s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.053s
[20:07:26][D][api:102]: Accepted 192.168.1.26
[20:07:26][D][api.connection:1210]: Home Assistant 2024.4.4 (192.168.1.26): Connected successfully
[20:07:26][D][time:049]: Synchronized time: 2024-04-30 20:07:26
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.015s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.059s
[20:08:06][D][api:102]: Accepted 192.168.1.26
[20:08:06][D][api.connection:1210]: Home Assistant 2024.4.4 (192.168.1.26): Connected successfully
[20:08:06][D][time:049]: Synchronized time: 2024-04-30 20:08:06
WARNING esp32-test @ 192.168.1.181: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esp32-test @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to esp32-test @ 192.168.1.181 in 0.051s
INFO Successful handshake with esp32-test @ 192.168.1.181 in 0.056s
```

ptr727, 2024-05-09:
Observation:

- I tested some more ESP32 devices I have, and some older devkit devices that do work with normal WiFi settings and BT proxy.
- I wondered if my PSU (PC USB port, Anker desk charger, bench PSU) had an influence on newer devices (different power stage) when using BT proxy.
- I added a USB Super Capacitor Power Filter module inline, and the Screek BP1 device now works with normal WiFi, same with a ESP32-S3 Feather.
- I also tried a lipo power bank, but same, I added the supercap inline with the powerbank, and it worked.

I had ruled out PSU due to testing three different PSU's (never an issue before using BT proxy), but it does seem that the PSU noise has a greater influences when using BT and WiFi on some devices with possibly inadequate power noise regulation?

pfeerick, 2024-05-10:
I had been running the really early BT Proxy config... from when it was first announced... and updating to 2024.4.2 broke it... hadn't had an issue before with a ESP32-DEV board being powered off USB, hanging off a desk lamp. Tried swapping to the current config, and no change. 

However, it seems like maybe it it is starting to fail, or there has been some timing changing which is causing it to brownout when both bluetooth and WiFi are started... as I swapped it with another ESP32-DEV, and that one is working. Adding capacitors to either VCC/3V3 made no change, but disabling the bluetooth-proxy did restore it to functioning :shrug: Pretty sure I had 2024.4.1 on it before.

ptr727, 2024-05-10:
> However, it seems like maybe it it is starting to fail

Did you try `fast_connect: true` to see if it made a difference?

hugh-martin, 2024-09-29:
I have the same issue.  As soon as I added `fast_connect: true`, the `[Errno 104]` errors disappeared.

ESPHome Add-on version 2024.9.1
Installation Type: HAOS (on VirtualBox)
HA version 2024.9.3
Platform / Framework: esp-idf
Board: M5Stack Atom Lite

YAML:

```
esphome:
  name: bt-proxy-1
  friendly_name: BT Proxy 1
  min_version: 2024.6.0
  name_add_mac_suffix: true

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"
    id: ota_esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true # Added to resolve [Errno 104] errors

esp32_ble_tracker:
  scan_parameters:
    # active: true
    active: false # Changed to false since the MacBook BT radio only allowed passive and the Level Lock worked.
    continuous: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset
```

Log data before adding fast_connect: true:

```
2024-09-29 11:50:21.169 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to bt-proxy-1 @ 192.168.107.182 in 1.103s
2024-09-29 11:50:21.362 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with bt-proxy-1-0ac0c0 @ 192.168.107.182 in 0.193s
2024-09-29 12:04:43.357 WARNING (MainThread) [aioesphomeapi.connection] bt-proxy-1-0ac0c0 @ 192.168.107.182: Connection error occurred: [Errno 104] Connection reset by peer
2024-09-29 12:04:43.358 INFO (MainThread) [aioesphomeapi.reconnect_logic] Processing unexpected disconnect from ESPHome API for bt-proxy-1-0ac0c0 @ 192.168.107.182
2024-09-29 12:04:43.695 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for bt-proxy-1-0ac0c0 @ 192.168.107.182: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.107.182', port=6053))]: [Errno 111] Connect call failed ('192.168.107.182', 6053) (SocketAPIError)
2024-09-29 12:04:43.695 INFO (MainThread) [aioesphomeapi.reconnect_logic] Trying to connect to bt-proxy-1-0ac0c0 @ 192.168.107.182 in the background
2024-09-29 12:04:49.101 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to bt-proxy-1-0ac0c0 @ 192.168.107.182 in 0.360s
2024-09-29 12:04:49.225 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with bt-proxy-1 @ 192.168.107.182 in 0.124s
[... 21 lines cut ...]

Log data after adding `fast_connect: true`:

```
2024-09-29 15:20:57.120 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to bt-proxy-1-0ac0c0 @ 192.168.107.182 in 0.118s
2024-09-29 15:20:57.251 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with bt-proxy-1-0ac0c0 @ 192.168.107.182 in 0.130s
2024-09-29 15:20:57.551 INFO (MainThread) [homeassistant.setup] Setting up homekit_controller
2024-09-29 15:20:57.561 INFO (SyncWorker_7) [homeassistant.components.synology_srm.device_tracker] Synology SRM scanner initialized
2024-09-29 15:20:57.562 INFO (MainThread) [homeassistant.setup] Setup of domain homekit_controller took 0.01 seconds
2024-09-29 15:20:57.577 INFO (MainThread) [homeassistant.setup] Setting up ibeacon
2024-09-29 15:20:57.578 INFO (MainThread) [homeassistant.setup] Setup of domain ibeacon took 0.00 seconds
2024-09-29 15:20:57.598 INFO (MainThread) [homeassistant.components.button] Setting up homekit_controller.button
2024-09-29 15:20:57.600 INFO (MainThread) [homeassistant.components.sensor] Setting up homekit_controller.sensor
2024-09-29 15:20:57.601 INFO (MainThread) [homeassistant.setup] Setting up lock
2024-09-29 15:20:57.601 INFO (MainThread) [homeassistant.setup] Setup of domain lock took 0.00 seconds
2024-09-29 15:20:57.602 INFO (MainThread) [homeassistant.components.lock] Setting up homekit_controller.lock
2024-09-29 15:20:57.614 INFO (MainThread) [homeassistant.components.device_tracker] Setting up ibeacon.device_tracker
2024-09-29 15:20:57.614 INFO (MainThread) [homeassistant.components.sensor] Setting up ibeacon.sensor
2024-09-29 15:20:57.797 INFO (MainThread) [homeassistant.setup] Setting up automation
2024-09-29 15:20:57.816 INFO (MainThread) [homeassistant.setup] Setup of domain automation took 0.02 seconds
2024-09-29 15:20:58.610 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up smartthings.binary_sensor
...
2024-09-29 18:06:27.736 ERROR (MainThread) [homeassistant.components.hydrawise] Timeout fetching hydrawise data
2024-09-29 18:07:00.624 INFO (MainThread) [homeassistant.components.hydrawise] Fetching hydrawise data recovered
```