esphome/issues #5755: HMC5883L Not fast updating

mkmer, 2024-05-02:
### The problem

Same issue was discovered here:
https://github.com/esphome/esphome/pull/6647#event-12676515088
Need to add similar code to allow for fast updates with this driver as well.

### Which version of ESPHome has the issue?

 2024.4.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-WROOM

### Component causing the issue

HMC5883L

### Additional information

Really just need to add the changes implemented for the QMC5883L ->   https://github.com/esphome/esphome/pull/6647#event-12676515088

tronikos, 2024-05-04:
Thanks for filing this. As I don't have an HMC5883L to test I was hesitant applying my change on this component too in case I break anything. Do you have an HMC5883L or QMC5883L?

mkmer, 2024-05-04:
I have one of the QMC's that is actually an HMC (as well as a "real" QMC) - this change does appear to give similar performance improvements and works.

closed 2024-05-05 by jesserockz