esphome/issues #5756: ESPHome device coming out of sleep mode exceedingly long connect time with good signal and does not update HomeAssistant sensors

JustEnoughDucks, 2024-05-02:
### The problem

I am making a dumb doorbell. The only requirements is that it 

- sends a sensor value change to Home Assistant when it has been pressed and it will also send its battery level before going to sleep
- It doesn't fails to update the sensor value (within a good wifi signal range) before going to sleep

The problem is that upon installing  the config to the ESP32-C3, the device will consistently work _until it goes to sleep for more than 10 minutes_, then it will be extremely inconsistent and never work completely correctly again.

The behavior after the 10 minute period is as follows:

- The device takes 20-90 seconds to connect. I think that is very far out of the reasonable time frame for a -41dB signal
- After connection, the device says that it has successfully sent the sensor values, but Home Assistant does not receive any sensor values
- ***If I click on "logs" in the ESPHome dashboard at any point when it is trying to connect, it will instantaneously connect and claim that it has sent the values (while not actually sending the values to HA). If I stay on this screen of the logs while waking up the device again, it will consistently work again until i get the following message:

`WARNING Can't connect to ESPHome API for dumb-doorbell @ 192.168.1.124: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.124', port=6053))] (TimeoutAPIError)`

*** I think this is an important because it indicates that it is a problem on the ESPhome end and not the device config. If ESPHome listens for a connection (or whatever it is doing when the log screen is up) then it connects right away. Then after the error message, it still connects, but ESPHome is the one that doesn't seem to relay the sensor values to Home Assistant

Maybe it would work better to simply broadcast an MQTT message instead of trying to couple with the API if the API doesn't seem to like handling very infrequent connections? It shouldn't be an networking issue because my other ESPHome devices that don't drop off the network have absolutely no problem sending simple sensor values and I have a DSMR reader that hands over values every 10 seconds.

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

<=2024.5.0

### What platform are you using?

ESP32

### Board

SEEED Xiao ESP32-C3

### Component causing the issue

deep sleep or esphome api itself

### Example YAML snippet

```yaml
esphome:
  name: dumb-doorbell
  friendly_name: Dumb Doorbell
  # on_boot documentation and priorities: https://esphome.io/components/esphome.html#on-boot
  on_boot:
    - priority: -100
      then: # At this priority, pretty much everything should already be initialized.
        - light.turn_on:
            id: doorbell_light
            brightness: 100%
            transition_length: 100ms
        - deep_sleep.prevent: sleep_routine
        - wait_until: api.connected
        
        - logger.log: 
            format: "Starting sensor updates..."
            level: INFO
        - light.turn_off:
            id: doorbell_light
            transition_length: 2s
        - output.turn_on: meas_gnd
        - repeat:
            count: 1
            then:
              - binary_sensor.template.publish:
                  id: doorbell_pressed
                  state: ON
              #Sensor updates are turned off and manually requested on boot. 
              #Then the ESP goes back to sleep when they're done (unless told to stay awake)

              # Request sensor updates
              - component.update: LIION
              - delay: 100ms
              - component.update: bat_v
              - component.update: bat_percent
        - output.turn_off: meas_gnd
        - deep_sleep.allow: sleep_routine
        - delay: 2s
        - binary_sensor.template.publish:
            id: doorbell_pressed
            state: OFF
        - if:
            condition:
              - binary_sensor.is_on: battery_updated
            then:
              - logger.log: 
                  level: INFO
                  format: "Data from battery received"

            else:
              - logger.log: 
                  format: "Battery update missing"
                  level: ERROR

        # Reset data update sensors states
        - binary_sensor.template.publish:
            id: battery_updated
            state: OFF
        - deep_sleep.enter: sleep_routine
      

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
    level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: <REDACTED>

ota:
  password: "espUpdate32"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.124
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  domain: ".web"
  fast_connect: true
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dumb-Doorbell Fallback Hotspot"
    password: "eiURjgTxvkLk"

captive_portal:

# Sensing routine Setup
binary_sensor:
  - platform: template
    name: "Doorbell Pressed"
    id: doorbell_pressed
  - platform: template
    name: "Battery Read"
    id: battery_updated
    internal: true
        

# Battery read ADC
sensor:
  - platform: adc
    pin: GPIO4
    name: "Doorbell A2 Voltage"
    id: LIION
    update_interval: never
    attenuation: 2.5db
    internal: true
  - platform: template
    name: "Doorbell Battery Voltage"
    id: bat_v
    unit_of_measurement: 'V'
    update_interval: never
    accuracy_decimals: 2
    icon: "mdi:battery"
    lambda: |-
      return (id(LIION).state * 4);
  - platform: template
    name: "Doorbell Battery Percentage"
    id: bat_percent
    unit_of_measurement: '%'
    update_interval: never
    accuracy_decimals: 0
    device_class: battery
    lambda: |-
      if(id(bat_v).state < 1)
      {return 0;}
      return ((id(bat_v).state-3) /1.2 * 100.00);
    on_value:
      then:
        - binary_sensor.template.publish:
            id: battery_updated
            state: ON

output:
  - platform: ledc
    pin: GPIO7
    id: led_out
    frequency: "1000Hz"
# Measurement GND active low to save leakage power
  - platform: gpio
    pin: GPIO5
    id: meas_gnd
    inverted: True
# LED setup
light:
  - platform: monochromatic
    output: led_out
    name: ${friendly_name} LED
    id: doorbell_light
    internal: true
    default_transition_length: 1s

# Deep sleep configuration
deep_sleep:
  #run_duration: 4s
  #sleep_duration: 3s
  wakeup_pin: 
    number: GPIO2
    mode: 
      input: True
      open_drain: False
      pulldown: True
    inverted: False
  id: sleep_routine
```

### Anything in the logs that might be useful for us?

```txt
: Running through setup()...<0x1b>[0m

<0x1b>[0;32m[I][wifi:303]: WiFi Connecting to <REDACTED>...<0x1b>[0m

<0x1b>[0;33m[W][component:157]: Component wifi set Warning flag: unspecified<0x1b>[0m
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038b254
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000
<0x1b>[0;32m[I][logger:158]: Log initialized<0x1b>[0m
<0x1b>[0;32m[I][app:029]: Running through setup()...<0x1b>[0m
<0x1b>[0;32m[I][wifi:303]: WiFi Connecting to <REDACTED>...<0x1b>[0m
<0x1b>[0;33m[W][component:157]: Component wifi set Warning flag: unspecified<0x1b>[0m
**1s delay**
<0x1b>[0;32m[I][wifi:597]: WiFi Connected!<0x1b>[0m
<0x1b>[0;32m[I][app:062]: setup() finished successfully!<0x1b>[0m
<0x1b>[0;33m[W][component:170]: Component wifi cleared Warning flag<0x1b>[0m
<0x1b>[0;33m[W][component:157]: Component api set Warning flag: unspecified<0x1b>[0m
<0x1b>[0;32m[I][app:100]: ESPHome version 2024.4.2 compiled on May  2 2024, 14:12:20<0x1b>[0m
**59 second delay**
<0x1b>[0;33m[W][component:170]: Component api cleared Warning flag<0x1b>[0m
<0x1b>[0;32m[I][main:259]: Starting sensor updates...<0x1b>[0m
<0x1b>[0;32m[I][main:290]: Data from battery received<0x1b>[0m
<0x1b>[0;32m[I][deep_sleep:117]: Beginning Deep Sleep<0x1b>[0m
```

ssieb, 2024-05-02:
The device does not do the connecting.  HA connects to the device.  If it doesn't know that the device is sleeping, it will keep attempting to connect with increasing back-off.  The dashboard logs use the api as well, so the device doesn't know the difference between that and HA connecting, which is why it immediately thinks it has sent the data when you connected for logs.  Is MDNS working?  Did you add the device before or after you included the deep sleep option?  There are several things you could improve with the config and you didn't include enough logs.  Come ask on discord.

JustEnoughDucks, 2024-06-17:
I can't use mDNS, I use ping instead because my ISP given router doesn't support mDNS. I use ping instead, defined as an environment variable in the compose file. I added the device after the deep sleep option because it is connected long enough to input the address and encryption key before it drops off. It sounds to me that this is an intentional esphome design decision and it is not possible to modify the ping interval.

I tried switching over to MQTT instead because it is like [300x faster to connect after getting wifi connection after deep sleep](https://community.home-assistant.io/t/esphome-api-vs-mqtt-for-deep-sleep-in-2022/440934/2), but that opens up its own can of worms of MQTT working everywhere on my entire docker stack as well as my whole home network except in esphome. I have asked in discord, so maybe I can get it sorted out there?

ssieb, 2024-06-24:
Why would you need to modify the ping interval?  Are you confusing the esphome dashboard with the HA integration?
I see you did ask on discord which is the right place for this.

closed 2024-06-24 by ssieb

JustEnoughDucks, 2024-06-24:
Maybe I am just misunderstanding the esphome and HA architecture, but you said HA pings the device with increasing backoff. If that behavior was not desired and more consistent pings are desired (like in this case), then I would need to modify the ping interval.

In this case, MQTT works better than the direct esphome API and HA integration. It connects withing 5 seconds every time and never has a connection issue like the esphome API.

ssieb, 2024-06-25:
If HA doesn't know that the device is sleeping, then HA tries to connect to the device with an increasing backoff.
If MDNS doesn't work on your network, then you will need to use mqtt.