esphome/issues #5757: Sonoff R4 Basic unable to update

techwithjake, 2024-05-02:
### The problem

I'm attempting to update my Sonoff R4 Basics using the yaml code from [https://devices.esphome.io/devices/Sonoff-BASIC-R4-v1.0](https://devices.esphome.io/devices/Sonoff-BASIC-R4-v1.0). I was able to upload and update them until ESPHome v2024.4.1. Since then I get:

```
INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/esphome-web-XXXXX.yaml...
Failed config

esp32: [source /config/esphome/esphome-web-XXXXX.yaml:9]
  
  Option 'variant' does not match selected board.
  variant: ESP32C3
  board: esp32dev
  framework: 
    type: esp-idf
    version: recommended
    sdkconfig_options: 
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: 10
``` 

### Which version of ESPHome has the issue?

2024.4.1+

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.4.2+

### What platform are you using?

ESP32-IDF

### Board

Sonoff R4 Basic

### Component causing the issue

Board Variant

### Example YAML snippet

```yaml
esp32:
  variant: ESP32C3
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
```

gromar88, 2024-05-03:
Same problem using HA 2024.5.X - I can't do update to esphome 2024.4.2

ssieb, 2024-05-03:
The example config is wrong and shouldn't have done that.  You need to pick a valid C3 board like `esp32-c3-devkitm-1`.

closed 2024-05-03 by ssieb

techwithjake, 2024-05-04:
@ssieb 
[https://esphome.io/components/esp32](https://esphome.io/components/esp32) needs to be updated then.

"Configuration variables:
- board (Required, string): The PlatformIO board ID that should be used. Choose the appropriate board from [this list](https://registry.platformio.org/packages/platforms/platformio/espressif32/boards) (the icon next to the name can be used to copy the board ID). This only affects pin aliases, flash size and some internal settings, if unsure choose a generic board from Espressif such as **`esp32dev`**."

ssieb, 2024-05-04:
> https://esphome.io/components/esp32 needs to be updated then.

I suppose it could be more clear, although it's technically still correct if you read it correctly.  I also gave you a "generic board from Espressif".  You could make a PR to make it better if you want.

> I received this config from ESPHome directly as well. https://devices.esphome.io/devices/Sonoff-BASIC-R4-v1.0

That's not from esphome.  Those configs are user supplied.  You could also make a PR to correct that.

sikaiser, 2024-05-06:
@techwithjake The following configuration works and replicates the same functionality as the example from https://devices.esphome.io/devices/Sonoff-BASIC-R4-v1.0 (except that it uses a `switch` instead of a `light`).

I'll try to make a pull request to update https://devices.esphome.io/devices/Sonoff-BASIC-R4-v1.0 if I get around to it.

You can first set up a bare bones install by using the intial setup wizard in the ESPhome dashboard. Then all you need to do is add the lines starting from `#magic switch mode`.

```
esphome:
  name: basic-r4
  friendly_name: basic-r4

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[ADD YOUR KEY]"

ota:
  password: "[ADD YOUR PASSWORD]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Basic-R4"
    password: "kh0CR8fSB6SF"

captive_portal:
    

#magic switch mode
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [ magic_switch ]

magic_switch:
  pin: 5
  on_switch:
    - switch.toggle: switch_1

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - switch.toggle: switch_1

output:
  - platform: gpio
    pin: GPIO4
    id: relay_1

switch:
  - platform: output
    name: Switch
    id: switch_1
    output: relay_1

status_led:
  pin:
    number: GPIO6
    inverted: yes
```

milandzuris, 2024-05-07:
For me esp-idf not work only arduino

ssieb, 2024-05-07:
With esp-idf, you will probably have to set the flash mode to `dio`.

milandzuris, 2024-05-08:
> With esp-idf, you will probably have to set the flash mode to `dio`.

dio???, first i flash arduino and after over ota flash idf