esphome/issues #5759: AP not RE-starting after initial start and then STA-wifi temporarily available

Hiekkaharju, 2024-05-03:
### The problem

When both WIFI SSID aka STA-wifi and AP are defined, the sequence: 
-reset the device when STA-wifi is not available - AP will start after ap_timeout
-make STA wifi available - device connects to STA-wifi and stops AP
-make STA wifi unavailable - device starts to scan for wifi. 
AP is not re-started even after much longer time than ap_timeout.

Since I've not noticed this earlier, I'd suspect that this might be a regression due to recent fast connect related changes, but I'm not able to verify that.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

esp32dev (ATOM LITE)

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esphome:
  name: test2
  friendly_name: test2

esp32:
  board: esp32dev
  framework:
    type: arduino
 
# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    api.service: INFO
    api: INFO

# Enable Home Assistant API
api:

ota:

web_server:
   local: true
   
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  ap:
    ssid: "Test2 Fallback Hotspot"
```

### Additional information

log :
[20240503 wo pr test2_logs redact.txt](https://github.com/esphome/issues/files/15198627/20240503.wo.pr.test2_logs.redact.txt)

Hiekkaharju, 2025-04-06:
I consider it to be a bug.

btw: since the device where this issue is significant to me does not do anything that would be really harmed by the side effects of my workaround, I use

```
button:
  - platform: restart
    id: restart_button

wifi:
  on_disconnect:
        - button.press: restart_button
````

nsty, 2025-04-10:
Thanks for your workaround. As of now, the connection is not too critical for me, only for informational purposes. So I just use the default not-connected timeout of 15 minutes to restart the device as a workaround.