esphome/issues #5762: Component htu21d.sensor set Warning flag: unspecified

Aculeasis, 2024-05-04:
### The problem

After updating firmware from 2024.2.2 to 2024.4.2, I started receiving these warnings from all my HTU21 sensors. The sensors work fine.

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

ESP8266, ESP32

### Component causing the issue

htu21d

### Example YAML snippet

```yaml
i2c:
  sda: 17
  scl: 16

sensor:
  - platform: htu21d
    temperature:
      name: "${friendly_name} Temperature"
    humidity:
      name: "${friendly_name} Humidity"
```

### Anything in the logs that might be useful for us?

```txt
[10:14:01][D][htu21d:058]: Got Temperature=39.1°C
[10:14:01][D][sensor:094]: 'BLE Gateway 3 Temperature': Sending state 39.10094 °C with 1 decimals of accuracy
[10:14:01][W][component:157]: Component htu21d.sensor set Warning flag: unspecified
[10:14:01][D][htu21d:081]: Got Humidity=5.2% Heater Level=0
[10:14:01][D][sensor:094]: 'BLE Gateway 3 Humidity': Sending state 5.24573 % with 1 decimals of accuracy
[10:14:01][W][component:170]: Component htu21d.sensor cleared Warning flag
```

sbrk-r, 2024-05-04:
There's been changes to add heater support for relatives it looks like, and added submodels. Looks like it should default to real htu21d with no heater level support.

All instances of setting warnings seem to be I2C read or write that fail (could be no ACK, timeout, etc).

https://github.com/esphome/esphome/pull/6511
@MRemy2 any ideas?

MRemy2, 2024-05-07:
> There's been changes to add heater support for relatives it looks like, and added submodels. Looks like it should default to real htu21d with no heater level support.
> 
> All instances of setting warnings seem to be I2C read or write that fail (could be no ACK, timeout, etc).
> 
> [esphome/esphome#6511](https://github.com/esphome/esphome/pull/6511) @MRemy2 any ideas?

I do have an ideea :)
We should wait for the commit to make it to the release version with the models of the sensors.
I have updated also to 2024.4.2 and see the messages.

As a quick fix you could add external components  in the yaml file and include the dev branch for the htu21d sensor:

external_components:
  - source: github://esphome/esphome@dev
    #refresh: 1min
    components: [ htu21d ]

- platform: htu21d
    model: htu21d
    update_interval: 5s
    temperature:
      name: "Temperature"

closed 2024-05-15 by Aculeasis