esphome/issues #5764: Dependencies not found despite installed on esp-idf framework

SodaWithoutSparkles, 2024-05-04:
### The problem

I was trying to use the IRremoteESP8266 library to implement control of a daikin air conditioner. Code of which can be found [here](https://github.com/SodaWithoutSparkles/esphome-daikin64). The issue is that, when using the esp-idf framework, esphome reports missing dependency, despite installing it.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Does not matter

### What platform are you using?

ESP32

### Board

Compile time issue, not even installed to hardware

### Component causing the issue

library manager(?)

### Example YAML snippet

```yaml
esphome:
  name: test-esp32
  friendly_name: test-esp32
  includes: 
    # Find this file at the linked repo
    - include/irdaikin.h
  libraries:
    - crankyoldgit/IRremoteESP8266@^2.8.6

esp32:
  board: esp32dev
  framework:
    # Change this to arduino and it would be fine
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Esp32 Fallback Hotspot"
    password: "3lengvbPyZj5"

time:
  - platform: homeassistant
    timezone: Asia/Hong_Kong
    id: esptime
[... 133 lines cut ...]
In file included from src/main.cpp:26:
src/irdaikin.h:2:10: fatal error: IRremoteESP8266.h: No such file or directory

*************************************************************************
* Looking for IRremoteESP8266.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:IRremoteESP8266.h"
* Web  > https://registry.platformio.org/search?q=header:IRremoteESP8266.h
*
*************************************************************************

 #include "IRremoteESP8266.h"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-esp32/src/main.o] Error 1
Compiling .pioenvs/test-esp32/app_trace/app_trace.o
========================= [FAILED] Took 50.62 seconds =========================
```

### Additional information

- https://github.com/SodaWithoutSparkles/esphome-daikin64
- https://github.com/esphome/feature-requests/issues/2670
- https://github.com/esphome/feature-requests/issues/1054#issuecomment-765096913
- https://github.com/esphome/feature-requests/issues/444#issuecomment-548166019

Gadgit83, 2024-07-13:
I get similar issues when trying to include external libraries, did you find a solution?

Gadgit83, 2024-07-13:
You may need a 'platformio.ini' file alongside your .yaml file in the ESPHome config folder with content something like this:
`
[env:test-esp32]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  crankyoldgit/IRremoteESP8266@^2.8.6

SodaWithoutSparkles, 2024-07-14:
@Gadgit83 but I am not using the arduino framework?