esphome/issues #5767: Cannot use ESPHome Web to flash Wemos ESP32-S2 Mini

poldim, 2024-05-05:
### The problem

When trying to flash the module, chrome recognizes the device, but the web flasher is unable to flash it

Chrome recognizes the module:
[image]

Selecting the file to install:
[image]

Error:
[image]

### Which version of ESPHome has the issue?

2024.5.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.5.1

### What platform are you using?

ESP32

### Board

Wemos ESP32-S2 Mini

### Additional information

Module schematic: https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf

The module has no serial chip and the USB is connected to pins GPIO19 and GPIO20:
[image]
The [ESP32-S2FH4 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf) calls out the following UART pins: 
[image]

Environment:  Macbook accessing web flasher via Chrome with module connected via USB-C

I can succesfully flash the module using the following: https://khaz.me/flashing-an-esp32-s2-mini-on-a-mac/

zeroflow, 2024-11-03:
I can confirm that the same behavior is true for Win10.

Both the integrated flasher and web.esphome.io show the same behavior when trying to flash an ESP32-S2.
Looking at the web serial port, it uploads the stub and then times out.

[image]

This issue is also known in esptool-js: https://github.com/espressif/esptool-js/issues/38
It does not sound like this will be fixed soon: 

https://github.com/espressif/esptool-js/issues/38#issuecomment-1637169053
> The main reason why this type of issues doesn't get resolved quickly is that they require someone with deep knowledge of both ESP boot and with web development skills (JS/TS) to contribute. Since these two topics are pretty far apart, people with both of these skills are hard to come by.

poldim, 2024-11-18:
@balloob I remember in one of your videos you mentioned you pulled together the right people to create the esphome web flasher.  Any chance any of them might know how to troubleshoot or solve this issue?