esphome/issues #5768: 'invalid escape sequence' in elf2bin.py

Masterz69, 2024-05-05:
### The problem

Getting followed on compile:

```
Building .pioenvs\zzz-d1mini\firmware.bin
C:\Users\IgMi\.platformio\packages\framework-arduinoespressif8266\tools\elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
C:\Users\IgMi\.platformio\packages\framework-arduinoespressif8266\tools\elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
esp8266_copy_factory_bin([".pioenvs\zzz-d1mini\firmware.bin"], [".pioenvs\zzz-d1mini\firmware.elf"])
```

So far it happens on first compilation of device YAML regardless on used components etc.
On subsequent compilations do not see it anymore.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

DeanCording, 2024-06-25:
Fixed in https://github.com/esp8266/Arduino/pull/9034

Floriszz, 2024-07-06:
Sorry for the 'noop' question. I try to understand how this works. Will the fix in esp8266 be automatically included in the Python Esphome package? So in order to get the fix run: pip esphome --upgrade when the new esphome is released(2024.6.7 ?) ?

ssieb, 2024-07-06:
Probably not.  esphome doesn't use the latest Arduino release and that patch might not even be in a release yet.

garry0garry, 2024-09-18:
ESPHome 2024.9.0, ESP8266
```
.platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)

```

amitkeret, 2024-10-29:
Can confirm persisting issue.

ESPHome 2024.10.2
pip installation
HA 2024.10.3
platform: ESP8266

```
Building ....
.platformio\packages\framework-arduinoespressif8266\tools\elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
.platformio\packages\framework-arduinoespressif8266\tools\elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
Creating BIN file ...
```

rjeffman, 2024-11-18:
This still an open issue. Will this ever be fixed?

hum4nizer, 2024-12-01:
I get the same error when compiling: .arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s' words = re.split('\s+', line)

Martin-d45, 2024-12-14:
it seem there i a code mistake in  the python script elf2bin.py.  
words = re.split('\s+', line)   Fail's  I think The IDE parses  it twice  and then the \ gets enterpreted.

my solution was:
open elf2bin.py in a text editor, and insert  an extra \ in  line 54 and 73:

before : words = re.split('\s+', line)
After:     words = re.split('\\s+', line) 

worked for me :-)

Floriszz, 2024-12-14:
Maybe because of github comments formatting, but both lines(before and after) are the same😀

rjeffman, 2024-12-15:
@Martin-d45 a better fix is to prepend the string with an `r` (raw string) as in `r'\s+'`, makes intent more clear.

Martin-d45, 2024-12-16:
yeah possibly,  but i stopped testing, when it was working.

edelmoral, 2025-01-14:
> @Martin-d45 a better fix is to prepend the string with an `r` (raw string) as in `r'\s+'`, makes intent more clear.

this worked for me

tgxn, 2025-01-29:
> This still an open issue. Will this ever be fixed?

> Fixed in [esp8266/Arduino#9034](https://github.com/esp8266/Arduino/pull/9034)

it _was_ fixed, back in 2023, https://github.com/esp8266/Arduino/pull/9034

It looks like it's because they haven't published any releases since 2023, with the latest `3.1.2`:
https://github.com/esp8266/Arduino/releases

There's some discussion here about the issue:  
https://github.com/esp8266/Arduino/issues/9161

madurapa, 2025-02-20:
You can use `dev` in the version to fix this warning (note this may break at any time)

```
esp8266:
  board: nodemcuv2
  framework:
    version: dev
```
More information on https://esphome.io/components/esp8266.html

Phil353556, 2025-05-01:
@Martin-d45 
Hello,
Thank you for the fix on both lines,  it works perfectly.
using : 
 Arduino 
 Version: 2.3.6
 Date: 2025-04-09T11:24:04.835Z
 CLI Version: 1.2.0
 On ubuntu 24.04.2 LTS (Noble Numbat)
Kind regards
Phil

BartGysens, 2025-05-22:
Issue remounted on ESPHome update : 2025.5.0

_/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)_

DerEmder, 2025-05-22:
Yep, same here. The whole espHome stuff became a real mess (especially regarding LD2410 and Neopixel) and it gets worse with every update.  Updates seem to be untested.

ssieb, 2025-05-24:
It's a warning, it's not something esphome can do anything about, and it won't cause any issues.

DerEmder, 2025-05-24:
use

`
framework: dev
`

then the warning is gone. Unfortunately the dev does not work well here. Using addressable light component OR partition/segment scheme crashes my ESP8266 - logging and OTA works but the API does not. Using the old arduino framework again gives the warning (just ignore it) and the addressable light is working again (but not the light partition/segment scheme).

It is all very frustrating - I assume there is a HUGE underlying problem in the espHome core. Atm I am fed up with this situation and I highly think about using my own software (MQTT) again instead of the mess that is espHome at the moment.

ssieb, 2025-05-24:
Again, the warning is not esphome's problem and any issues caused by changing the framework are your problem because that's explicitly not supported.  And no one else thinks it's a mess.  If you have problems, then come ask for help on discord.

DerEmder, 2025-05-25:
Discord? Why? We are here, aren't we. This is the place for issues and help, isn't it?

cvwillegen, 2025-05-30:
Here's a one-liner that you can run on your HomeAssistant command line to fix this warning:

``
sudo docker exec -it `sudo docker ps -qf "name=esphome"` sed -i "s/'\\\\s/r'\\\\s/" /data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py
``

Hope this helps!

umtauscher, 2025-06-03:
@ssieb 
> Again, the warning is not esphome's problem and any issues caused by changing the framework are your problem because that's explicitly not supported. And no one else thinks it's a mess. If you have problems, then come ask for help on discord.

Hmm please enlighten me. Perhaps I am too ignorant to understand:
Why is this not the problem of ESPHome Builder?
ESPHome manages all the libraries and obviously a part of a library throws a warning.  A warning is a warning and means the syntax is not correct and the compiler guessed how to interpret it. At least this is my understanding.
How can the user fix this? 
I have no means to correct the library myself.

Please explain

ssieb, 2025-06-03:
You've been given instructions for fixing it locally.  But you need to use the advanced ssh addon configured to get into the host.

umtauscher, 2025-06-03:
Thanks, I wasn‘t aware of a second ssh addon.

ziomimmo77, 2025-06-18:
RAM:   [====      ]  43.5% (used 35616 bytes from 81920 bytes)
Flash: [=====     ]  52.5% (used 547873 bytes from 1044464 bytes)
Building .pioenvs/multisensor/firmware.bin
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
esp8266_copy_factory_bin([".pioenvs/multisensor/firmware.bin"], [".pioenvs/multisensor/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/multisensor/firmware.bin"], [".pioenvs/multisensor/firmware.elf"])

Today nobody fixed the problem.

ssieb, 2025-06-18:
@ziomimmo77 how about reading the other comments before adding another unhelpful one?

lakusz1github, 2025-06-28:
the same for me:

/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)

ziomimmo77, 2025-06-28:
> the same for me:
>
> /data/cache/platformio/packages/framework-arduinoespressif8266/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
>   words = re.split('\s+', line) [...]

[image]

Fixed with that!

olealm, 2025-06-29:
> > And no one else thinks it's a mess. If you have problems, then come ask for help on discord.
> 
> No. I am here. I am not on Discord and never will be. And yes, ESPhome is a big mess atm. I am not the only one, believe me. I am siting here after the 2025.5.0/1 update trying to re-flash my controllers OTA ... most times they time out and are in a boot loop. I only added partition/segment or addressable_set to my light functions - my ESP8266's go into boot loops after that. All(!) of them worked before the update. The same witjh my esp32cam's .. there are A LOT of issues posted here because of the update.

I have the same ESPHome problems. All my 101 ESPHome adapters (mix of 8266 and Esp32) worked just fine, including OTA. All were running encrypted communication. Since approx. 2025.x editions it has not even been lottery doing any changes via OTA. None of the once I did connected after the OTA. For the few I have deared to try and update, and after boot-loop and not being able to connect to the unit, I ended up re-flashing from scratch connected to a computer. For one 8266 not even that did work (tried the usual, deleting the integration for that unit, copying the .yaml, then deleting it from ESPHome GUI, restarting, re-adding the .yaml to \ESPHome\ etc.), and I ended up trying to flash it with "encryption:" disabled in the api: then re-adding everything to ESPHome and Home Assistant integration. Low and behold this time it was found. Being able to connect after disabling encryption could be a coincident I guess, who knows. Last year even mass-OTA'ing was just a matter of doing. Yes, ESPHome is a mess atm.

olealm, 2025-06-30:
> I have the same ESPHome problems. All my 101 ESPHome adapters (mix of 8266 and Esp32) worked just fine, including OTA. All were running encrypted communication. Since approx. 2025.x editions it has [...]

For completeness, let me just state that it looks like most of my problems are related to a display component. The boot loops are accompanied with the good ole "OOM".. Was never a problem earlier, but it looks like later versions of the ESPHome api (or other sensor components?) need more RAM. If I remove the display component it connects and work fine. The displays were frankly no more than nice-to-have so not a problem really. I'll replace the units with ESP32's if the OLED display is actually needed.