esphome/issues #5769: [cst816][pca9554] is an invalid option for [interrupt_pin]

lboue, 2024-05-06:
### The problem

Hello.

I'm looking for the right ESPHome syntax to configure a touchscreen connected via a PCA9554 I/O expander. This is the YAML config:
```
touchscreen:
  platform: cst816
  id: cst816_touchscreen
  interrupt_pin:
      pca9554: pca9554a_device
      number: 3
```
And I am getting this error: ```[pca9554] is an invalid option for [interrupt_pin]. Please check the indentation.```

I didn't find it here:
https://esphome.io/components/touchscreen/index.html#touchscreen-components

Regards

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.1

### What platform are you using?

ESP32-IDF

### Board

Adafruit Qualia ESP32-S3

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
# PCA9554 I/O Expander
# https://esphome.io/components/pca9554.html
pca9554:
  - id: pca9554a_device
    address: 0x3F

# 2.1" 480x480 round display use CST826 touchscreen with I2C Address at 0x15
touchscreen:
  platform: cst816
  id: cst816_touchscreen
  interrupt_pin:
      pca9554: pca9554a_device
      number: 3
```

randybb, 2024-05-06:
Interrupts via expanders are not supported.
Expanders have their own interrupt pin, but since it is not connected and not supported in esphome it cannot detect interrupt so this cannot work.

closed 2024-05-06 by randybb

lboue, 2024-05-06:
@randybb 
Technically, it works with this board and this screen. In fact, it works when programmed with Arduino or CircuitPython. This is a limitation of ESPHome and not of the PCA9554 chip.

randybb, 2024-05-06:
If your program is polling GPIOs constantly, then yes, but once you start using interrupts, then such approach is not going to work.

ssieb, 2024-05-06:
There's no benefit to using the interrupt pin on an expander.  If you have to query the expander to get the pin state, then you might as well just query the touch screen directly.