esphome/issues #5771: mitsubishi Climate do not return current Temp

sosso-b, 2024-05-07:
### The problem

Since the last update Current version: 2024.4.2
mitsubishi Climate return always 0 for the current temperature when i chance to daikin then i get directly back the current temperature.
Since that never add any issue was working nice for more then 3 years 

### Which version of ESPHome has the issue?

Current version: 2024.4.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
remote_transmitter:
  pin: D2
  carrier_duty_percent: 50%

climate:
  - platform: mitsubishi       # adjust to match your AC unit!
    name: "Clim etage"
    sensor: temp_etage

# Creation du sensor

sensor:
  - platform: homeassistant
    name: "Current temperature"
    entity_id: sensor.smoke_sensor_2
    id: temp_etage
    internal: true
    accuracy_decimals : 1
```

killer0071234, 2024-05-11:
I got the same issue, but the temperature at the climate object in HA is always 0°C.
The sensor object itself shows the correct temperature.
Esphome version: 2024.4.2

my yaml snippet:
```
  - platform: dallas
    index: 0
    name: "Temperatur"
    id: klima_temperatur
    filters:
      - sliding_window_moving_average:
          window_size: 15
          send_every: 15

remote_receiver:
  id: rcvr
  pin:
    number: D6
    inverted: True
    mode: INPUT_PULLUP
  tolerance: 25%

remote_transmitter:
  pin: D5
  carrier_duty_percent: 50%

climate:
  - platform: mitsubishi
    name: "Vorraum"
    receiver_id: rcvr
    id: klima
    icon: mdi:air-conditioner
    sensor: klima_temperatur
    visual:
      min_temperature: 18
      max_temperature: 26
      temperature_step: 0.5
```
I am not sure when it started to show just 0 value, because I did not use it during the winter months.

BR Daniel

bossoq, 2024-05-18:
Add this line to `esphome/components/mitsubishi/mitsubishi.cpp` after line 59

`traits.set_supports_current_temperature(true);`

[Add support current temperature to mitsubishi climate ir](https://github.com/bossoq/esphome/commit/8f666bfdef9814f78c883188cd1444c0ad5045fe)

sosso-b, 2024-05-18:
Hello bossoq,

Thank you for your help, but can you give me how and where i can access to mitsubishi.cpp.
Then i will add this line 

Many thanks

killer0071234, 2024-05-18:
Hello @bossoq 
Can you add this change as PR?
Thanks Daniel

> Add this line to `esphome/components/mitsubishi/mitsubishi.cpp` after line 59
> 
> `traits.set_supports_current_temperature(true);`
> 
> [Add support current temperature to mitsubishi climate ir](https://github.com/bossoq/esphome/commit/8f666bfdef9814f78c883188cd1444c0ad5045fe)

sosso-b, 2024-05-18:
I arrive to do it using the custom external component, thank you 
[image]

Thank you solved for me

closed 2024-05-18 by sosso-b