esphome/issues #5772: ESPHome does not autodiscover devices even though they are visible via mDNS

andyearnshaw, 2024-05-07:
### The problem

I have a strange problem. I migrated Home Assistant OS with the ESPHome addon from a Raspberry Pi 4 to a NUC with Proxmox, and none of my existing devices are discovered (to be able to "adopt"). I've tried both an LXC container and the ESPHome addon for Home Assistant, both dashboards remain completely empty.

All devices appear to be visible (at least, to Avahi) from the Proxmox terminal, as well as the ESPHome container/addon:

```
root@esphome:~# avahi-browse _esphomelib._tcp
+   eth0 IPv4 delicacy-smart-plug-3                         _esphomelib._tcp     local
+   eth0 IPv4 delicacy-smart-plug-2                         _esphomelib._tcp     local
+   eth0 IPv4 delicacy-smart-plug-1                         _esphomelib._tcp     local
+   eth0 IPv4 tv-bed                                        _esphomelib._tcp     local
```

It appears that something is not working correctly with the zeroconf setup, but there doesn't appear to be any logging in this area of the codebase.

```
2024-05-07 14:16:33,945 DEBUG Using selector: EpollSelector
2024-05-07 14:16:33,949 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /root/config/...
2024-05-07 14:16:33,949 DEBUG Updating dashboard entries
2024-05-07 14:16:33,952 DEBUG Using icmplib in privileged=True mode
2024-05-07 14:16:36,929 INFO 304 GET /ping (192.168.1.239) 2.49ms
2024-05-07 14:16:39,918 INFO 304 GET /ping (192.168.1.239) 0.86ms
2024-05-07 14:16:39,920 DEBUG Updating dashboard entries
2024-05-07 14:16:39,921 INFO 304 GET /devices (192.168.1.239) 1.81ms
2024-05-07 14:16:42,915 INFO 304 GET /ping (192.168.1.239) 0.89ms
2024-05-07 14:16:45,914 INFO 304 GET /ping (192.168.1.239) 0.73ms
```

ESPHome remains empty:

[image]

If I add a configuration yaml for each host, ESPHome will connect. It's just the autodiscovery that fails. Please let me know if there's any way I can help further debug this.

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

randybb, 2024-05-07:
There is no such feature. You need to copy all yaml files to the new machine. On devises is only a binary, so there is no way to import them from working devices.
Adoption is a special feature only for devices with a prebuild binary.

closed 2024-05-07 by randybb

andyearnshaw, 2024-05-07:
My mistake then, I thought I had seen this feature work on my raspberry pi, that must have been the prebuilt binary.