esphome/issues #5775: Ultrasonic sensor jsn_sr04t does not measure less than 250mm

kristi5, 2024-05-09:
### The problem

I am using jsn_sr04t in mode 1 (continous measurement via serial) to measure water level in tank.

Shortest measured range in my sensor is around 220mm but any measurement below 250mm is not reported by the component. 
Instead the log contains lot of lines with `Invalid data read from sensor`.

The component should consider any measurement which passed checksum control as valid and the measurement should be provided to HA.

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

jsn_sr04t

### Anything in the logs that might be useful for us?

```txt
I have plenty of following lines in log

[20:07:43][W][jsn_sr04t.sensor:044]: Invalid data read from sensor: FF.00.E0.DF
[20:07:43][W][jsn_sr04t.sensor:044]: Invalid data read from sensor: FF.00.E0.DF
[20:07:43][W][jsn_sr04t.sensor:044]: Invalid data read from sensor: FF.00.E0.DF
[20:07:43][W][jsn_sr04t.sensor:044]: Invalid data read from sensor: FF.00.E0.DF
```
```

### Additional information

Checked the [code ](https://github.com/esphome/esphome/blob/026c3a69b89aa718218c4539b0ef7d204ca1e76a/esphome/components/jsn_sr04t/jsn_sr04t.cpp#L39) and there is fixed constant limiting that all measurements below 250 are considered as invalid.

Why the measurement is considered as invalid when the checksum control passed?

zagi988, 2024-05-17:
As far as I know, since I've been using it with custom sensor - it doesn't measure under 20 cm. Why it shows unknown under 25 instead of just showing 20 when under 20, I don't know. But yeah, it's an issue.

adam-kulagowski, 2024-05-29:
There several revisions and datasheets for this sensor. According to them the minimum range is from 20cm up to 25cm. However as long as the checksum is OK I believe we should allow lower values between 20 and 25 cm. @jesserockz should I make a PR for this?

adam-kulagowski, 2024-05-29:
**Last minute update:** in my case sensors sometimes throws a value lower than 25cm when there is an issue w/ measurement (garbage, echo etc). So for me, 25cm limit works just fine, so I withdraw the idea of PR.

kristi5, 2024-05-30:
According to me its not invalid data as the log msg suggest while the checkum is OK.

My water tank has highest level around 20cm from the top where the sensor is located. My experience is that sensors are returning 20cm for whatever shorter measurement which is still OK but no measurement at all is not ok.

In my case i will not get any measurement from the sensor (in HA its reported as unknown) and it may last for days or months unless the level in the tank drops below 25cm or more from the top.

So when i will not get any measurement from the sensor then i do not know if the tank is full or sensor stop working or is unavailable.

Any suggestion how to handle this no-measurement state in esphome?

kristi5, 2024-05-30:
If sensor sometimes throws wrong measurement (but with correct checksum) due to some echoes or unstable water level then filtering those values is task for sensor filters. The esphome sensor component jsn_sr04t should provide any valid (CRC OK) value anytime it is received from the hardware. Rest should be handled by the user inside yaml file.

ericu65, 2024-12-16:
Same issue noticed on my side as well. JSN-SR0T4-V3.0 spec'd at 21cm-600cm bellow 25cm will fail returning values with:
`[16:19:59][W][jsn_sr04t.sensor:051]: Invalid data read from sensor: FF.00.CE.CD` (range should be 206mm)

Checksum is correct and FF.00.D0.CF is another raw output example (range should be 208mm).

Associated PR: https://github.com/esphome/esphome/pull/6023

@Mafus1 is there any reason why you cut off below 250mm? I think the filters set as below should have the same effect for people who want to ignore data below 250mm:
```
  filters:
    - clamp:
        min_value: 25
        ignore_out_of_range: true
```

zagi988, 2025-02-20:
Has this been resolved?
Because I'm still getting 

> [W][jsn_sr04t.sensor:051]: Invalid data read from sensor: FF.00.C8.C7
when it should read 200 mm.

Mafus1, 2025-02-21:
There were two reasons I cut it off below 25cm.
- Different data sheets for this sensor mention differen cut off values.
- For the sensors I tested, anythung below 25cm the sensor started to jump up to 600cm.

I think the only real solution would be to remove the limit and just output the data as long as the checksum is valid. But this would require most users to apply filters.