esphome/issues #5778: ESP8266 GPIO mode open drain on IO16 not working

oerkel47, 2024-05-10:
### The problem

I am using a WemosD1 Mini type board. Configuring pin D0 (which afaik is GPIO16) as open drain does not work. There are no errors or warning, but it just does not work. The IO will not pull to ground when supposed to. Other Pins work.

As far as I understand the ESP8266 Datasheet, open drain mode should basically work on all IO.

I suspect it could be related to the Pin not supporting PWM/analog mode. Does open drain make use of "analog" mode related framework functions and is thus ignored?

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1 Mini NodeMcu ESP8266-12F

### Component causing the issue

gpio

### Example YAML snippet

```yaml
output:
  - platform: gpio
    id: relay
    inverted: true
    pin:
      number: D0
      mode: OUTPUT_OPEN_DRAIN
switch:
  - platform: output
    output: 'relay'
    name: "AC Relay"
```

ssieb, 2024-05-10:
GPIO16 is special and doesn't support that mode.

closed 2024-05-10 by ssieb

oerkel47, 2024-05-10:
A warning/error would be great then.