esphome/issues #5781: Ethernet Component not working with AZDelivery W5500 Ethernet Shield (SPI)

AndySymons, 2024-05-10:
### The problem

Main issue: I am pairing an [Espressif Devkit v4](https://www.ebay.co.uk/itm/254803716289) with an [AZDelivery W5500 Ethernet module](https://www.amazon.co.uk/dp/B07CYVRMPL). The configuration yaml compiles and loads correctly, and the program initially says it is connected, but after that all requests are timed out. 

Side issue 1: DNS is not resolved. I have to use manual-IP to connect at all. 
Side issue 2: The [documentation](https://esphome.io/components/ethernet.html) says that reset and interrupt are optional, but if I do not include them I get errors. 
Side issue 3: The [documentation](https://esphome.io/components/ethernet.html) says I can set a clock speed of up to 80MHz, but if I do that I get errors.  

### Which version of ESPHome has the issue?

I don't know where to find the number, but it is the latest version installed 2024-04-29 18:07:01 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

core-2022.11.5

### What platform are you using?

ESP32

### Board

AZDelivery W5500 Ethernet Shield

### Component causing the issue

Ethernet W5500 SPI

### Example YAML snippet

```yaml
esphome:
  name: hardwired-alarm-interface
  friendly_name: Hardwired Alarm Interface

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "BKVgvPgWeTB9FUDehvoT8JWCH/iEcpIDnG4hGd6OACY="

ota:
  password: "1e78adf2c5aa91872a69e7594a033450"

ethernet:
  type: W5500
  cs_pin: 18
  clk_pin: 19
  mosi_pin: 21
  miso_pin: 23
  reset_pin: 22
  interrupt_pin: 36

  manual_ip:
    static_ip: 192.168.0.42
    gateway: 192.168.0.1
    subnet: 255.255.255.0
```

[... 114 lines cut ...]
[10:11:46][C][mdns:115]: mDNS:
[10:11:46][C][mdns:116]:   Hostname: hardwired-alarm-control
[10:11:46][C][ota:096]: Over-The-Air Updates:
[10:11:46][C][ota:097]:   Address: 192.168.0.42:3232
[10:11:46][C][ota:100]:   Using Password.
[10:11:46][C][ota:103]:   OTA version: 2.
[10:11:46][C][api:139]: API Server:
[10:11:46][C][api:140]:   Address: 192.168.0.42:6053
[10:11:46][C][api:142]:   Using noise encryption: YES
[10:11:46][D][esp-idf:000]: E (3070) w5500.mac: w5500_send_command(123): send command timeout

[10:11:46][D][esp-idf:000]: E (3070) w5500.mac: emac_w5500_transmit(534): issue SEND command failed

[10:11:46][D][esp-idf:000]: E (4271) w5500.mac: w5500_send_command(123): send command timeout

[10:11:46][D][esp-idf:000]: E (4271) w5500.mac: emac_w5500_transmit(534): issue SEND command failed

[10:11:52][D][esp-idf:000]: E (22658) w5500.mac: w5500_send_command(123): send command timeout

[10:11:52][D][esp-idf:000]: E (22658) w5500.mac: emac_w5500_receive(578): issue RECV command failed

[10:12:31][D][esp-idf:000]: E (61484) w5500.mac: w5500_send_command(123): send command timeout

[10:12:31][D][esp-idf:000]: E (61484) w5500.mac: emac_w5500_transmit(534): issue SEND command failed
```

AndySymons, 2024-05-13:
I am sorry that no one has responded to this issue yet :-(  

Shooting in the dark, but in attempt to narrow down the main issue, I did several more tests and recorded the log files, which are attached. 

## Initial state 
As in my original posting…  
- There is no manual IP.
- All 6 pins are physically connected (four SPI plus Reset and interrupt).
      cs_pin: 18
      clk_pin: 19
      mosi_pin: 21
      miso_pin: 23
      reset_pin: 22
      interrupt_pin: 36
- Clock speed is default. 

## Test 1 
I removed the definition of the interrupt pin, wondering if it is necessary because the ESPHome documentation says it is optional and the AZ-Delivery example with Arduino does not use it. 
Result: no new errors. Same errors as the initial state. 
Conclusion: not sure. Maybe there is a default? Still need to actually cut the wire (test 3). 

## Test 2
I removed the definition of the reset pin, wondering if it is necessary because the ESPHome documentation says it is optional, although the AZ-Delivery example with Arduino does use it. 
Result: ‘reset’ errors and could not load the program at all. 
Conclusion: reset is mandatory. 

## Test 3 
I restored the reset pin, and physically cut the wire to the interrupt pin because the ESPHome documentation says it is optional and the AZ-Delivery example with Arduino does not use it.  
(It is till not defined in the config since test 1).  
Result: still fails but gives different errors [Ethernet: 237]
Conclusion: inconclusive, but these errors look more serious (?), so I will continue to assume for now that the interrupt pin is mandatory.

##  Test 4 
I restored the config to the initial state, i.e. all six pins physically connected and configured, no manual IP and default clock frequency. I ran it to get a fresh log of the errors. There are many, and it is hard to see which is the most fundamental error and which are knock-on errors. I checked by doing a manual reset after the initial load but that made no difference --  same errors again. 
 
## Test 5
I previously had the impression that the device connected better when a manual IP was specified. I tried specifying a manual IP, but it made no difference this time. The errors are the same. 
I therefore withdrew the manual IP again.  

## Test 6
I previously noted that the clock speed cannot be set to 80MHz. I verified that by putting it in the config, installing it and keeping the log file. 
I withdrew the clock_speed entry from the config.  

## CONCLUSIONS SO FAR 
1. The need for an interrupt pin is unclear. Leaving it out seems to create more serious errors, but it needs to be re-tested when a situation is reached when no other errors are being reported. In the meantime, I note that the ESPHome documentation could be clearer about the use for this pin and the consequences of not connecting and/or defining it. 
2. The reset pin is clearly important. Without reset, no program can be loaded. Perhaps there is an awkward way of manually resetting the board when necessary, but I did not pursue that. The ESPHome documentation should be much clearer on this point; ‘optional’ does not cover it!  
3. Manual IP is not necessary (I withdraw my previous comment that this appeared to be the case).  
4. The clock speed cannot be set to 80MHz and the ESPHome documentation is probably wrong about the default being 29.97MHz; the error log says “the driver always tries to find the closest frequency to your configuration”.  So setting the clock speed at all only makes sense if you want to slow it down for some reason! 

## NEXT STEPS? 
I have cleared up some side issues, and eliminated some causes, but still not found the central problem. If no one has an answer, I would at least like some guidance on how to proceed with further diagnostics. 
It would help a lot if someone could tell me some more about what the error messages in the log mean. (I cannot find any list online). 
The following look like they may be important. They occur several times, with variations in the numbers  in brackets (what do they mean?). 
In all cases except the removal of reset, the configuration was successfully compiled and uploaded. The error messages occur after the line ‘[C][ethernet:034]: Setting up Ethernet…’. 

[D][esp-idf:000]: E (679) w5500.mac: w5500_send_command(123): send command timeout
[D][esp-idf:000]: E (680) w5500.mac: emac_w5500_start(279): issue OPEN command failed
[D][esp-idf:000]: E (690) w5500.mac: emac_w5500_set_link(422): w5500 start failed
[D][esp-idf:000]: E (700) esp_eth: eth_on_state_changed(121): ethernet mac set link failed
[D][esp-idf:000]: E (711) w5500.phy: w5500_update_link_duplex_speed(84): change link failed
[D][esp-idf:000]: E (721) w5500.phy: w5500_get_link(108): update link duplex speed failed
[D][esp-idf:000]: E (721) esp_eth: esp_eth_start(272): phy get link status failed
[D][esp-idf:000]: E (2517) w5500.mac: emac_w5500_transmit(534): issue SEND command failed

So the problem is presumably somewhere in the Ethernet setup. But where?? I would very much appreciate any help an expert may be able to give to determine whether there is a bug in the ESPHome code or something I can do differently in the config. 

## Attachments
[240513 0850 test 2 log - error when RST not defined.txt](https://github.com/esphome/issues/files/15292972/240513.0850.test.2.log.-.error.when.RST.not.defined.txt)
[... cut ...]
[240513 0913 test 4 log - INT restored (back to original).txt](https://github.com/esphome/issues/files/15292974/240513.0913.test.4.log.-.INT.restored.back.to.original.txt)
[240513 0924 test 5 log - manual IP added (no difference).txt](https://github.com/esphome/issues/files/15292976/240513.0924.test.5.log.-.manual.IP.added.no.difference.txt)
[240513 0939 test 6 log - clock speed set to 80MHz.txt](https://github.com/esphome/issues/files/15292977/240513.0939.test.6.log.-.clock.speed.set.to.80MHz.txt)

AndySymons, 2024-05-19:
19 May. I tried a different W5500 component, the USR-E1 (a nice little module), this time with an ESP32-S2, and got the same result -- the program compiles and loads, the device has an IP address, but it does not respond to Ethernet signals. 

[image]

#### This therefore seems to be a general problem with the ESPHome Ethernet Component for W5500. 

I guess either there is something missing in the libraries, or in the documentation? 

My config below and log file attached ... 
```
esphome:
  name: hardware-alarm-interface-3
  friendly_name: Hardware Alarm Interface 3

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "o6O/pTaTUYUGrab9xftGr46QnvFTVvXvkpaj0UeGBN8="

ota:
  password: "73e414cd91ef846e8c757e5f02ce39d6"
  
ethernet:
  type: W5500
  reset_pin: GPIO8
  interrupt_pin: GPIO9
  cs_pin: GPIO10
  mosi_pin: GPIO11
  clk_pin: GPIO12
  miso_pin: GPIO13
```

[logs_hardware-alarm-interface-3_run.txt](https://github.com/esphome/issues/files/15369077/logs_hardware-alarm-interface-3_run.txt)

nagyrobi, 2024-05-19:
Please contact us on the Discord channel linking to this isdue.

AndySymons, 2024-06-08:
I am unable to log in to Discord! Whenever I try it says that verification os required. After I verify by phone it says this number is already in use (of course it is).

AndySymons, 2024-06-08:
Done

nagyrobi, 2024-06-08:
Works now for you?

AndySymons, 2024-06-21:
Why? Has there been a fix? What version of the code do I need?

AndySymons, 2024-06-21:
It looks like I accidentally closed Update ethernet.rst #3846. I think that was due to inexperience in using  Github. In any case my proposed document change was only to say teh W5500 is not working at the moment. 

I have recently got a dfferent W5500 module working. I will retest soon with the AZDelivery board.

dominik4545, 2024-07-06:
@AndySymons: I am thinking also to buy the module for my project. Did you get it working meanwhile?

halogen28372, 2024-11-08:
My LilyGO-T-ETH with the 5500 seems to have the same problem. I did find that using the Arduino framework worked but this isn't a good solution for me.

lcnittl, 2024-11-18:
Can confirm that the same problem exists with the Adafruit Ethernet FeatherWing, which is also using the W5500 chip, in ESPHome 2024.10.3.
Thank you for all your detective work so far @AndySymons

AndySymons, 2024-11-26:
Thanks for the tip. I'll try that

MAE-91, 2025-01-06:
Hi, did the W5500 work for you?
 I'm trying to use it with the ESP32-WROOM-32E-N16, but I'm encountering the same issues.

halogen28372, 2025-01-06:
Yes, I had to include some extra text, I guess the default values are bad for this board. You can search for the post I started I'm pretty sure I listed it there. I found the solution on another post here.