esphome/issues #5785: Compiling 2 or more nodes at the same time.

WNTLtd, 2024-05-13:
### The problem

OK - So this isn't so much as an issue - because I wouldn't have thought that most people would actually do what I just did - BUT - I thought I'd flag it up anyway maybe (if not already) it goes in the docs somewhere.

I had the HASS app open on my Mac - and I updated the yaml on a couple of ESPHome nodes.
I also had the same page open as a web gui in Chrome.
While the app was doing a compile for one node - I thought - I wonder if it can do two at the same time?
So I instigated a second compile of the other node I updated, from within chrome (all based on the same HASS instance)

Suffice to say - about 10 secs in, it all stopped - I lost comms with HASS, and HASS rebooted.

Everything came back up as normalI and I could run the compiles again one at a time.

So I guess the general rule of thumb - don't try and run two or more compiles for your ESPHome devices AT THE SAME TIME!! LOL.

### Which version of ESPHome has the issue?

2.4.5.0b3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

24.5.3

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

n/a

ssieb, 2024-05-13:
What are you running HA on?  You're most likely running out of memory, although that still shouldn't cause it to reboot.  But unless you have a lot of RAM, you shouldn't do two compiles at once.

WNTLtd, 2024-05-13:
Running on an Odroid HA Blue.
4Gb ram.
So I didn't/don't know if the reboot was because ESPHome just wasn't built to be able to do two or more compiles at the same time - or if HASS was running out of hardware resources?

Perhaps Jesse could chip in here...?
Is it theoretically possible to do more than one compile at the same time?

ssieb, 2024-05-13:
It's theoretically possible, but 4GB might not be enough RAM.

WNTLtd, 2024-05-13:
Fair enough. 👍

Rudd-O, 2024-05-14:
> Is it theoretically possible to do more than one compile at the same time?

I compile 3 or 4 at a time, running on a Xen VM with 4 GB RAM, doesn't even break a sweat.

Your hardware is bad, it is overheating, or your PSU is too weak for the jerb.