esphome/issues #5786: Problems with WireGuard and esp8266 (esp-12f)

delthia, 2024-05-14:
### The problem

When uploading an script with WireGuard, I can ping the board, but the web interface doesn't work. Changing the board and uploading the same script to an esp32 allows me to use the web interface through WireGuard without any issues.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

4-relay board with esp-12f module

### Component causing the issue

Wireguard

### Example YAML snippet

```yaml
esphome:
  name: riego

esp8266:
  board: esp12e

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""
  reboot_timeout: 0s

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Riego Fallback Hotspot"
    password: ""

captive_portal:

light:
  - platform: status_led
    name: "Status"
    pin:
      number: GPIO2
      inverted: true

web_server:
  port: 80

wireguard:
  address: 10.7.0.6
  private_key: =
  peer_endpoint: wg.domain
  peer_public_key: =

  netmask: 255.255.255.255

  peer_port: 51820

  peer_preshared_key: =

  peer_allowed_ips: 10.7.0.1/24

  peer_persistent_keepalive: 25s

  update_interval: 20s

time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Madrid
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org
```

### Additional information

The logs say that the peer is online, and on the server I can see traffic between the server and the client coming from the keepalive messages. I've tried using the public domain name to access the server, but also directly the local ip to test inside the network. Both things fail but work on the esp32

asergunov, 2024-06-06:
Got the same issue. It looks like wireguard disables all the outgoing connections except allowed ones. Once I deactivate or before it's connected it works just fine. Worked around by adding SNTP server in local network and adding it in server list

```yaml
time:
  - platform: sntp
    id: sntp_time
    timezone: Europe/Madrid
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 10.7.0.1 # deploy sntp server on this address
```

Also be aware of https://github.com/esphome/issues/issues/5884

asergunov, 2024-06-06:
Still can't make it work. Maybe it's [esp_wireguard](https://github.com/trombik/esp_wireguard) bug

delthia, 2024-06-06:
I managed to make it work by removing peer_allowed_ips, but I think that with that it worked on the esp32

atlantden, 2024-08-08:
Have same issue. peer_allowed_ips removed but still not working

delthia, 2024-08-08:
Does your setup work with the same config elsewere? I recommend trying on a computer to see that it isn't a problem with the confgiuration. Or using an esp32 where for me worked, if your problem happens with other model.

I find it weird that you would use port 80 for wireguard, ports under 1024 are generally reserved and although I haven't tried, I'd be surprised if wg worked within that range. I recommend using any other port. You could test with the default 51820 and then change it if you want

atlantden, 2024-08-09:
i use esp8266
i dont use 80 gor wireguard its just webface of my module that is open trough wireguard but not ipen in browser
Wireguard connection works fine
port 6053  (for HA API) seems not ipened trough wireguard at all
i dont have esp32 now to test wireguard on it but i think that is software bug

prchal, 2025-01-10:
Hi all,
same here. Can't connect to HA. Nothing in logs:
`[D][wireguard:098]: WireGuard remote peer is online (latest handshake 2025-01-10 09:23:46 CET)`
Ping to device works.
`64 bytes from 172.25.162.69: icmp_seq=145 ttl=255 time=12.0 ms`
[image]
ESP32 works fine.