esphome/issues #5788: ESP-IDF mqtt uses secured connection by default?

MGWGIT, 2024-05-14:
### The problem

I try to send unsecured mqtt message to my broker, but receive esp-idf errors caused tls connection fails.

Why tls error occured for insecure publish? `[E][mqtt.idf:162]: Last error code reported from esp-tls: 0x8008`

### Which version of ESPHome has the issue?

2024.4.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
esphome:
    name: esp-xxx
    platformio_options:
      board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      COMPILER_OPTIMIZATION_SIZE: y
    advanced:
      ignore_efuse_mac_crc: false
  variant: esp32s3

mqtt:
  broker: "iot.fr-par.scw.cloud"
  port: 1883
  username: "my_user_id"
  #username: "tester"
  password: ""
  topic_prefix: "test"
  skip_cert_cn_check: true
```

### Anything in the logs that might be useful for us?

```txt
[14:09:41][V][mqtt.idf:119]: Event dispatched from event loop event_id=7
[14:09:41][V][mqtt.idf:122]: MQTT_EVENT_BEFORE_CONNECT
[14:09:42][V][esp-idf:000]: E (206625) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
[14:09:42]
[14:09:42][V][esp-idf:000]: E (206628) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
[14:09:42]
[14:09:42][V][esp-idf:000]: E (206629) MQTT_CLIENT: MQTT connect failed
[14:09:42]
[14:09:42][V][mqtt.idf:119]: Event dispatched from event loop event_id=0
[14:09:42][E][mqtt.idf:160]: MQTT_EVENT_ERROR
[14:09:42][E][mqtt.idf:162]: Last error code reported from esp-tls: 0x8008
[14:09:42][E][mqtt.idf:163]: Last tls stack error number: 0x0
[14:09:42][E][mqtt.idf:165]: Last captured errno : 0 (Success)
[14:09:42][V][mqtt.idf:119]: Event dispatched from event loop event_id=2
[14:09:42][V][mqtt.idf:131]: MQTT_EVENT_DISCONNECTED
[14:09:42][W][mqtt:323]: MQTT Disconnected: TCP disconnected.
[14:09:42][I][mqtt:234]: Connecting to MQTT...
[14:09:42][V][esp-idf:000]: I (206640) MQTT_CLIENT: Client asked to disconnect
```

MGWGIT, 2024-05-14:
Is it because `set(CONFIG_MQTT_TRANSPORT_SSL "y")` in cmake?

spuder, 2024-10-11:
Normally port 1883 is for unencrypted. Are you intending to send this to port 8883 ?

fiatguy85, 2025-11-17:
I was able to eliminate the error, by configuring a TLS connection and migrating to port 8883 to use it.  Ideally, this should not be necessary.

Ulrar, 2026-03-09:
Same here out of nowhere after updating, it's build with tls even though I'm not using it, making it fail to connect