esphome/issues #5790: ESPHome 2024.5.0 + D1 Mini = The connection dropped immediately after encrypted hello

stickpin, 2024-05-15:
### The problem

The issue starts to happen after upgrading to ESPHome 2024.5.0.
My LD2410 presence sensors are working fine with version 2024.4.2.

The firmware compilation log is attached: [logs_bedroom-presence-sensor_run.txt](https://github.com/esphome/issues/files/15319373/logs_bedroom-presence-sensor_run.txt)

I will try to rollback now and see if i will be able to recover the sensors.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP8266

### Board

D1

### Example YAML snippet

```yaml
esphome:
  name: kitchen-presence-sensor
  friendly_name: Kitchen Presence Sensor

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Kitchen-Presence-Sensor"
    password: "password"

captive_portal:

uart:
  id: uart_bus
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

[... 163 lines cut ...]
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
WARNING Can't connect to ESPHome API for kitchen-presence-sensor @ 10.10.10.154: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0). (HandshakeAPIError)
INFO Trying to connect to kitchen-presence-sensor @ 10.10.10.154 in the background
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.003s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.005s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.004s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.004s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.007s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.005s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.004s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.003s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.005s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.004s
WARNING kitchen-presence-sensor @ 10.10.10.154: Connection error occurred: kitchen-presence-sensor @ 10.10.10.154: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).
INFO Successfully connected to kitchen-presence-sensor @ 10.10.10.154 in 0.004s
```

vasyna, 2024-05-15:
disable encryption. This work with esphome 2024.5.0

stickpin, 2024-05-15:
@vasyna I don't see any changes related to encryption in esphome 2024.5.0.
Do you know why it's not working anymore?

vasyna, 2024-05-15:
I can’t say, but the problem is only on devices paired with ESP8266 + LD

CEZ15, 2024-05-15:
Exact same fault here. I have multiple PIR sensors still functioning as intended, but every LD2410 on my network is failing with the same error. Also, seem to be unable to re-flash any of them.

stickpin, 2024-05-15:
Disabling encryption doesn't help much.

```
WARNING bedroom-presence-sensor @ 10.10.10.152: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for bedroom-presence-sensor @ 10.10.10.152
WARNING Disconnected from API
INFO Successfully connected to bedroom-presence-sensor @ 10.10.10.152 in 0.003s
WARNING bedroom-presence-sensor @ 10.10.10.152: Connection error occurred: bedroom-presence-sensor @ 10.10.10.152: EOF received
WARNING Can't connect to ESPHome API for bedroom-presence-sensor @ 10.10.10.152: bedroom-presence-sensor @ 10.10.10.152: EOF received (SocketClosedAPIError)
INFO Trying to connect to bedroom-presence-sensor @ 10.10.10.152 in the background
INFO Successfully connected to bedroom-presence-sensor @ 10.10.10.152 in 0.003s
WARNING bedroom-presence-sensor @ 10.10.10.152: Connection error occurred: bedroom-presence-sensor @ 10.10.10.152: EOF received
INFO Successfully connected to bedroom-presence-sensor @ 10.10.10.152 in 0.006s
WARNING bedroom-presence-sensor @ 10.10.10.152: Connection error occurred: bedroom-presence-sensor @ 10.10.10.152: EOF received
```

stickpin, 2024-05-15:
> Exact same fault here. I have multiple PIR sensors still functioning as intended, but every LD2410 on my network is failing with the same error. Also, seem to be unable to re-flash any of them.

yes, I had to go and flash it one by one with the cable.

stickpin, 2024-05-15:
@CEZ15 I did a rollback to ESPHome 2024.4.2 addon from HA backup and installed the firmware on top of the newer one.
After that, everything is working as before.

KennoPa, 2024-05-15:
ESP8266 + LD2410 : same issue, router shows ip up and running, as I disabled webui for the performance issues, I cannot access via Web server to confirm. I will wait a fix and will not revert esphome version.

MallocArray, 2024-05-15:
I don't think it is specific to the LD2410, as others are seeing it without this, but still D1 Minis

benalexau, 2024-05-16:
I had the same log messages and an unresponsive device after upgrading to 2024.5.0 on an m5Stack Atom Lite ESP32. To troubleshoot I used https://web.esphome.io/ to view the log. This revealed repeated reboots due to `esphome assert failed: vQueueDelete queue.c:2140 (pxQueue)`. Based on this [forum post](https://forum.arduino.cc/t/esp32-assert-failed-vqueuedelete-queue-c-2131-pxqueue/1122763) I resolved the issue by disabling the ESP32 Improv service:

```yaml
#esp32_improv:
#  authorizer: none
```

poudenes, 2024-05-16:
Same issue with different board:
Revert back and device is working again.

```
#########################################################
# Below all fixed settings for dressoir PCB
#########################################################
substitutions:
  devicename: meek-dressoir
  friendly: Diningroom
  friendly_switch_1: Dressoir
  friendly_switch_2: Balcony
  ip: 192.168.100.207
  neopixel: GPIO02 # (D4)
  touch_power: GPIO16 # (D0)
  gpio_touch1: GPIO14 # (D5)
  gpio_touch2: GPIO13 # (D7)
  gpio_relay1: GPIO05 # (D1)
  gpio_relay2: GPIO4 # (D2)
  gpio_relay3: GPIO15 # (D8)

#########################################################
# Everything below can be copy/paste without problem
#########################################################
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s
  fast_connect: true
  output_power: 20.0
  manual_ip:
    static_ip: ${ip}
    gateway: 192.168.100.1
    subnet: 255.255.255.0
    dns1: 192.168.100.1
  ap:
    ssid: ${devicename}
    password: !secret password
    channel: 4

  on_disconnect:
    - light.turn_on:
        id: neopixel
        red: 1
        green: 1
        blue: 1

    - light.turn_on:
        id: neopixel
        effect: strobe

esphome:
  name: ${devicename}
  on_boot:
    priority: 600
    then:
      - switch.turn_off: touch_power
      - delay: 2s
      - switch.turn_on: touch_power
      - light.turn_on:
          id: neopixel
          red: 1
          green: 1
          blue: 1

esp8266:
  board: esp01_1m
  restore_from_flash: true

api:
[... 276 lines cut ...]
    name: "${friendly_switch_1} - Touch"
    icon: mdi:gesture-tap-button
    on_click:
      - min_length: 10ms
        max_length: 500ms
        then:
          - switch.toggle: switch1
      - min_length: 1000ms
        max_length: 2000ms
        then:
          - switch.toggle: switch_longpress1

  - platform: gpio
    pin: ${gpio_touch2}
    name: "${friendly_switch_2} - Touch"
    on_click:
      - min_length: 10ms
        max_length: 500ms
        then:
          - switch.toggle: switch2
      - min_length: 1000ms
        max_length: 2000ms
        then:
          - switch.toggle: switch_longpress2
```

KennoPa, 2024-05-16:
> I don't think it is specific to the LD2410, as others are seeing it without this, but still D1 Minis

Maybe but same board with cc1101 works fine and did not brake after update.
Will wait for the fix

poudenes, 2024-05-16:
maybe interesting. This is not the first time that a new update brick the connection/wifi etc. In 2023 I revert 2 or 3 times back to   the version I used before the update. (To bad I don't remember the versions anymore)

GLSSoftware, 2024-05-16:
same problem here with Wemos D1. 
Restored to 2024.4.2

Wheemer, 2024-05-16:
Not sure how to revert the devices since I can't even connect to them. It just keep saying connecting when I try to flash the older version. Grounding d3 doesn't get them into flash mode. Seems I may have to unsolder all connections. What a royal PITA.

CEZ15, 2024-05-16:
> Not sure how to revert the devices since I can't even connect to them. It just keep saying connecting when I try to flash the older version. Grounding d3 doesn't get them into flash mode. Seems I may have to unsolder all connections. What a royal PITA.

I found desoldering the RX / TX pins would then allow them to be flashed. Not sure why these pins being connected to sensors stop the device from allowing flashing, but at least you don't need to desolder the entire board

j-f-pinto, 2024-05-16:
> I found desoldering the RX / TX pins would then allow them to be flashed. Not sure why these pins being connected to sensors stop the device from allowing flashing, but at least you don't need to [...]

The ESP8266 only have one UART with TX and RX and it is use for flashing. If you have any device attached on those pins it will prevent the flashing.

It is also good practice to add on the Logger: section the command baud_rate: 0. This command disable the log mensagens on the serial port.

Wheemer, 2024-05-16:
> I found desoldering the RX / TX pins would then allow them to be flashed. Not sure why these pins being connected to sensors stop the device from allowing flashing, but at least you don't need to [...]

I just tried that and nothing, then I also unsoldered the 5v, still nothing...

Wheemer, 2024-05-16:
> The ESP8266 only have one UART with TX and RX and it is use for flashing. If you have any device attached on those pins it will prevent the flashing.
>
> It is also good practice to add on the Logger: [...]

Yes logger was already at 0.

tonertiffi, 2024-05-16:
> disable encryption. This work with esphome 2024.5.0

> disable encryption. This work with esphome 2024.5.0

Don't feel alone here i have the same issue on only the LD2410 sensors connected to an esp8266 just about identical to yours, restored to a version before re flashed them and they working again

tonertiffi, 2024-05-16:
will wait for a new version to see if that fixes it

ondoteam, 2024-05-16:
Not only with LD sensors. I seeing same behavior with bme680 and not with the board using the LD

vazquezjm, 2024-05-16:
Just come to say I'm experiencing the exact same issue but with a Wemos D1 mini and LD2410c

`The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.5.0).`

Logger baud rate set to 0:
```
# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
```

toogooda, 2024-05-17:
Yes also broke my D1 Mini presence sensors no idea how to revert back to an older version of esp home :(

vazquezjm, 2024-05-17:
> no idea how to revert back to an older version

Neither do I, but at least I was just testing

toogooda, 2024-05-17:
> Yes also broke my D1 Mini presence sensors no idea how to revert back to an older version of esp home :(

Actually it did an automatic backup when I upgraded and I was just now able to do the partial restore.
All working now back on 2024.3.2

Mikescotland, 2024-05-17:
Same problem here. Esp8266 and Ld2410, not working but connection OTA to flash works.
This update disabled all my presence sensors.
Wtf esphome?? Nothing listed in breaking changes..

arhills, 2024-05-17:
Same issues for me as well, reverted to previous version, unsoldered everything and flashed as a new device with usb cable, then flashed the previous yaml. All working fine again.

stoffies00711, 2024-05-17:
> Same issues for me as well, reverted to previous version, unsoldered everything and flashed as a new device with usb cable, then flashed the previous yaml. All working fine again.

Exactly the same issue and restore as @arhills above.

marky852, 2024-05-17:
I was running several D1 minis with the LH2410 nice and dandy on 2024.4.2. I upgraded and all hell broke loose. I flashed and reflashed, compiled and recompiled  to no avail. The things wouldn’t stay connected. I was on the verge of seeing red, but I was able to go back to 2024.4.2 and all is well.  My BME680 was acting weird to with the new upgrade. Had to retrograde that too.

stoffies00711, 2024-05-17:
**Here is how I fixed mine.....**
I think the D1 Mini is just that... a "mini".  I've always had timeout issues updating it and had to boot to safe-mode just get updates done.

I went through the yaml and hashed out everything except the necessary sensors I needed for my automations to work. Since I use the radartool app to calibrate the ld2410 via bluetooth anyways, I removed all the number and select inputs.  
The D1mini now works with the new esphome update and updates upload within seconds without timeout.  I honestly think that all the sensors, numbers and input selects were just too much for the d1.  I will calibrate with the app if needed but have all sensors I need for now. Just leave the bluetooth control in place so that it can be enabled when calibration is needed.

Hope this helps
[image]

stoffies00711, 2024-05-17:
how do I place my yaml here without pasting all the text?

marky852, 2024-05-17:
[image]

With ESPHome 2024.4.2 my D1 Mini is running flawlessly with all the bells and whistles and plus some.  

The update was giving me connectivity issues.

AdrianGarside, 2024-05-18:
I'm seeing something similar with 2024.4.2. I can flash perfectly fine OTA and it will connect fine and show logs. But as soon as I reboot it, it becomes inaccessible. All I can do at this point is do a reflash:

WARNING Can't connect to ESPHome API for esphome-web-1f07e0 @ 192.168.0.80: esphome-web-1f07e0 @ 192.168.0.80: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.4.2). (HandshakeAPIError)

saboaua, 2024-05-18:
I have encountered the same issue with only my d1 mini's with LD2410C and the only solution at the moment is to revert to 2024.4.2 and flash one by one connected to the PC to get them working. 

I tried deleting the encryption and reflashed, removed the device reinstalled it, and still not working correctly and the log doesn't work on 2024.5 when trying to look and the log for the LD2410C.

"Esphome Connection error occurred EOF received."

TomHBP, 2024-05-18:
Also have this problem. D1 mini with LD2410B.
I also have 3 LD2410's with the ESP-S2, and these seem to still be working.

Mikescotland, 2024-05-18:
It looks it's a problem with combination with ESP8266

jlt24, 2024-05-19:
Exact same issue here.  I was eventually able to reflash, I'm not sure what I did besides removing encryption, but it still doesn't work.  ESP just keep crashing.

davebeckster, 2024-05-19:
Ditto, same issues. Anyone working on it? Restoring backup didn't work for me.

freddujour, 2024-05-19:
> **Here is how I fixed mine to run on 2024.5.0** without having to revert back to the older version
> 
> I think the D1 Mini is just that... a "mini". I've always had timeout issues updating it and had to boot to safe-mode just get updates done.
> 
> I went through the yaml and hashed out everything except the necessary sensors I needed for my automations to work. Since I use the radartool app to calibrate the ld2410 via bluetooth anyways, I removed all the number and select inputs. The D1mini now works with the new esphome **2024.5.0** update and updates upload within seconds without timeout. I honestly think that all the sensors, numbers and input selects were just too much for the d1. I will calibrate with the app if needed but have all sensors I need for now. Just leave the bluetooth control in place so that it can be enabled when calibration is needed.
> 
> Hope this helps
> 
> Below is the list of items I kept. ![ld2410](https://private-user-images.githubusercontent.com/87973552/331737178-d10dff75-a709-4652-84c3-823002146e87.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxMDMyODEsIm5iZiI6MTcxNjEwMjk4MSwicGF0aCI6Ii84Nzk3MzU1Mi8zMzE3MzcxNzgtZDEwZGZmNzUtYTcwOS00NjUyLTg0YzMtODIzMDAyMTQ2ZTg3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE5VDA3MTYyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlmMjkxNDE0ZTA5NThhNjc2YmU5NThkNzk2YThjZGZhMzc4ODMzMWI0ZjE1MTQ1MDk4ZDU1ZDE0NzQyN2EyM2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zQnWg7zSGOKtNfyb-N-aD3BB_LsAUFv6WCLgK5vPWNM)

Worked for me on D1_mini

stoffies00711, 2024-05-19:
https://github.com/esphome/issues/issues/5790#issuecomment-2118419138
The above is a temp fix.  You can always enable the LD2410's other features again (if you need them) after the issue has been resolved.
Personally, my sensors are much more responsive now.

playersct, 2024-05-19:
This same problem
Wemos D1 mini + LD2410B

addon in HA: 2024.4.2
firmware in Wemos D1 mini: 2024.4.2
status: work

addon in HA: 2024.5.0
firmware in Wemos D1 mini: 2024.4.2
status: work

addon in HA: 2024.5.0
firmware in Wemos D1 mini: 2024.5.0
status: error

TwardyBobek, 2024-05-20:
This same problem
Wemos D1 mini + LD2410B
I'm waiting for new update

baldarn, 2024-05-20:
This same problem
Wemos D1 mini + LD2410B

Wheemer, 2024-05-20:
There's a new update out with zero mention of this issue being resolved, disappointing.

Pimentoso, 2024-05-20:
@stoffies00711 haven't tried your fix yet but sounds like a performance issue. What could it be in the last release that raised the computational load? Maybe the new event entities support?
Also looks like most of the merged PRs are only tested against ESP32 boards. It's a shame since D1 minis are so widely used

stoffies00711, 2024-05-20:
@Pimentoso I agree.  Over here the D1 mini is much cheaper and it works well.  I own only two ESP32 boards which I bought to experiment with Bluetooth proxy.  ESP8266 should be on top if the test-list.  I'm guessing that 90% of the 'branded' flashable consumer devices worldwide are 8266 based.
The devs are magicians in my eyes.  I'm sure this issue is already on their snag-list to resolve.

henk65, 2024-05-20:
> **Here is how I fixed mine to run on 2024.5.0** without having to revert back to the older version
>
> I think the D1 Mini is just that... a "mini". I've always had timeout issues updating it and had [...]

Yep, worked for me, my LD2410 + D1 mini combination is working again on ESPHome 2024.5.1
Of course this is temporary, waiting for a real fix.

smic717394, 2024-05-20:
> **Here is how I fixed mine to run on 2024.5.0** without having to revert back to the older version
>
> I think the D1 Mini is just that... a "mini". I've always had timeout issues updating it and had [...]

Hi, 

Whats your code for the LD2410?

stoffies00711, 2024-05-20:
@smic717394 forgive me - I do not know how to post a yaml here without it looking weird so I'm attaching a txt file.
You'll see it's the standard setup.  I just 'disabled' everything I didn't need.  I recommend you keep the "control bluetooth" button in case you need to enable it, like I did, so the LD2410 can be discovered via the app.

When the issue with the D1 is resolved you can just remove all the '#' to enable all features again. 

Hope it helps
[ensuite-sensor.txt](https://github.com/esphome/issues/files/15380068/ensuite-sensor.txt)

smic717394, 2024-05-20:
I think the problem is the mini d1, I have another senor with a dfrobot SEN0395 and it does the same with the  2024.5.0 and with the latest 2024.5.1, on the dfrobot one if I remove the web_server and the http_request, it works, but it takes much longer that the 2024.4. I made bin files of the good 2024 firmware, and I will disable for this sensors the firmware lalue si even if I update the epshome, will not ask to update the firmware on this sensors.

stoffies00711, 2024-05-20:
Yes I must take back what I said about the D1 Mini.
I have another variant of the board (Lolin or something) and I'm not very impressed with that one.  I have 4 Wemos D1 minis and they are awesome.

Since it worked fine with the previous version 2024.4.2 it might be something completely different causing the communication issue.  Someone suggested it was the API encryption, someone mentioned modbus with TX/RX pins...  I'm just a consumer and do not have enough knowledge on the matter...
      `

davebeckster, 2024-05-21:
I can no longer install anything to my 2410 D1Mini devices that are using
the tx/rx pins. Neither by OTA or USB.    Others have stated they had to
unsolder from the rx/tx pins to install new firmware.  Is that where we are
at??  Disassembling our sensors to recover them?  I hope I'm just having a
"brain fart" and missing something.

davebeckster, 2024-05-21:
Not sure if there is a better way, but unsoldering the pins works for me.  PIA.

poudenes, 2024-05-21:
> Same here... Kinda weird you would need to disabled encryption or a bunch of code to get it working again while there were no issues in the previous version

Im not that into the code. But first what I think: Memory issue. Not enough memory to store the code and run all the services.

poudenes, 2024-05-21:
I have 12: ESP-12 ESP8266MOD Wi-Fi Module soldered on a circuit board. They are my wall swiches. (I didn't create them)
With latest ESPhome devices went offline. I had this already in past with some versions. Never found what it was. Because a new  update later is working again.  Beside that. There was a ESPHome version that bricked my whole network. None of the devices were able to connect router. At the end I had to reset my router at all.... (Asus GTAX-11000)

I always have for all my devices a local BIN file to revert back when needed. If a ESPhome version is working for at least 1 month without issues I replace the backup BIN files.

This latest version ESPhome again devices are gone....

FGOD1983, 2024-05-21:
that is a good thing to do indeed, maybe I should do the same before updating ESPHome next time... For now I just reverted back to the 2024.4 version which solved the issues for me and I will just hold off on updating the ESPHome add-on until I see it mentioned in the fixes

Wheemer, 2024-05-21:
Two releases since they messed up the code and no fix, is anyone of importance actually checking on these issues reported?

It would seem there is no testing done before code changes are pushed out. Zero actual interaction here is very disconcerting.

otto2704, 2024-05-21:
same here after upgrading to esphome 24.5 coming from 24.3.2
but I am using an esp8266 nodemcu without sensors (only four buttons and a relay).
I can revert using the backup created when I updated esphome within HA...

otto2704, 2024-05-21:
if needed, I can upload my yaml here, as there are no sensors or hardware needed to test, this could be useful?

[57 more replies]

closed 2024-05-28 by stickpin