esphome/issues #5791: 2024.5.0 Voice assistant, no speaker sounds

hugobloem, 2024-05-15:
### The problem

I am using the ESP32-S3-BOX (non 3) firmware from esphome/firmware. However, after updating to esphome 2024.5 I get no voice return. The text on the display does come up correctly and opening the audio link in a browser plays the audio as normal.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
[11:05:18][D][voice_assistant:591]: Speech recognised as: "Tell me a joke."
[11:05:18][D][text_sensor:064]: 'text_request': Sending state 'Tell me a joke.'
[11:05:18][W][component:237]: Component voice_assistant took a long time for an operation (240 ms).
[11:05:18][W][component:238]: Components should block for at most 30 ms.
[11:05:18][D][voice_assistant:563]: Event Type: 5
[11:05:18][D][voice_assistant:596]: Intent started
[11:05:19][D][voice_assistant:563]: Event Type: 6
[11:05:19][D][voice_assistant:563]: Event Type: 7
[11:05:19][D][voice_assistant:619]: Response: "I'm here to assist with your smart home. How can I help you today?"
[11:05:19][D][text_sensor:064]: 'text_response': Sending state 'I'm here to assist with your smart home. How can I help you today?'
[11:05:19][D][voice_assistant:563]: Event Type: 98
[11:05:19][D][voice_assistant:704]: TTS stream start
[11:05:19][D][esp-idf:000][speaker_task]: I (258604) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=8

[11:05:19][D][esp-idf:000][speaker_task]: I (258612) I2S: I2S0, MCLK output by GPIO2

[11:05:19][D][esp-idf:000][speaker_task]: I (258618) ESP32_S3_BOX: I2S0, MCLK output by GPIO0

[11:05:19][D][esp-idf:000][speaker_task]: I (258622) AUDIO_PIPELINE: link el->rb, el:0x3d85c2c8, tag:raw, rb:0x3d85c438

[11:05:19][D][esp-idf:000][speaker_task]: I (258629) AUDIO_ELEMENT: [raw-0x3d85c2c8] Element task created

[11:05:19][D][esp-idf:000][speaker_task]: I (258635) AUDIO_ELEMENT: [i2s-0x3d85c024] Element task created

[11:05:19][D][esp-idf:000][speaker_task]: I (258640) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8064151 Bytes, Inter:63740 Bytes, Dram:63740 Bytes

[11:05:19][D][esp-idf:000][i2s]: I (258646) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[11:05:19][D][esp-idf:000][i2s]: I (258648) I2S_STREAM: AUDIO_STREAM_WRITER

[11:05:19][D][esp-idf:000][speaker_task]: I (258652) AUDIO_PIPELINE: Pipeline started

[11:05:20][W][component:237]: Component voice_assistant took a long time for an operation (280 ms).
[11:05:20][W][component:238]: Components should block for at most 30 ms.
[11:05:20][D][voice_assistant:563]: Event Type: 8
[11:05:20][D][voice_assistant:639]: Response URL: "http://192.168.1.102:8123/api/tts_proxy/8e80ff9caa1ef21e0bcaaea38ac66211b3483bab_en-gb_2cdeae300d_tts.microsoft.wav"
[11:05:20][D][voice_assistant:439]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[11:05:20][D][voice_assistant:445]: Desired state set to STREAMING_RESPONSE
[11:05:20][D][voice_assistant:563]: Event Type: 2
[11:05:20][D][voice_assistant:653]: Assist Pipeline ended
[11:05:21][D][esp-idf:000][speaker_task]: W (260212) AUDIO_PIPELINE: There are no listener registered

[11:05:21][D][esp-idf:000][speaker_task]: I (260219) AUDIO_PIPELINE: audio_pipeline_unlinked

[11:05:21][D][esp-idf:000][speaker_task]: W (260226) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE

[11:05:21][D][esp-idf:000][speaker_task]: I (260235) I2S: DMA queue destroyed

[11:05:21][D][esp-idf:000][speaker_task]: W (260243) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERMINATE

[11:05:21][D][esp-idf:000][speaker_task]: W (260251) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE

[11:05:21][D][esp-idf:000][speaker_task]: I (260291) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=8

[11:05:21][D][esp-idf:000][speaker_task]: I (260299) I2S: I2S0, MCLK output by GPIO2

[11:05:21][D][esp-idf:000][speaker_task]: I (260309) ESP32_S3_BOX: I2S0, MCLK output by GPIO0

[11:05:21][D][esp-idf:000][speaker_task]: I (260317) AUDIO_PIPELINE: link el->rb, el:0x3d85c2c8, tag:raw, rb:0x3d85c438

[11:05:21][D][esp-idf:000][speaker_task]: I (260325) AUDIO_ELEMENT: [raw-0x3d85c2c8] Element task created

[11:05:21][D][esp-idf:000][speaker_task]: I (260333) AUDIO_ELEMENT: [i2s-0x3d85c024] Element task created

[11:05:21][D][esp-idf:000][speaker_task]: I (260338) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8064243 Bytes, Inter:63832 Bytes, Dram:63832 Bytes

[11:05:21][D][esp-idf:000][i2s]: I (260345) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[11:05:21][D][esp-idf:000][i2s]: I (260348) I2S_STREAM: AUDIO_STREAM_WRITER

[11:05:21][D][esp-idf:000][speaker_task]: I (260350) AUDIO_PIPELINE: Pipeline started

[11:05:25][D][voice_assistant:563]: Event Type: 99
[11:05:25][D][voice_assistant:712]: TTS stream end
[11:05:25][D][voice_assistant:310]: End of audio stream received
[11:05:25][D][voice_assistant:439]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[11:05:25][D][voice_assistant:445]: Desired state set to RESPONSE_FINISHED
[11:05:27][D][esp-idf:000][speaker_task]: W (266700) AUDIO_PIPELINE: There are no listener registered

[11:05:27][D][esp-idf:000][speaker_task]: I (266707) AUDIO_PIPELINE: audio_pipeline_unlinked

[11:05:27][D][esp-idf:000][speaker_task]: W (266716) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE

[11:05:27][D][esp-idf:000][speaker_task]: I (266723) I2S: DMA queue destroyed
```

Findarato, 2024-05-16:
Some one might want to merge all the related issues into one, or invent a search function ;-)

ayavilevich, 2024-05-16:
I have the same issue on a box 3 device after updating to 2024.5.0 .
The wav file in the log has the right content and the right content is displayed on the display but there is no audio except for popping sounds.
How can I rollback to a previous firmware version to confirm this is the cause?

hugobloem, 2024-05-16:
If you have a back up of the older ESPHome add-on you can roll it back and then reinstall the firmware on your Box. I unfortunately did not back up my ESPHome instance :cry:

hugobloem, 2024-05-16:
> Some one might want to merge all the related issues into one, or invent a search function ;-)

The issue is different though... Before there was actually audio, now there isn't.

Pat0856, 2024-05-16:
I have also have the same issue on a box 3 device after updating to 2024.5.0 .

tbrasser, 2024-05-16:
I first thought this was an issue with music assistant, however I do get output, just very high-pitched, so maybe there is a sampling/bitrate issue at the root of this?

smcnaught1, 2024-05-17:
Also have the issue with ESP32-S3-BOX-3 wth latest ESPHome 2024.5.0.  Audio does work with the M5Stack Atom Echoes.

WarC0zes, 2024-05-17:
> Audio does work with the M5Stack Atom Echoes.

Hi,
It doesn't work for me with an Atom echo. After the voice response (no sound), the atom echo reboot.
I downgraded to version 2024.4.2 and it works.

almoney, 2024-05-17:
Where is the downgrade procedure documented?  I see the same issue.

cryptk, 2024-05-17:
> Where is the downgrade procedure documented? I see the same issue.

Assuming you are running it as a home-assistant addon, you should have a backup from before the upgrade (Home Assistant offers to make these by default), you have to restore that backup.  If you have a full backup, you can do a partial restore and only restore the ESPHome addon.

almoney, 2024-05-17:
> Assuming you are running it as a home-assistant addon, you should have a backup from before the upgrade (Home Assistant offers to make these by default), you have to restore that backup. If you have [...]

I just installed ESP Home as I just got a ESP32 S3 Box 3 so when I installed ESPHome it took me to 2024.5.0 so I had no way to go back.  

I solved by going back using [https://github.com/khenderick/esphome-legacy-addons/blob/main/README.md](url) an now have audio responses working.

rccoleman, 2024-05-18:
The tip at the end of [this issue ](https://github.com/esphome/issues/issues/5810#issue-2302645723)description fixed audio playback for me.  Essentially, pinning the `esp-idf` version to `version: 4.4.6`, implying that [this](https://github.com/esphome/esphome/pull/6703) caused the regression.

ayavilevich, 2024-05-18:
confirming esp-idf version change as a condition to the bug.
adding

```
esp32:
  framework:
    type: esp-idf
    version: 4.4.6
```

to the yaml resolves the issue.

almoney, 2024-05-18:
> confirming esp-idf version change as a condition to the bug. adding
> 
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 4.4.6
> ```
> 
> to the yaml resolves the issue.

For the new guys like me, which yaml file gets this?

Pat0856, 2024-05-18:
confirming also esp-idf version change as a condition to the bug. adding

esp32:
  framework:
    type: esp-idf
    version: 4.4.6
to the yaml resolves the issue.

mad-tunes, 2024-05-20:
I find the same after updating to esphome 2024.5.1 as I did with 2024.5.0.
`version: recommended` occasionally plays the first sound but then nothing more, `version: 4.4.6` plays everything OK.

cl0ud6uru, 2024-05-21:
Just wanted to chime in that I'm also having the No sound issues on both my S3-Box3 devices. They worked fine, albeit way to quite, until the recent updates.

hargcore, 2024-05-21:
Experienced the same issue with the chirp sound playing but no Text to Speech response from HA on a brand new S3 implementation, I had installed the ESPHOME update - As suggested added this to the YAML and I now have voice response.

enoquefcd, 2024-05-22:
same issue on HA Core 2024.5.4
ESPHome 2024.5.2
no audio output and takes too long listening

styphonthal, 2024-06-02:
I just got the esp32-s3-box-3, after installing the ESPhome software it would respond to voice, do the command, but only text would appear. I did that roleback using esp32 version 4.4.6, now it responds with voice. It does take a little time to respond, and I am using Whisper and Piper for audio.

Also sometimes it freezes after I give a command, it does the command, responds on screen, then freezes before any audio. It will stay frozen until unplugged or using the reboot button.

adamlc, 2024-06-03:
I've bought a brand new esp32-s3-box-3 to give voice assistant a try. Whatever I've tried I can't get audio to work. I just get a clicking sound where there should be audio.

I've tried downgrading esphome to `2024.4.2` and esp-idf to `4.4.6` but don't have any joy sadly. Tried all different combinations of just downgrading esphome, just downgrading esp-idf. Downgrading both etc.

The microphone however has seemed to work fine, regardless of version!

Not sure if theres a way I can test if my speaker is faulty or not. But it seems lots of people have issues in general!

tannisroot, 2024-06-03:
> I've bought a brand new esp32-s3-box-3 to give voice assistant a try. Whatever I've tried I can't get audio to work. I just get a clicking sound where there should be audio.
>
> I've tried downgrading [...]

Is the factory demo firmware capable of playing audio for you?

tannisroot, 2024-06-03:
> > > I've bought a brand new esp32-s3-box-3 to give voice assistant a try. Whatever I've tried I can't get audio to work. I just get a clicking sound where there should be audio.
> > > I've tried downgrading esphome to `2024.4.2` and esp-idf to `4.4.6` but don't have any joy sadly. Tried all different combinations of just downgrading esphome, just downgrading esp-idf. Downgrading both etc.
> > > The microphone however has seemed to work fine, regardless of version!
> > > Not sure if theres a way I can test if my speaker is faulty or not. But it seems lots of people have issues in general!
> > 
> > 
> > Is the factory demo firmware capable of playing audio for you?
> 
> Just tried flashing the factory demo and the speaker is working perfectly there. So definitely something with esphome!

Can you show the config you were flashing while specifying esp-idf to version 4.4.6?

adamlc, 2024-06-04:
> Can you show the config you were flashing while specifying esp-idf to version 4.4.6?

I tried it both as copying the full yaml and also as a package like this, neither work

```yaml
substitutions:
  name: esp32-s3-box-3-05a96c
  friendly_name: Office Assistant
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ***
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.68.26
    gateway: 192.168.68.1
    subnet: 255.255.255.0
esp32:
  framework:
    type: esp-idf
    version: 4.4.6
```

Interestingly when I flashed from the [ESPHome Projects](https://esphome.io/projects/?type=bluetooth) the sound worked! 

But when adopting in to ESPHome it forces a recompile, at that point it then broke. So whatever version has been built on the projects site works.

sammcj, 2024-06-04:
Downgrading to 4.4.6 fixed it for me as well, would be good to get this updated to support 5.2.x (5.2.2 is current).

tannisroot, 2024-06-05:
Upgrading to ESPHome 2024.5.5 fixed the issue for me.

adamlc, 2024-06-05:
Can also confirm that ESPHome 2024.5.5 fixes the issue! All working perfectly now :)

rechichidaniel, 2024-06-05:
Do you still have to use esp32 framework version 4.4.6 after updating ESPHome to 2024.5.5 ? I set mine back to recommended and I still have no sound.

```yaml
esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
```

sammcj, 2024-06-05:
Yeah ESPHome 2024.5.5 seems just as broken for me, I had to set mine back to esp-idf 4.4.6

almoney, 2024-06-05:
ESPHome speech working for me on Atom Echo and ESP32-s3.  I did have to power cycle the Atom Echo once but working fine now without:

# Last known good build for esp32 4.4.6
esp32:
  framework:
    type: esp-idf
    version: 4.4.6

mad-tunes, 2024-06-05:
2024.5.5 makes my assistant (DIY, based on a esp32s3) respond once at most even with the 4.4.6 framework set.
After that it sits there doing nothing until I restart it.
With the earlier 2024.5.* versions setting the framework did the trick.

I've got 2024.4.something installed alongside the latest for now, using that works fine.

tannisroot, 2024-06-06:
> Do you still have to use esp32 framework version 4.4.6 after updating ESPHome to 2024.5.5 ? I set mine back to recommended and I still have no sound.
>
> ```yaml
> esp32:
>   board: esp32-s3-devkitc-1 [...]

I'm using the template config for BOX3 (voice-assistant/esp32-s3-box-3.yaml) with the option to set esp-idf to 4.4.6 removed. I also cleaned the build directory beforehand just to make sure, and just to double check I reflashed the firmware and the sound still works.

ayavilevich, 2024-06-06:
> I'm using the template config for BOX3 (voice-assistant/esp32-s3-box-3.yaml) with the option to set esp-idf to 4.4.6 removed. I also cleaned the build directory beforehand just to make sure, and just to double check I reflashed the firmware and the sound still works.

Same here.
For those of you who don't have it working, try to rebuild and re-upload. Make sure it succeeds in both of those operations. You can then check the log of the device to see that it is indeed running the latest build (look for build date/time).

adamlc, 2024-06-06:
> I'm using the template config for BOX3 (voice-assistant/esp32-s3-box-3.yaml) with the option to set esp-idf to 4.4.6 removed. I also cleaned the build directory beforehand just to make sure, and [...]

Yep same here!

maxi1134, 2024-06-16:
I am unable to get the microphone to work with the following code:
Does anyone has any heads-up?
```
substitutions:  
  micro_wake_word_model: alexa
  name: esp32-s3-box-3-04dc50
  friendly_name: ESP32 S3 Box 3 04dc50
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
esp32:
  board: esp32-s3-devkitc-1
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
      ```

ESP4Ever, 2024-06-17:
Hi!
HW: AI-Thinker ESP32 Audio Kit v2.2 (ESP32 A1S)
ESPhome: 2024.5.5
ISSUE:  no sound on I2S speaker component. If media player component is used instead - response played only once and then the voice assistant can't be started again till ESP restart but the media player still can play files/radios

cryptk, 2024-06-25:
FWIW, on my ESP32-S3-BOX (not the BOX-3), I now have a fully working voice assistant again with the recent 2024.6.X releases without the ESP-IDF version workaround and things are working properly.

maxi1134, 2024-06-26:
Still no reaction to my "Alexa"s with the latest firmware and this code:
```

substitutions:
  name: esp32-s3-box-3-0505ec
  friendly_name: ESP32 S3 Box 3 Office
packages:
  esphome.voice-assistant: github://esphome/firmware/wake-word-voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: Sd+//=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

hugobloem, 2024-06-26:
It works for me on both the ESP32-S3-BOX and the ESP32-S3-BOX 3. I use the default firmware for both as provided in [esphome/firmware](https://github.com/esphome/firmware/tree/main/wake-word-voice-assistant). I use the on-device wake-word detection which works well, but switching to openWakeWord on HASS does not seem to work. Although, I am not sure whether that is an esphome or openWakeWord issue. 

Please comment a :+1: if it has been resolved for you and we can close this issue.

ayavilevich, 2024-06-26:
I upgraded from a working version 2024.5.5 to latest 2024.6.2 and at first the device stopped responding to commands.
I think what happens is that during the upgrade the configuration of the device changes and it is no longer in sync with the configuration of the device in HA. So the device was doing VAD and trying to have HA to perform wake word detection where HA config was set for wake word to be done on the device.
Once I toggled "Wake word engine location" back and forth it started working as before. If the device is not working for you after the firmware upgrade, try the toggle and comment a 👍 if it resolves your issue.

maxi1134, 2024-06-26:
I have this code but it is still not working 

I get a white screen on the ESP box but no answer to either alexa or okay nabu
```
---
substitutions:
  name: esp32-s3-box-office
  friendly_name: ESP32 S3 Box Office
  loading_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/loading_320_240.png
  idle_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/idle_320_240.png
  listening_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/listening_320_240.png
  thinking_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/thinking_320_240.png
  replying_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/replying_320_240.png
  error_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/error_320_240.png
  timer_finished_illustration_file: https://github.com/esphome/firmware/raw/main/voice-assistant/casita/timer_finished_320_240.png

  loading_illustration_background_color: "000000"
  idle_illustration_background_color: "000000"
  listening_illustration_background_color: "FFFFFF"
  thinking_illustration_background_color: "FFFFFF"
  replying_illustration_background_color: "FFFFFF"
  error_illustration_background_color: "000000"

  voice_assist_idle_phase_id: "1"
  voice_assist_listening_phase_id: "2"
  voice_assist_thinking_phase_id: "3"
  voice_assist_replying_phase_id: "4"
  voice_assist_not_ready_phase_id: "10"
  voice_assist_error_phase_id: "11"
  voice_assist_muted_phase_id: "12"
  voice_assist_timer_finished_phase_id: "20"

  # These unique characters have been extracted from every test file of every language available on https://github.com/home-assistant/intents (14 March 2024)
  allowed_characters: " !#%'()+,-./0123456789:;<>?@ABCDEFGHIJKLMNOPQRSTUVWYZ[]_abcdefghijklmnopqrstuvwxyz{|}°²³µ¿ÁÂÄÅÉÖÚßàáâãäåæçèéêëìíîðñòóôõöøùúûüýþāăąćčďĐđēėęěğĮįıļľŁłńňőřśšťũūůűųźŻżŽžơưșțΆΈΌΐΑΒΓΔΕΖΗΘΚΜΝΠΡΣΤΥΦάέήίαβγδεζηθικλμνξοπρςστυφχψωϊόύώАБВГДЕЖЗИКЛМНОПРСТУХЦЧШЪЭЮЯабвгдежзийклмнопрстуфхцчшщъыьэюяёђєіїјљњћאבגדהוזחטיכלםמןנסעפץצקרשת،ءآأإئابةتجحخدذرزسشصضطظعغفقكلمنهوىيٹپچڈکگںھہیےংকচতধনফবযরলশষস়ািু্చయలిెొ్ംഅആഇഈഉഎഓകഗങചജഞടഡണതദധനപഫബഭമയരറലളവശസഹാിീുൂെേൈ്ൺൻർൽൾაბგდევზთილმნოპრსტუფქყშჩცძჭხạảấầẩậắặẹẽếềểệỉịọỏốồổỗộớờởợụủứừửữựỳ—、一上不个中为主乾了些亮人任低佔何作供依侧係個側偵充光入全关冇冷几切到制前動區卧厅厨及口另右吊后吗启吸呀咗哪唔問啟嗎嘅嘛器圍在场執場外多大始安定客室家密寵对將小少左已帘常幫幾库度庫廊廚廳开式後恆感態成我戲戶户房所扇手打执把拔换掉控插摄整斯新明是景暗更最會有未本模機檯櫃欄次正氏水沒没洗活派温測源溫漏潮激濕灯為無煙照熱燈燥物狀玄现現瓦用發的盞目着睡私空窗立笛管節簾籬紅線红罐置聚聲脚腦腳臥色节著行衣解設調請謝警设调走路車车运連遊運過道邊部都量鎖锁門閂閉開關门闭除隱離電震霧面音頂題顏颜風风食餅餵가간감갔강개거게겨결경고공과관그금급기길깥꺼껐꼽나난내네놀누는능니다닫담대더데도동됐되된됨둡드든등디때떤뜨라래러렇렌려로료른를리림링마많명몇모무문물뭐바밝방배변보부불블빨뽑사산상색서설성세센션소쇼수스습시신실싱아안않알았애야어얼업없었에여연열옆오온완외왼요운움워원위으은을음의이인일임입있작잠장재전절정제져조족종주줄중줘지직진짐쪽차창천최추출충치침커컴켜켰쿠크키탁탄태탬터텔통트튼티파팬퍼폰표퓨플핑한함해했행혀현화활후휴힘，？"

  micro_wake_word_model: alexa

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: dio
  project:
    name: esphome.voice-assistant
    version: "2.0"
  on_boot:
    priority: 600
    then:
      - script.execute: draw_display
      - delay: 30s
      - if:
          condition:
            lambda: return id(init_in_progress);
          then:
            - lambda: id(init_in_progress) = false;
            - script.execute: draw_display

esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

psram:
[... 679 lines cut ...]
      - id: timer_finished_page
        lambda: |-
          it.fill(id(idle_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(casita_timer_finished), ImageAlign::CENTER);
      - id: error_page
        lambda: |-
          it.fill(id(error_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(casita_error), ImageAlign::CENTER);
      - id: no_ha_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_ha), ImageAlign::CENTER);
      - id: no_wifi_page
        lambda: |-
          it.image((it.get_width() / 2), (it.get_height() / 2), id(error_no_wifi), ImageAlign::CENTER);
      - id: initializing_page
        lambda: |-
          it.fill(id(loading_color));
          it.image((it.get_width() / 2), (it.get_height() / 2), id(casita_initializing), ImageAlign::CENTER);
      - id: muted_page
        lambda: |-
          it.fill(Color::BLACK);
          id(draw_timer_timeline).execute();
          id(draw_active_timer_widget).execute();

```

almoney, 2024-06-29:
> I upgraded from a working version 2024.5.5 to latest 2024.6.2 and at first the device stopped responding to commands. I think what happens is that during the upgrade the configuration of the device [...]

This got me up and running after my upgrade.

cyberhuman, 2024-07-21:
I have just bought a ESP32-S3-BOX-3 and have the same issue, no speaker sound with ESPHome.
I tried setting the esp-idf version to 4.4.6 and rebuilding the firmware but it didn't help either.

tannisroot, 2024-07-21:
> I have just bought a ESP32-S3-BOX-3 and have the same issue, no speaker sound with ESPHome. I tried setting the esp-idf version to 4.4.6 and rebuilding the firmware but it didn't help either.

How did you flash it?

cyberhuman, 2024-07-22:
Hmm, false alarm.
I re-flashed the device using the ready-made project at https://esphome.io/projects/ and the sound was back.
Then I realized there was a mistake in my custom configuration; after fixing the mistake, the sound works 🤦🏻‍♂️
Sorry for bothering you!

mleglise, 2024-10-25:
Still a problem with ESP32-S3-BOX-3 and ESPHome 2024.10.2.
No sound from the speaker, even though the TTS link works.
Tried with and without setting esp-idf to 4.4.6.
Tried with and without sammcj's volume component.

Here's my latest config file.
```
substitutions:
  name: esp32-s3-box-3-52f964
  friendly_name: ESP32 S3 Box 3 Living Room Assistant
  micro_wake_word_model: alexa
packages:
  esphome.voice-assistant: github://esphome/wake-word-voice-assistants/esp32-s3-box-3/esp32-s3-box-3.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  compile_process_limit: 1

########################################################
# https://github.com/sammcj/esphome-esp-s3-box-3-volume
  on_boot:
    then:
      - lambda: |-
          auto volume_component = id(speaker_volume);
          volume_component->set_volume(0.9);

external_components:
  - source: github://sammcj/esphome-esp-s3-box-3-volume@main
    components: [esp_box_volume]
    refresh: always

esp_box_volume:
  id: speaker_volume

########################################################
# attempted fix for speaker not working
esp32:
  framework:
    type: esp-idf
    version: 4.4.6

#voice_assistant:
#  volume_multiplier: 10.0

api:
  encryption:
    key: !secret api_key_s3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Here's the log of it reacting to "Alexa, what is today's date?"

```
[15:17:51][D][micro_wake_word:162]: The 'Alexa' model sliding average probability is 0.951 and most recent probability is 1.000
[15:17:51][D][micro_wake_word:123]: Wake Word 'Alexa' Detected
[15:17:51][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[15:17:51][D][micro_wake_word:129]: Stopping Microphone
[15:17:51][D][esp_adf.microphone:234]: Stopping microphone
[15:17:51][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[15:17:51][D][esp-idf:000][filter]: W (4689948) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT

[15:17:51][D][esp-idf:000][read_task]: E (4689951) AUDIO_ELEMENT: [filter] Element already stopped

[15:17:51][D][esp-idf:000][read_task]: W (4689982) AUDIO_PIPELINE: There are no listener registered

[15:17:51][D][esp-idf:000][read_task]: I (4689986) AUDIO_PIPELINE: audio_pipeline_unlinked
[... 153 lines cut ...]
[15:18:07][D][esp-idf:000][read_task]: I (4705861) AUDIO_PIPELINE: link el->rb, el:0x3d04b454, tag:i2s, rb:0x3d04b868

[15:18:07][D][esp-idf:000][read_task]: I (4705869) AUDIO_PIPELINE: link el->rb, el:0x3d04b5c8, tag:filter, rb:0x3d04d8a8

[15:18:07][D][esp-idf:000][read_task]: I (4705878) AUDIO_ELEMENT: [i2s-0x3d04b454] Element task created

[15:18:07][D][esp-idf:000][read_task]: I (4705883) AUDIO_THREAD: The filter task allocate stack on external memory

[15:18:07][D][esp-idf:000][read_task]: I (4705889) AUDIO_ELEMENT: [filter-0x3d04b5c8] Element task created

[15:18:07][D][esp-idf:000][read_task]: I (4705895) AUDIO_ELEMENT: [raw-0x3d04b6f8] Element task created

[15:18:07][D][esp-idf:000][read_task]: I (4705900) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:16573699 Bytes, Inter:122332 Bytes, Dram:122332 Bytes

[15:18:07][D][esp-idf:000][i2s]: I (4705906) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1

[15:18:07][D][esp-idf:000][filter]: I (4705911) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1

[15:18:07][D][esp-idf:000][filter]: I (4705917) RSP_FILTER: sample rate of source data : 16000, channel of source data : 2, sample rate of destination data : 16000, channel of destination data : 1

[15:18:07][D][esp-idf:000][read_task]: I (4705922) AUDIO_PIPELINE: Pipeline started

[15:18:07][D][esp_adf.microphone:273]: Microphone started
[15:18:07][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
```

lx-80, 2025-11-02:
One year later the problem is still blocking the playback on my ESP32 S3 Box3. 
Unfortunately i also can't pin the framework to the older Version anymore: ESP Home tells me that "Only ESP-IDF 5.0+ is supported." 

Any other workarounds or solutions? How do you guys use your s3-Boxes?

Thanks