esphome/issues #5794: Wake word not responding after update

itnassol, 2024-05-15:
### The problem

I have 3 old Google Gen2s that I have converted to Assist, new boards and they have been working perfectly for 3 months now.

This morning I updated the ESPhome to 2024.5.0.

Now they speakers only respond once, If I reboot them they respond to the wake word, then nothing it’s like they are just not listening.

UPDATE - Just re-installed the 2024.4.2 version and everything is working again.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP8266

collinstork, 2024-05-16:
This is also happening on Korvo1 v5 boards.  Also working on 2024.4.2

Willber1, 2024-05-18:
Same exact symptoms here, 2024.5.0 responds only once, reboot allows another response.

gregkg, 2024-05-18:
Exactly what I experienced with the Onju PCB replacement for google mini v2 after updating to ESPHome 2024.5.0 (on latest version of HA). Reverted to previous version of ESPHome & they are working again. Only thing I noticed in the logs were several warnings of " i2s_audio.media_player took a long time for an operation" after completing the first & only wakeword.

Willber1, 2024-05-18:
Thanks for the feedback. Same warning here.
Can you enlighten me on how to roll back ESPhome?
Cheers

gregkg, 2024-05-18:
> Thanks for the feedback. Same warning here. Can you enlighten me on how to roll back ESPhome? Cheers
> […](#)

I use the ESPHome add on in Home Assistant. Before every update, I create a backup of the add onversion I’m currently using. So it simply a restore of that.

Willber1, 2024-05-19:
Today I learned something......
Always did full backups and never considered the partial restore capabilities.
That fixed the issue!
Many thanks for your help.
Cheers,
Bill

Bandit3112011, 2024-05-20:
Same here and same after update to 2024.5.1

itnassol, 2024-05-20:
Thank you Bandit3112011, saved me a job.

Willber1, 2024-05-20:
Ya bummer, just go into Backups, select Partial and look for ESPhome update of 2024.4.2 and flash that.
Update the device and you should be good. Other units retain the 2024.5.1 version. Hopefully a future update will fix this. It might be related to the API calls.
Cheers.

PhilippeP62, 2024-06-01:
Add me to the list , same problem , same symptoms , same solution ...

PhilippeP62, 2024-06-01:
Ok pr#6849 solved the problem for me , thanks to @gnumpi

itnassol, 2024-06-02:
> Ok pr#6849 solved the problem for me , thanks to @gnumpi

Can you let us know how you did that? thanks

PhilippeP62, 2024-06-02:
You can add this to your config to load the i2s_audio component from the pull request:
```yaml
external_components:
  - source: github://pr#6849
    components: i2s_audio
    refresh: 0s
```

itnassol, 2024-06-02:
> You can add this to your config to load the i2s_audio component from the pull request:
> 
> ```yaml
> external_components:
>   - source: github://pr#6849
>     components: i2s_audio
>     refresh: 0s
> ```

So, just to be sure, I update the ESP first add this to my config yaml and it should work?

PhilippeP62, 2024-06-02:
Update esphome to 2024.5 , edit your device config, update the device , test

itnassol, 2024-06-06:
Mic not as responsive as it should be, used to be able to call assit from across the room, now I have to be right on it.

gnumpi, 2024-06-06:
> Mic not as responsive as it should be, used to be able to call Assit from across the room, now I have to be right on it.
> 
> Quick update- returned to 2024.5.4 and the mic picks up the wake word instantly and responded well, install 2024.5.5 and you have to be right next door to the speaker and mic for it to hear you.

Hey, that's strange as only the ring buffer size was changed. I will see if this is related anyway. Could you share your config?

itnassol, 2024-06-06:
> Hey, that's strange as only the ring buffer size was changed. I will see if this is related anyway. Could you share your config?

No worries, here is the config from one of them....

`substitutions:
  name: "dr-ada"
  friendly_name: "Dr Ada"
  wifi_ap_password: "password"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

logger:
api:
  encryption:
   key: "Q9sZOVZj9005+tDi9Cdwr3xPwm2+LmrD/LBRs1SC3Hc="
  services:
    - service: start_va
      then:
        - voice_assistant.start
    - service: stop_va
      then:
        - voice_assistant.stop

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: "${wifi_ap_password}"

captive_portal:

[... 447 lines cut ...]
              ESP_LOGE("touch_calibration", "Invalid button ID (%d)", button);
              return;
          }

switch:
  - platform: template
    name: Use Wake Word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - script.execute: turn_on_wake_word
    on_turn_off:
      - script.execute: turn_off_wake_word

  - platform: template
    name: Listen DR
    id: listen_dr
    optimistic: true
    on_turn_on:
      - switch.turn_off: use_wake_word
      - delay: 1s
      - voice_assistant.start_continuous
    on_turn_off:
      - switch.turn_on: use_wake_word`

gnumpi, 2024-06-06:
So far I couldn't reproduce your issue with my hardware, which HA version are you running?

itnassol, 2024-06-07:
[image]

itnassol, 2024-06-07:
It really is very strange, I had the original problem with it not responding to the wake word at all after 2024.4.2. That was fixed with the addition of the code.

external_components:

source: github://pr#6849
components: i2s_audio
refresh: 0s

With this I could install 2024.5.4 and worked well... It still works after I installed 2024.5.5, but I need to be standing right on top of the speaker for it to respond, is does then respond as expected.

closed 2024-06-07 by jesserockz

Mugga6315, 2024-06-08:
For me this fix is no longer working for 2024.5.5:
source: github://pr#6849
components: i2s_audio
refresh: 0s

Its still, that only the first wakeword usage is being answered. And after that no wake word detection at all

gnumpi, 2024-06-08:
embarrassing, but I created a new bug while fixing another one ...
Just, created a new PR for this https://github.com/esphome/esphome/pull/6875

Mugga6315, 2024-06-08:
Works again, thank you very much.

itnassol, 2024-06-09:
So, I am not quite sure where we are now, I have tried the 2024.5.5: again with
source: github://pr#6849
components: i2s_audio
refresh: 0s

Its still, that only the first wake word that is answered. And after that no wake word detection at all still.

itnassol, 2024-06-09:
Ok, So I have spent the last 3 hours trying to get this to work..

I have tried reinstalling several times. and still only responds to the wake word once.

I have found a couple of lines in the logs that might help. these are..

[image]
and
[image]

Any help would be great as i want to keep my system up to date but at the moment I am having to keep it on 2024.4.2

thanks

itnassol, 2024-06-10:
> > So, I am not quite sure where we are now, I have tried the 2024.5.5: again with source: github://pr#6849 components: i2s_audio refresh: 0s
> > Its still, that only the first wake word that is answered. And after that no wake word detection at all still.
> 
> You have to use PR #6875 just like gnumpi has written. PR #6849 is not working for 2024.5.5

Thank you Mugga6315,

I apologise for what might seem like very stupid questions, I am getting my head around Home Assistant, but the whole gitgub is a new thing for me, but I am getting there. Not too bad for an old man lol..

So, just to clarify I replace....

source: github://pr#6849
components: i2s_audio
refresh: 0s

With 

source: github://pr#6875
components: i2s_audio
refresh: 0s

Thanks again for you patience.

Mugga6315, 2024-06-10:
@itnassol 
No problem, we all are learning everyday.

Your example is correct:
source: github://pr#6875
components: i2s_audio
refresh: 0s