esphome/issues #5795: After upgrade to version 2024.5.0 of ESPHOME my esp stop providing temperature

Frazou1, 2024-05-15:
### The problem

After upgrade to version 2024.5.0 of ESPHOME my esp stop providing temperature

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP8266

### Board

esp_pool

### Component causing the issue

dallas

### Example YAML snippet

```yaml
esphome:
  name: esp-pool
  friendly_name: ESP_Pool

esp8266:
  board: esp01_1m

dallas:
  - pin: GPIO2 

sensor:
  - platform: dallas
    address: 0xe20313979404a328
    name: "Temperature Piscine"
    id: temp

  - platform: template
    name: "Temperature_F"
    id: temperature_f
    lambda: return id(temp).state * 9/5+32;
    update_interval: 3s
    
  # https://esphome.io/components/sensor/adc.html
  - platform: adc
    pin: A0
    id: ph
    name: "pH Sensor"
    update_interval: 1s
    unit_of_measurement: pH
    # https://esphome.io/components/sensor/index.html#sensor-filters
    filters:
      - median:
          window_size: 7
          send_every: 4
          send_first_at: 3
      # Measured voltage -> Actual pH (buffer solution)
      - calibrate_linear:
          - 0.59 -> 7.0
          - 0.71 -> 4.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Pool Fallback Hotspot"
    password: "XXX"

captive_portal:

i2c:
  sda: GPIO4
  scl: GPIO5

display:
  - platform: lcd_pcf8574
    dimensions: 16x2 
    address: 0x27
    id: lcd_screen
    #update_interval: 1s
    lambda: |-
      it.print(10, 0, "T:");
      it.printf(12, 0, "%.1f", id(temperature_f).state);
      it.printf(0, 0, "pH: %.2f", id(ph).state);
      if (id(ph).state < 4) {
        it.print(0, 1, "Very acidic");
      }
      else if (id(ph).state >= 4 && id(ph).state < 5) {
        it.print(0, 1, "Acidic");
      }
      else if (id(ph).state >= 5 && id(ph).state < 7) {
        it.print(0, 1, "Acidic-ish");
      }
      else if (id(ph).state >= 7 && id(ph).state < 8) {
        it.print(0, 1, "Neutral");
      }
      else if (id(ph).state >= 8 && id(ph).state < 10) {
        it.print(0, 1, "Alkaline-ish");
      }
      else if (id(ph).state >= 10 && id(ph).state < 11) {
        it.print(0, 1, "Alkaline");
      }
      else if (id(ph).state >= 11) {
        it.print(0, 1, "Very alkaline");
      }
```

### Anything in the logs that might be useful for us?

```txt
[15:19:46][W][component:170]: Component dallas cleared Warning flag
[15:19:46][E][dallas.sensor:114]: Requesting conversion failed
[15:19:46][W][component:157]: Component dallas set Warning flag: unspecified
[15:19:46][D][sensor:093]: 'Temperature Piscine': Sending state nan °C with 1 decimals of accuracy
[15:19:49][D][sensor:093]: 'Temperature_F': Sending state nan  with 1 decimals of accuracy
```

ssieb, 2024-05-15:
Is the sensor detected?

Frazou1, 2024-05-15:
If i look into the logs, yes :  [16:50:35][C][dallas.sensor:075]: DallasComponent:
[16:50:35][C][dallas.sensor:076]:   Pin: GPIO2
[16:50:35][C][dallas.sensor:077]:   Update Interval: 60.0s
[16:50:35][D][dallas.sensor:082]:   Found sensors:
[16:50:35][D][dallas.sensor:084]:     0xe20313979404a328
[16:50:35][C][dallas.sensor:089]:   Device 'Temperature Piscine'
[16:50:35][C][dallas.sensor:089]:     Device Class: 'temperature'
[16:50:35][C][dallas.sensor:089]:     State Class: 'measurement'
[16:50:35][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[16:50:35][C][dallas.sensor:089]:     Accuracy Decimals: 1
[16:50:35][C][dallas.sensor:097]:     Address: 0xe20313979404a328
[16:50:35][C][dallas.sensor:098]:     Resolution: 12

ssieb, 2024-05-15:
I don't see any changes in that component.  What was the last version that worked?
Also, GPIO2 is probably not a good pin to be using.  Isn't the LED on that pin?

tomasvida, 2024-05-27:
After today's update to esphome 2024.5.3, I have exactly the same problem.
esp32dev
dallas on gpio4
It worked without any problems until now, it is not a new project.

darthklaus59, 2024-05-28:
With version ESPHOME 2024.5.4 everything is fine again.

0x7d7b, 2024-06-08:
I still have that issue with 2024.5.5.

ssieb, 2024-06-08:
You could try this PR: https://github.com/esphome/esphome/pull/6860