esphome/issues #5798: [ETH RTL8201] Static IP stop working after update to 2024.5.0

kzajac83, 2024-05-15:
### The problem

After toady (15.05.2024) update ESPHome to 2024.5.0 static IP configuration stop working and switch into DHCP for ethernet module RTL8201.  

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP32

### Board

ESP32DEV  lilygo-eth

### Component causing the issue

etherent

### Example YAML snippet

```yaml
esphome:
  name: esphome--lilygo-eth

esp32:
  board: esp32dev
  framework:
    type: arduino

ethernet:
  type: RTL8201
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  power_pin: GPIO12
  phy_addr: 0
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.15.25
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.15.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

ota:
  password: #####

psram:
  mode: quad
  speed: 80MHz

logger:
  id: component_logger
  level: VERBOSE
  #baud_rate: 115200
 
time:
#  - platform: sntp
#    id: sntp_time
#    timezone: Europe/Warsaw
#    - servers:
#      0.pool.ntp.org
#      1.pool.ntp.org
#      2.pool.ntp.org
  #platform: sntp
  #servers: 192.168.13.1 
  #id: time_sntp
   platform: homeassistant
   id: homeassistant_time
#   timezone: Europe/Warsaw

external_components:
 # - source: github://oxan/esphome-stream-server
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]
  

wmbus:
  frequency: 868.950
  sync_mode: True
  log_unknown: True
  miso_pin: GPIO13
  mosi_pin: GPIO14
  clk_pin:  GPIO15
  cs_pin:   GPIO32
  gdo0_pin: GPIO33
  gdo2_pin: GPIO34
  #led_pin:
  #  number: GPIO5
  #led_blink_time: "300ms"
  clients:
      name: "HA"
      ip_address: "192.168.15.5"
      port: 1234
      #format: HEX

api:
  encryption:
    key: "CbH3NgcR+ZfqlldGXymtYvdCMOmsTMepQ3LL2CWsLyg="

sensor:
  - platform: wmbus
    meter_id: 0x0123456
    type: apator162
    key: "00000000000000000000000000000000"
    lqi:
      name: "lqi"
    rssi:
      name: "RSSI"
    total_water_m3:
      name: "CWU Lok"
      filters:
       - offset: 0.0

text_sensor:
  - platform: wmbus
    name: "Text debug for Apator 16-2"
```

### Anything in the logs that might be useful for us?

```txt
After connect serial adapter (without PoE eth) got:

▒▒▒▒▒▒ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, A                                                                                                                                   PB: 80000000 Hz
[   456][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I][logger:156]: Log initialized
[C][ota:483]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[V][esp32.preferences:126]: sync: key: 233825507, len: 4
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written,                                                                                                                                    0 failed
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[C][ethernet:034]: Setting up Ethernet...
[V][ethernet:617]: Hardware default RTL8201 RMII Mode Register is: 0x0FFA
[V][ethernet:624]: Setting RTL8201 RMII Mode Register to: 0x1FFA
[V][ethernet:376][sys_evt]: [Ethernet event] ETH started (num=0)
[V][ethernet:433]: DHCP Client Status: 0
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[W][ethernet:241]: Connecting via ethernet failed! Re-connecting...
[V][ethernet:433]: DHCP Client Status: 2
[...]

should not be DHCP but static IP 192.168.15.25.
```

### Additional information

Can be connected with:

Add PHY register writes to enable external clock on Ethernet with RTL8201  #6704  therefore call @jesserockz  to confirm or deny.

console08, 2024-05-16:
I'm running into the same problem after updating to 2024.5.0 and I also use "clk_mode: GPIO0_IN". I have an esp32dev board running the esp32_ble_tracker component with the esp-idf framework and it uses an RTL8201F [M91CT21 GN07B]. It's a Gemmy "Orchestra of Lights" Smart Hub board with an RTL8201 and 25MHz crystal.

I believe pin 12 is floating after checking with a magnifier but have not verified with an oscilloscope. ESPHome initializes and prints all of the normal component startup log messages to the serial port and the ethernet link comes up at 100Mbps, but then the ethernet component is returning an "insufficient TX buffer size" error and the esp32 does not respond to external connection attempts over the wired ethernet network.

> ...
[C][api:139]: API Server:
[C][api:140]:   Address: 192.168.xxx.65:6053
[C][api:142]:   Using noise encryption: YES
[D][esp-idf:000][tiT]: E (11152) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size
[D][esp-idf:000][tiT]: E (64507) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size
[D][esp-idf:000][tiT]: E (124507) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size
[D][esp-idf:000][tiT]: E (184507) esp.emac: emac_esp32_transmit(229): insufficient TX buffer size

heythisisnate, 2024-05-21:
@console08 @matwho Do you know for sure if the board you are using has an external clock? you may try setting `clk_mode GIPO_OUT` and see if that helps.

@kzajac83 the change I made in https://github.com/esphome/esphome/pull/6704 is included in 2024.5 already. Its possible that this change caused problems on other boards with this RTL8201 chip.

heythisisnate, 2024-05-24:
@kzajac83 have you been able to resolve this by changing clk_mode setting?

If no, can you share a link to the schematic for your hardware?

kzajac83, 2024-05-24:
@heythisisnate This is PoE board with module CC1101 (868 MHz) and I placed it in very hard accessible place to the best received signals from antenna then if you can check without touching it then maybe help this links:

https://www.lilygo.cc/products/t-eth-lite?variant=43120887038133

https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series

kzajac83, 2024-05-24:
PS.  Hope will help:

https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/schematic/T-ETH-Lite-ESP32.pdf
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/tree/master/schematic

heythisisnate, 2024-05-24:
@kzajac83 will you please enable VERBOSE logging on the Ethernet component and look for two lines like this:

```
[16:39:42][V][ethernet:615]: Hardware default RTL8201 RMII Mode Register is: 0x0FFA
[16:39:42][V][ethernet:622]: Setting RTL8201 RMII Mode Register to: 0x1FFA
```

And please share them here. I am interested in the initial value of the register.

kzajac83, 2024-05-24:
Set to VERBOSE but cannot found what you asking in logs hence send all

```
INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/wmbusreder-nowy.yaml...
INFO Updating https://github.com/SzczepanLeon/esphome-components.git@main
WARNING GPIO12 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Warsaw'
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esphome--lilygo-eth (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|-- SPI @ 2.0.0
|-- SmartRC-CC1101-Driver-Lib @ 2.5.7+sha.b8c6af4
|-- wmbus-drivers @ 0.0.0+20240516003403.sha.1ad30e4
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ethernet/esp_eth_phy_jl1101.c.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/psram/psram.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling .pioenvs/esphome--lilygo-eth/src/esphome/components/socket/socket.cpp.o
[... 244 lines cut ...]
[12:39:17][V][crc:031]:     calculated: 0x6421, read: 0x6421
[12:39:17][V][mbus:115]: Validating CRC for Block5
[12:39:17][V][crc:031]:     calculated: 0x591B, read: 0x591B
[12:39:17][V][mbus:115]: Validating CRC for Block6
[12:39:17][V][crc:031]:     calculated: 0xC379, read: 0xC379
[12:39:17][V][mbus:115]: Validating CRC for Block7
[... cut ...]
[12:39:17][D][sensor:094]: '0x8767345 CWU RSSI': Sending state -62.00000 dBm with 0 decimals of accuracy
[12:39:17][V][wmbus:121]: Publishing 'total_water_m3' = 0.1310
[12:39:17][V][sensor:043]: '0x8767345 CWU Lok 9': Received new state 0.131000
[12:39:17][D][sensor:094]: '0x8767345 CWU Lok 9': Sending state 0.13100 m³ with 3 decimals of accuracy
[12:39:17][V][wmbus:228]: Will send RTLWMBUS telegram to 192.168.15.5:1234 via TCP
[12:39:17][W][component:237]: Component wmbus took a long time for an operation (411 ms).
[12:39:17][W][component:238]: Components should block for at most 30 ms.

```

kzajac83, 2024-05-24:
Sorry but now cannot do then (maybe in weekend but not promise) because probably ETH communication will stop working and I will need use 5m ladder in a non-conformist place to catch the device and connect to USB port. 

Maybe @console08 @matwho can help and deliver the logs?

heythisisnate, 2024-05-24:
I totally understand.
I will try to buy one of the Lilygo for my own testing.

heythisisnate, 2024-05-24:
I just purchased one Lilygo T-Eth-lite with RTL8201 from AliExpress. I will try to solve this problem after it arrives. It may be a couple weeks. Thanks for your patience.

SzczepanLeon, 2024-05-24:
> And please share them here. I am interested in the initial value of the register.

I think You have it in first post:

```
[C][ethernet:034]: Setting up Ethernet... [V][ethernet:617]: Hardware default RTL8201 RMII Mode Register is: 0x0FFA [V][ethernet:624]: Setting RTL8201 RMII Mode Register to: 0x1FFA
```

heythisisnate, 2024-05-24:
> You have it in first post

Ah yes you're right. I missed that. Ok the initial hardware default is the same in my hardware ... so I don't know yet what the problem is. I will troubleshoot more when I receive the Lilygo board.

heythisisnate, 2024-05-24:
> However, there is a massive packet loss

Interesting. This is exactly the problem we had with the Konnected/wESP32 design that the change in https://github.com/esphome/esphome/pull/6704 fixed. Once I get my hands on the Lilygo I will work on a solution that works for both designs.

heythisisnate, 2024-05-24:
For those of you having this problem, I'm interested if you can see the first letter of the date code on the RTL8201 on your board? Here's an example, this batch code starts with M. What we noticed was a change in the default behavior of the chip from batch codes starting with N and later. I am interested if these boards have older or newer chips.
[image]

egiljae, 2024-05-25:
I have both M and N revision, and I have tried various configurations. I see no difference in behavior between them, ie. no connection with `clk_mode: GPIO0_IN`, connection but with packet loss with `clk_mode: GPIO0_OUT`.

[The documentation for T-eth-lite](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series#:~:text=the%20FAQ%20below%3E-,Since%20T%2DLITE%2DESP32%20uses%20IO0%20as%20the%20clock%20input%2C%20the%20default%20working%20level%20of%20DTR%20is%20high%20level%2C%20so%20after%20the%20download%20is%20completed%2C%20you%20need%20to%20remove%20the%20connection%20between%20DTR%20and%20IO0%2C%20so%20that%20ETH%20will%20work%2C%20otherwise%20the%20initialization%20will%20fail,-T%2DLITE%2DESP32) mentions that the board uses IO0 as clock input.

heythisisnate, 2024-05-25:
Ok I think the best approach here is to revert the change that automatically sets the register value, and instead add an option to set arbitrary register values in ESPHome configs at build time. For example:

```
ethernet:
  type: RTL8201
  set_phy_register:
    - page: 7
      value: 0x1FFA
```

This should make everyone happy and be compatible with the wide range of PHY configurations.

I'm away from work this weekend (it's a holiday weekend now in the U.S.) but I will work on it when I'm back next week.

Sorry for any headache caused!

heythisisnate, 2024-05-30:
I've just opened a PR esphome/esphome#6836 that reverts the previous change that automatically sets the PHY register values and adds new config options to arbitrarily set PHY registers as needed. This is better because we don't make assumptions about your hardware configuration and allow full flexibility for firmware developers to tweak register settings very easily.

closed 2024-06-05 by jesserockz

kzajac83, 2024-06-05:
@heythisisnate I understood that the change has been implement in 2024.5.5 therefore what I should change in my config in current version 2024.5.5 to working ethernet? Right now working config for 2024.4.2 is:

```
ethernet:
  type: RTL8201
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  power_pin: GPIO12
  phy_addr: 0
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.15.25
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.15.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
```

heythisisnate, 2024-06-06:
@kzajac83 The change has been merged but not released yet. It looks like it will make it into the 2024.6.x release. You do not need to change anything in your config. It should work again starting on the 2024.6 release.

egiljae, 2024-06-22:
I'm still experiencing packet loss with esphome version 2024.6.1 on LilyGo T-eth-lite. Does this board require additional ethernet configuration now?

console08, 2024-06-22:
@egiljae I don't have a LilyGo T-eth-lite to test but can you verify you rolled your clk_mode back to GPIO0_IN after changing it for testing earlier? No new config options should be needed. I didn't see any packet loss on my RTL8201 board after about 20 minutes of continuous pinging on 2024.6.1 just now. Can you increase the log level and see if any errors are thrown when packets are dropped?

kzajac83, 2024-06-22:
I have lilygo PoE and from smartphone via WiFi ping 360x and only one ping lost, but can be also due to WiFi, for me general working for last release ESPhome.

egiljae, 2024-06-22:
Yes, I changed it back to GPIO0_IN. I can't see any obvious errors or warning related to the issue in the logs, with level set to VERBOSE.
However, this issue seems to be present on the N revision. The M revision of the Realtek chip has 0% packet loss.

egiljae, 2024-06-22:
Actually, using
```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```
instead of `type: arduino` solves the issue on the N-revision board as well.