esphome/issues #5801: Esp32 runs hotter on Esphome 2024.5.0 firmware

Tmbao, 2024-05-16:
### The problem

I have observed that my ESP32 Bluetooth proxy seems to run hotter after updating to 2024.5.0. There is no other logic changes made to the board and the ambient temperature has been staying pretty much the same. 

[image]

The issue isn't observed on my other esp32 boards which aren't used as Bluetooth proxy 

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy
  friendly_name: Bluetooth Proxy

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

wifi:
  ap:

api:
logger:
ota:
improv_serial:

dashboard_import:
  package_import_url: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
  - platform: safe_mode
    name: Safe Mode Boot
    entity_category: diagnostic
```

WarC0zes, 2024-05-17:
Hi,
I just noticed the same problem with two Atom Lites. The temperature has increased with version 2024.5.0.
[image]

[image]

Tmbao, 2024-05-23:
Hi @WarC0zes yeah that seems to work for me as well. Thank you 🙇 ! So this seems to be something with the (esp-idf)[https://github.com/espressif/esp-idf] rather than esphome.

closed 2024-05-23 by Tmbao

WarC0zes, 2024-06-06:
Hi,
with ESP32, esphome 2024.5.5 and esp-idf 4.4.7. ESP32 take 8-10 degrees more. return to esp-idf 4.4.6, the temperature returns to normal.
I don't think you should have closed the issue.

Tmbao, 2024-06-30:
Hi I closed this because this seems to be due to the underlying esp-idf platform, not ESPHome