esphome/issues #5802: Web interface do not show device configs

Saentist, 2024-05-16:
### The problem

After update to latest DEV version no more device configs show in web ui.

[image]

 all config files are in place 
 even new device add disappear immediately. 

### Which version of ESPHome has the issue?

 2024.6.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP8266

Saentist, 2024-05-18:
2024.6.0-dev20240518
Same situation

ThomasFTL, 2024-05-19:
All actual updates are running.
ESPhome 2024.5.0 Home Assistant AddOn
Home Assistant 2024.5.4

The Web Ui show no content. Only the headers of the tables for the data and logs and the footer to change sheme are shown. I see no data and no logs.
But Home Assistant gets all data and logs.

Saentist, 2024-05-20:
@ThomasFTL  this is Issue is for managment webUI
For device WebUI check if your browser don't block web content
```
https://esphome.io/_static/webserver-v1.min.css
https://esphome.io/_static/webserver-v1.min.js
```
https://esphome.io/components/web_server.html

ThomasFTL, 2024-05-20:
@Saentist  thats work.
I test with Mozilla, Chrome and Edge.
The code of the css file and the js file are displayed as text.
But, when I call the ESP's IP, I get the two empty tables.
If I display the ESP logs in HA under ESP home, this display works.

Saentist, 2024-05-20:
@ThomasFTL create NEW issue and post YAML config file of device

ThomasFTL, 2024-05-20:
@Saentist 
I have around 10 ESP home devices that have been working flawlessly for over 2 years.
Since the last update to ESP home, the websites of all ESPs no longer work.
I see only the frame.

Saentist, 2024-05-20:
@ThomasFTL 
I have same problem with devices 
`/events`
do not return data

Saentist, 2024-05-20:
```yaml
devices-list.ts:253 Uncaught (in promise) TypeError: i.toLocaleLowerCase is not a function
    at devices-list.ts:253:4
    at Array.sort (<anonymous>)
    at devices-list.ts:249:8
    at r (polling-collection.ts:40:29)
```
this is a repeating message with browser is not happy with it.

ThomasFTL, 2024-05-20:
Update from 2024.5.0 to 2024.5.1: The same problem. No improvement.

olonsoft, 2024-05-20:
Just compiled and updated a device and I have the same error as @ThomasFTL shows in his screenshot.
Chrome shows this error in console:
```
www.js:3 Uncaught (in promise) Error: invalid template strings array
    at Ct (www.js:3:335)
    at Mt (www.js:3:1057)
    at new H (www.js:3:1221)
    at T._$AC (www.js:3:4731)
    at T.g (www.js:3:4433)
    at T._$AI (www.js:3:4089)
    at qt (www.js:3:7409)
    at yt.update (www.js:3:7811)
    at yt.performUpdate (www.js:1:6064)
    at yt.scheduleUpdate (www.js:1:5716)
```

ThomasFTL, 2024-05-20:
I have one ESP 32 Cam-Projekt. Thats work. But all ESP 8266 have the Problem with the Web-Interface.

ThomasFTL, 2024-05-20:
Why are ESP32, IDF, Libretiny and Pico W components also compiled with an esp12e?
Are they also integrated by the linker?

randybb, 2024-05-20:
guys, this issue is about the dashboard, for the web_server problem is opened another issue

vvrein, 2024-05-20:
Another fix is here https://github.com/esphome/issues/issues/5793#issuecomment-2120983660

Saentist, 2024-05-21:
If disabled caching on devices, web interface is show without problem.
but strange message appear.
`Could not allocate memory for JSON document! Requested 512 bytes, largest free heap block: 512 bytes`

ssieb, 2024-05-26:
Fixed with https://github.com/esphome/esphome/pull/6797

Saentist, 2024-05-27:
@ssieb cannot confirm, still blank dashboard devices list.
```
Uncaught (in promise) TypeError: i.toLocaleLowerCase is not a function
    at devices-list.ts:253:4
    at Array.sort (<anonymous>)
    at devices-list.ts:249:8
    at r (polling-collection.ts:40:29)
```
```
The resource http://10.244.21.87:8123/api/hassio_ingress/-J94cKKnPBFYq23v4xZUvlsJ9eTbwTncftyvBEf4MAs/static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 
was preloaded using link preload but not used within a few seconds from the window's load event. 
Please make sure it has an appropriate `as` value and it is preloaded intentionally.
```

ssieb, 2024-05-27:
Sorry, this got overrun with people from the other issue and I missed that.

Saentist, 2024-05-28:
2024.6.0-dev20240528
still not devices

Saentist, 2024-05-29:
2024.6.0-dev20240529
nothing

ssieb, 2024-05-29:
There's no need to keep posting like that unless you think there was something submitted that might change it.
This is something very specific to your install since no one else is seeing this.
Maybe try re-installing the add-on.

Saentist, 2024-05-30:
@ssieb any steps for debugging?
I use HAOS so nothing specific can be different from others with same setup.

ssieb, 2024-05-30:
It looks like what happens when a proxy doesn't pass websockets.  Check the browser network log and see if there are any errors.

Saentist, 2024-05-30:
> It looks like what happens when a proxy doesn't pass websockets. Check the browser network log and see if there are any errors.

I try different browsers same situation, no devices show, even if create new it disappear after wizard finish.

ssieb, 2024-06-02:
I don't see any issues in the HAR file.  All the devices are there and things seem to be working.  I don't know why only you would be getting that error.

ssieb, 2024-06-02:
I don't see any issues in the HAR file.  All the devices are there and things seem to be working.

Saentist, 2024-06-03:
@ssieb  Dashboard do not show devices, page generator have something wrong I guess. 

[image]
[image]
[image]

sql-aus-hh, 2024-06-06:
I am having the same issue. Two different computers, 
It seems to work if reload the integration, but just for one refresh (F5), then ESPhome-screen keeps empty

[image]

sql-aus-hh, 2024-06-06:
If you open the "issue" tab in Chrome you will receive the following hint... but I don't know if this is the reason:

[image]

Saentist, 2024-06-14:
@sql-aus-hh 
can you share what addons are also installed, maby is some conflicting one

Saentist, 2024-06-27:
Is any one else have this problem?
Any debugging suggestions are welcome.

Saentist, 2024-08-05:
Update
Installed ESP home on another virtual pc and ESPHome dashboard again stop to show device.

JovanVuceljic, 2024-09-29:
Saw comment in [this issue](https://github.com/esphome/issues/issues/5873) that bad yaml can cause the issue. 
Did ssh into `root/homeassistant/esphome` and removed the latest edited one by one until the UI  showed up.
HA addon for ssh can be used for this. I did Terminal & SSH.

Saentist, 2024-10-01:
> Saw comment in [this issue](https://github.com/esphome/issues/issues/5873) that bad yaml can cause the issue. Did ssh into `root/homeassistant/esphome` and removed the latest edited one by one until the UI showed up. HA addon for ssh can be used for this. I did Terminal & SSH.

it import it from devices
very bad situation better to see from where error come not to guess.

igorbaikov, 2025-05-21:
I think it's a bug of how names are parsed. Numbers are not parsed as a string, so it messes with esphome's dashboard. Rename devices to have letters together with numbers, i.e. 8266.yaml -> esp8266.yaml.

You can see it when you are in editing mode in esphome: the name of the device can't be just a number, it shows an error.

Saentist, 2025-05-22:
> I think it's a bug of how names are parsed. Numbers are not parsed as a string, so it messes with esphome's dashboard. Rename devices to have letters together with numbers, i.e. 8266.yaml -> [...]

10x @igorbaikov 
This is so unlogical bug but definitely this is a problem.
more interesting, this was not a bug before,
and if only one device there also no problem.
Parser need to be fixed!

Saentist, 2025-05-25:
After try to update outdated firmware of one of esp devices all they disappear again
/no number starting devices/