esphome/issues #5803: Dallas temperature sensor not found after upgrade to ESPHome 2024.5.0

mrucidlo, 2024-05-16:
### The problem

Dallas temperature sensor was normally working normally with ESPHome 2024.4.x version. I have only one sensor connected. After upgrade to ESPHome 2024.5.0, sensor is not found.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP32

### Board

WeMos WROOM

### Component causing the issue

dallas temperature sensor

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

dallas:
  - pin: GPIO04

sensor:
  - platform: dallas
    address: 0xbe00000f1b381628
    name: "Water Heater Temperature"
    #accuracy_decimals: 1
```

### Anything in the logs that might be useful for us?

```txt
[15:59:37][C][dallas.sensor:075]: DallasComponent:
[15:59:37][C][dallas.sensor:076]:   Pin: GPIO4
[15:59:37][C][dallas.sensor:077]:   Update Interval: 60.0s
[15:59:37][W][dallas.sensor:080]:   Found no sensors!
[15:59:37][C][dallas.sensor:089]:   Device 'Water Heater Temperature'
[15:59:37][C][dallas.sensor:089]:     Device Class: 'temperature'
[15:59:37][C][dallas.sensor:089]:     State Class: 'measurement'
[15:59:37][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[15:59:37][C][dallas.sensor:089]:     Accuracy Decimals: 1
[15:59:37][C][dallas.sensor:097]:     Address: 0xbe00000f1b381628
[15:59:37][C][dallas.sensor:098]:     Resolution: 12

...

[15:59:58][W][component:170]: Component dallas cleared Warning flag
[15:59:59][W][dallas.sensor:269]: 'Water Heater Temperature' - Scratch pad checksum invalid!
[15:59:59][W][component:157]: Component dallas set Warning flag: unspecified
```

andreas-bulling, 2024-05-16:
Have you tried/can you try https://github.com/persuader72/esphome-components ?

BinnenBeest, 2024-05-17:
I have the same issue, annoying 🤨

BinnenBeest, 2024-05-17:
> Have you tried/can you try https://github.com/persuader72/esphome-components ?

I don’t think this is related. I have a Dallas soldered to a gpio on esp32.

BinnenBeest, 2024-05-20:
Not solved with ESPHome 2024.5.1

3dogzzz, 2024-05-21:
I had this issue and I switched the GPIO pin I was using and that seemed to fix the issue.

BinnenBeest, 2024-05-22:
> I had this issue and I switched the GPIO pin I was using and that seemed to fix the issue.

I’m using GPIO pin 21.
It worked for years but after the upgrade it no longer does. I don't see a hardware cause. 
What pin are you using?

3dogzzz, 2024-05-22:
> > I had this issue and I switched the GPIO pin I was using and that seemed to fix the issue.
> 
> I’m using GPIO pin 21. It worked for years but after the upgrade it no longer does. I don't see a hardware cause. What pin are you using?

I’m using GPIO23 now.

mrucidlo, 2024-05-25:
After some playing around, nothing with the board I have and ESPHome version 2024.5.3 worked. 
So I replaced Dallas ESPHome component (sorry ESPHome) with https://github.com/nrandell/dallasng. Temperature sensor is back, same pin as before.

My code now looks like:

external_components:
  - source: github://nrandell/dallasng

dallasng:
  - pin: GPIO4
    update_interval: 30s

sensor:
  - platform: dallasng
    index: 0
    name: "Water Heater Temperature"
    resolution: 12
    filters:
      - filter_out: NAN

BinnenBeest, 2024-05-25:
> After some playing around, nothing with the board I have and ESPHome version 2024.5.3 worked. So I replaced Dallas ESPHome component (sorry ESPHome) with https://github.com/nrandell/dallasng. [...]

Thanks for sharing, this works for me as well

Dominator86, 2024-05-25:
I updated to 2024.5.3 and now my temperature sensors also stopped working.
Guess I'll give dallasng a try for the time being.

jasper-bosch, 2024-06-01:
> After some playing around, nothing with the board I have and ESPHome version 2024.5.3 worked. So I replaced Dallas ESPHome component (sorry ESPHome) with https://github.com/nrandell/dallasng. [...]

Thanks, this worked for me too!

sphen13, 2024-06-04:
> After some playing around, nothing with the board I have and ESPHome version 2024.5.3 worked. So I replaced Dallas ESPHome component (sorry ESPHome) with https://github.com/nrandell/dallasng. Temperature sensor is back, same pin as before.
> 

thank you - thank you.  crazy that i had to dig so far to find the answer and that esphome has not been able to address for this ESP board?

randybb, 2024-06-04:
Maybe you can test the new one_wire component that will replace the original dallas component https://github.com/esphome/esphome/pull/6860

danila1994, 2024-06-25:
I don't know if I am late or not but still.
ESPHome: 2024.6.2

//Example configuration entry

one_wire:

    platform: gpio
    pin: 13
    id: dallas_1
    platform: gpio
    pin: 12
    id: dallas_2

sensor:

    platform: dallas_temp
    address: 0xff03109794599728
    name: "Water Temperature from CV"
    one_wire_id: dallas_1 # Explicitly specify the one_wire_id
    platform: dallas_temp
    address: 0xf203029794031128
    name: "Water Temperature to CV"
    one_wire_id: dallas_2 # Explicitly specify the one_wire_id

Arkoxs, 2024-06-26:
When using the new configuration one_wire -> dallas_temp the same scratchpad errors are reported. The sensors do register.

`dallas-test`
```
substitutions:
  devicename: "dallas-test"
  upper_devicename: "Dallas Test"
  device_friendly_name: ""
  device_comment: ""
  api_encryption_key: !secret api_encryption
  ota_password: !secret ota_password
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_fallback_password: !secret fallback_password

esp8266:
  board: d1_mini

<<: !include common/common.yaml
<<: !include common/homeassistant.yaml
<<: !include common/default-sensors.yaml

esphome:
    name: $devicename
    friendly_name: $device_friendly_name
    comment: $device_comment

# Exposed switches.
switch:
  # Switch to restart the sensor.
  - platform: restart
    name: $upper_devicename Restart

# Initialize Dallas Temperature sensors on pin D1
one_wire:
  - platform: gpio
    pin: GPIO05

sensor:
  - platform: dallas_temp
    name: T1
    address: 0xbd000803e35e3a10
  - platform: dallas_temp
    name: T2
    address: 0xe9000803e39ff510
  - platform: dallas_temp
    name: T3
    address: 0xe0d41f740e64ff28
  - platform: dallas_temp
    name: T4
    address: 0xa1f643770e64ff28
  - platform: dallas_temp
    name: T5
    address: 0xa6bd27770e64ff28
```

The output is this:
```
[14:53:24][C][logger:185]: Logger:
[14:53:24][C][logger:186]:   Level: DEBUG
[14:53:24][C][logger:188]:   Log Baud Rate: 115200
[14:53:24][C][logger:189]:   Hardware UART: UART0
[14:53:24][C][gpio.one_wire:016]: GPIO 1-wire bus:
[14:53:24][C][gpio.one_wire:017]:   Pin: GPIO5
[14:53:24][W][gpio.one_wire:078]:   Found no devices!
[14:53:24][C][restart:068]: Restart Switch 'Dallas Test Restart'
[14:53:24][C][restart:070]:   Icon: 'mdi:restart'
[14:53:24][C][restart:090]:   Restore Mode: always OFF
[14:53:24][C][dallas.temp.sensor:029]: Dallas Temperature Sensor:
[14:53:24][C][dallas.temp.sensor:034]:   Address: 0xbd000803e35e3a10 (DS18S20)
[... 36 lines cut ...]
[14:53:25][C][wifi_info:011]: WifiInfo SSID 'Dallas Test SSID'
[14:53:25][C][wifi_info:012]: WifiInfo BSSID 'Dallas Test BSSID'
[14:53:25][C][wifi_info:009]: WifiInfo IPAddress 'Dallas Test IP'
[14:53:48][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[14:53:49][W][dallas.temp.sensor:074]: 'T5' - reading scratch pad failed bus reset
[14:53:49][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[14:53:49][D][sensor:093]: 'T5': Sending state nan °C with 1 decimals of accuracy
[14:53:49][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[14:53:50][W][dallas.temp.sensor:074]: 'T1' - reading scratch pad failed bus reset
[14:53:50][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[14:53:50][D][sensor:093]: 'T1': Sending state nan °C with 1 decimals of accuracy
[14:53:52][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[14:53:53][W][dallas.temp.sensor:074]: 'T2' - reading scratch pad failed bus reset
[14:53:53][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[14:53:53][D][sensor:093]: 'T2': Sending state nan °C with 1 decimals of accuracy
[14:53:53][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[14:53:54][W][dallas.temp.sensor:074]: 'T4' - reading scratch pad failed bus reset
[14:53:54][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[14:53:54][D][sensor:093]: 'T4': Sending state nan °C with 1 decimals of accuracy
[14:53:57][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[14:53:58][W][dallas.temp.sensor:074]: 'T3' - reading scratch pad failed bus reset
[14:53:58][W][component:157]: Component dallas_temp.sensor set Warning flag: bus reset failed
[14:53:58][D][sensor:093]: 'T3': Sending state nan °C with 1 decimals of accuracy
[14:54:09][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```

danila1994, 2024-06-26:
> When using the new configuration one_wire -> dallas_temp the same scratchpad errors are reported. The sensors do register.
>
> `dallas-test`
>
> ```
> substitutions:
>   devicename: "dallas-test" [...]

Try that

one_wire:
  - platform: gpio
    pin: 5
    id: dallas_1

sensor:
  - platform: dallas_temp
    name: T1
    address: 0xbd000803e35e3a10
	one_wire_id: dallas_1
  - platform: dallas_temp
    name: T2
    address: 0xe9000803e39ff510
	one_wire_id: dallas_1
  - platform: dallas_temp
    name: T3
    address: 0xe0d41f740e64ff28
	one_wire_id: dallas_1
  - platform: dallas_temp
    name: T4
    address: 0xa1f643770e64ff28
	one_wire_id: dallas_1
  - platform: dallas_temp
    name: T5
    address: 0xa6bd27770e64ff28
	one_wire_id: dallas_1

Arkoxs, 2024-06-26:
@danila1994 Wow, that actually worked, thanks:

I've added:
```
one_wire:
  - platform: gpio
    pin: GPIO05
    id: yellow_line

sensor:
  - platform: dallas_temp
    name: T1
    address: 0xbd000803e35e3a10
    one_wire_id: yellow_line
....
```

```
[16:54:10][D][dallas.temp.sensor:054]: 'T5': Got Temperature=25.9°C
[16:54:10][D][sensor:093]: 'T5': Sending state 25.93750 °C with 1 decimals of accuracy
[16:54:20][D][dallas.temp.sensor:054]: 'T1': Got Temperature=25.7°C
[16:54:20][D][sensor:093]: 'T1': Sending state 25.68750 °C with 1 decimals of accuracy
[16:54:22][D][dallas.temp.sensor:054]: 'T3': Got Temperature=25.9°C
[16:54:22][D][sensor:093]: 'T3': Sending state 25.93750 °C with 1 decimals of accuracy
[16:54:27][D][dallas.temp.sensor:054]: 'T4': Got Temperature=25.9°C
[16:54:27][D][sensor:093]: 'T4': Sending state 25.87500 °C with 1 decimals of accuracy
[16:54:31][D][dallas.temp.sensor:054]: 'T2': Got Temperature=25.6°C
[16:54:31][D][sensor:093]: 'T2': Sending state 25.62500 °C with 1 decimals of accuracy
```

ssieb, 2024-06-26:
Adding the bus id doesn't make any difference.  It was more likely picking up the updated code from the new release.

closed 2024-06-26 by ssieb