esphome/issues #5804: OTA stops working after upgrade to 2024.5.0

andreas-bulling, 2024-05-16:
### The problem

Since the upgrade to the latest ESPHome I suddenly cannot flash one of my Shelly devices OTA anymore. It has a 8266 chip with 1MB of flash. I see

```
INFO ESPHome 2024.5.0
INFO Reading configuration /config/esphome/entrance.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing entrance (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
RAM:   [=====     ]  51.1% (used 41900 bytes from 81920 bytes)
Flash: [======    ]  59.7% (used 611149 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.09 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.106
INFO Uploading /data/build/entrance/.pioenvs/entrance/firmware.bin (615296 bytes)
INFO Compressed to 418344 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP8266

### Board

shelly

mvaneijken, 2024-05-20:
Same here for version 2024.5.1. In this case a Athom Smartplug v1.

```
INFO ESPHome 2024.5.1
INFO Reading configuration /config/esphome/smartplug.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing smartplug (board: esp8285; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  48.8% (used 39936 bytes from 81920 bytes)
Flash: [======    ]  59.3% (used 607045 bytes from 1023984 bytes)
========================= [SUCCESS] Took 2.34 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.33
INFO Uploading /data/build/smartplug/.pioenvs/smartplug/firmware.bin (611200 bytes)
INFO Compressed to 419248 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

Mika255, 2024-05-20:
Same here for Athom Presence sensor 

`Processing athom-presence-sensor-f9a753 (board: esp8285; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  49.1% (used 40240 bytes from 81920 bytes)
Flash: [======    ]  59.4% (used 608089 bytes from 1023984 bytes)
========================= [SUCCESS] Took 3.78 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of athom-presence-sensor-f9a753.local
INFO  -> 192.168.1.71
INFO Uploading /data/build/athom-presence-sensor-f9a753/.pioenvs/athom-presence-sensor-f9a753/firmware.bin (612240 bytes)
INFO Compressed to 414748 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
 [ERROR] /config/esphome/athom-presence-sensor-f9a753.yaml`

pgale, 2024-05-21:
I have both the Athom smart plug x5 and the Athom presence sensor x10 - ALL fail with this issue. Never used to happen before 2024.5.0. Maybe the size of the compiled file has increased?

I wish there was a way to EXCLUDE devices from updates - I see NO reason at all to update them. It takes ages to compile for each device, despite them being the same. There has got to be a better way.

thePunderWoman, 2024-05-23:
Same here. I'll have to skip the update for the Athom devices I have. It's been a pain to try to figure out how I can update, even plugging in via USB isn't working.

pgale, 2024-05-23:
You don't NEED to update. For now, I rolled back to 2024.4.x and restarted/re-flashed all of my devices. I skipped the 2024.5 release.

andreas-bulling, 2024-05-23:
Try the following - this has worked for me:

1. Compile a minimum version of the firmware - just esphome/device as well as wifi configuration.
2. I copied the whole yaml configuration into a text file as a backup and removed everything except for the first 20 lines in the ESPHome yaml.
3. Install that firmwware OTA
4. Make sure the device connects properly to Wifi and HA
5. Copy and paste the full yaml configuration from your backup file
6. Flash that

andreas-bulling, 2024-05-24:
Pretty annoyingly, though, I have to do this workaround every single time I want to upgrade my sensor.

ssieb, 2024-05-28:
Yes, there was an accidental size increase that is now fixed.

closed 2024-05-28 by ssieb