esphome/issues #5808: Device offline but reachable after update to ESPHome 2024.5.0

FGOD1983, 2024-05-17:
### The problem

one of my D1 Mini can still do OTA updates and is pingable over mDNS and IP, but refuses to go online in both ESPHome as in HA.

I changed the way ESPHome checks if devices are online, which brought it back online in ESPHome (although going offline again from time to time), but HA still doesn't see the device.

After reverting back to my previous version (latest 2024.4 version), I was able to reflash the device and it straight on came back online in both HA and ESPHome.

The device in question has an LD2410 sensor connected to RX/TX and ledstrip data to D4(GPIO2), which all 3 should be of no issue.

I also read online that it might help to set a static IP, but this one already has static IP set.

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.5.3

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Example YAML snippet

```yaml
esphome:
  name: naambord-esp8266
  friendly_name: Naambord-ESP8266

esp8266:
  board: d1

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "{API_KEY}"  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.0.32
    gateway: 191.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.22
    dns2: 192.168.0.102

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Naambord-ESP8266"
    password: "{AP_PASS}"

captive_portal:

ota:
  password: "{OTA_PASS}"

# LED strip
light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812X
    pin: GPIO2
    num_leds: 34
    name: "NeoPixel Light"

# LD2410 sensor
uart:
  id: ld2410_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  uart_id: ld2410_uart
  throttle: 1500ms
  id: ld2410_comp

select:
  - platform: ld2410
    distance_resolution:
      name: distance resolution
    baud_rate:
      name: baud rate
    light_function:
      name: light function
    out_pin_level:
      name: out pin level

button:
  - platform: ld2410
    factory_reset:
      name: "factory reset"
    restart:
      name: "restart"
    query_params:
      name: query params

number:
  - platform: ld2410
    timeout:
      name: timeout
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
    light_threshold:
      name: light threshold

text_sensor:
  - platform: ld2410
    version:
      name: "presenece sensor version" 
    mac_address:
      name: "presenece sensor mac address"

switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode" 

binary_sensor:
  - platform: ld2410
    has_target:
      name: "Presence"
    has_moving_target:
      name: "Moving Target"
    has_still_target:
      name: "Still Target"
    out_pin_presence_status:
      name: out pin presence status
  - platform: gpio
    pin: D6
    name: gpio out pin presence
    device_class: presence

sensor:
  - platform: ld2410
    moving_distance:
      name: "Moving distance (cm)"
    still_distance:
      name: "Still Distance (cm)"
    moving_energy:
      name: "Move Energy (%)"
    still_energy:
      name: "Still Energy (%)"
    detection_distance:
      name: "Distance Detection (cm)"
    g0: 
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1: 
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2: 
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3: 
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4: 
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5: 
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6: 
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7: 
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8: 
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy
    light:
      name: light
```

### Anything in the logs that might be useful for us?

```txt
Sometimes it gives a handshake timeout, other times it gives a socket error (error 111)
```

Mikescotland, 2024-05-17:
Same here, with ld2410 mmWave as well (2 devices same behaviour) . Can update, working fine, but throws error after either to enable encryption on device or turn off on host.

buuenke, 2024-05-18:
same here AFTER update my Lily can not be reached anymore...???

Pimentoso, 2024-05-21:
Check #5790