esphome/issues #5810: Media player can't play mp3s after updating to 2024.5.0

mad-tunes, 2024-05-17:
### The problem

Media player doesn't play mp3s after update to 2024.5.0

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.5.3,  Supervisor 2024.05.1 ,Operating System 12.3, Frontend 20240501.1

### What platform are you using?

ESP32

### Board

ESP32-C3-Zero like here: https://www.aliexpress.com/item/1005006524672028.html

### Component causing the issue

esp-idf framework / adf_pipeline media player

### Example YAML snippet

```yaml
Anything using media_player.play_media, like this button, fails to play the .mp3

button:
  - platform: template
    name: Test sound
    id: test_sound
    icon: "mdi:speaker-play"
    disabled_by_default: true           # Shows entity in HA, but disabled by default
    on_press:
      - media_player.play_media: "http://192.168.5.5:8123/local/Voice_assistant/im_listening.mp3"

Any tts.speak from HA fails to play too:
  - service: tts.speak
    metadata: {}
    data:
      cache: true
      media_player_entity_id: media_player.jarvis_media_player
      message: "Test TTS message"
      language: en_US
      options:
        voice: en_US-ryan-medium
    target:
      entity_id: tts.piper

Whole YAML for my assistant/smart speaker:

substitutions:                          # Substitutions
  friendly_name: Jarvis
  host_name: jarvis
  static_ip: 192.168.5.8
  log_level: DEBUG                      # NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE
#  log_level: VERBOSE                      # NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE

  start_volume: 45%
  # Voice assistant
  mic_volume_multiplier: "10.0"         # MIC volume multiplier
  # Mic_volume_multiplier: "5.0"
  noise_suppression_level: "3"
  # Mic
[... 479 lines cut ...]
```

### Additional information

Rolling back the update to 2024.5.0 and re-uploading to the device made it play mp3s reliably again.

I found that if I stick with 2024.5.0 and change
```
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: recommended
```
to
```
esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: 4.4.6
```
...then things play OK again

rccoleman, 2024-05-18:
This fixed audio playback for me on my Onju voice devices too!  Thanks for the tip!

mad-tunes, 2024-05-20:
I find the same after updating to esphome 2024.5.1 as I did with 2024.5.0.
`version: recommended` occasionally plays the first sound but then nothing more, `version: 4.4.6` plays everything OK.

mad-tunes, 2024-06-05:
I'm still having similar trouble after updating to 2024.5.5.
Playing either an mp3 or TTS only tend to work for the first time. After hearing the wake word, I speak to it and it just gets stuck and sits there. It might listen and respond to me once, but then it needs a restart to do anything else. 
Still works OK if I go pack to esphome 2024.4.*, but setting `version: 4.4.6` **doesn't** seem to make things work reliably with 2024.5.5 like it did with the earnlier 2024.5.*

mad-tunes, 2024-06-25:
I'm still having similar trouble with 2024.6.2
With version: recommended, it might respond once but then gets stuck and needs a restart before it'll respond to anything else.
Setting version: 4.4.6 doesn't make it any better so I'll still have to stick with 2024.4.2 for now

mad-tunes, 2024-07-06:
The behaviours changed _slightly_ since I installed esphome `24.6.6`.
With `version: recommended`, it listens and responds OK first time, but on the second try it detects the wake word and starts listening, I ask something and then it gets stuck and needs a reset.
I see this in its log though, which I don't think I did prior to `24.6.6`:

```
[22:38:12][E][main:650]: ****MWW Wake word detected
[22:38:12][W][component:237]: Component micro_wake_word took a long time for an operation (63 ms).
[22:38:12][W][component:238]: Components should block for at most 30 ms.
[22:38:12][D][esp-idf:000]: I (32079) AUDIO_THREAD: The http task allocate stack on external memory

[22:38:12][D][esp-idf:000]: I (32082) AUDIO_ELEMENT: [http-0x3de17d34] Element task created

[22:38:12][D][esp-idf:000]: I (32085) AUDIO_THREAD: The decoder task allocate stack on external memory

[22:38:12][D][esp-idf:000]: I (32088) AUDIO_ELEMENT: [decoder-0x3de17eec] Element task created

[22:38:12][D][esp-idf:000][http]: I (32090) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1

[22:38:12][D][esp-idf:000][decoder]: I (32093) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[22:38:12][D][esp_audio_sources:097]: Streamer status: 2
[22:38:12][D][esp_audio_sources:098]: decoder status: 2
[22:38:13][D][esp-idf:000][http]: I (32159) HTTP_CLIENT: Body received in fetch header state, 0x3fcc211c, 1704

[22:38:13][D][esp-idf:000][http]: I (32163) HTTP_STREAM: total_bytes=2520

[22:38:13][D][esp-idf:000][http]: W (32179) HTTP_STREAM: No more data,errno:0, total_bytes:2520, rlen = 0

[22:38:13][D][esp-idf:000][http]: I (32182) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0

[22:38:13][D][esp-idf:000][decoder]: I (32186) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2

[22:38:13][I][HTTPStreamReader:129]: [ * ] Receive music info from mp3 decoder, sample_rates=32000, bits=16, ch=1
[22:38:13][D][adf_i2s_out:114]: update i2s clk settings: rate:32000 bits:16 ch:1
[22:38:13][D][adf_i2s_out:127]: Set final i2s settings: 32000
[22:38:13][D][esp-idf:000][decoder]: W (32200) AUDIO_ELEMENT: OUT-[decoder] AEL_IO_ABORT

[22:38:13][D][esp-idf:000][decoder]: W (32205) MP3_DECODER: output aborted -3

[22:38:13][D][esp-idf:000][decoder]: I (32209) MP3_DECODER: Closed

[22:38:13][D][esp_adf_pipeline:302]: State changed from PREPARING to STARTING
[22:38:13][I][adf_media_player:135]: got new pipeline state: 2
[22:38:13][D][adf_i2s_out:127]: Set final i2s settings: 32000
[22:38:13][D][esp-idf:000]: I (32225) AUDIO_ELEMENT: [i2s_out-0x3de180b8] Element task created

[22:38:13][D][esp-idf:000]: I (32228) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2123355 Bytes, Inter:165420 Bytes, Dram:165420 Bytes

[22:38:13][D][esp-idf:000][http]: I (32231) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1

[22:38:13][D][esp-idf:000][decoder]: I (32233) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[22:38:13][D][esp-idf:000][i2s_out]: I (32236) AUDIO_ELEMENT: [i2s_out] AEL_MSG_CMD_RESUME,state:1

[22:38:13][D][esp-idf:000][i2s_out]: I (32238) I2S_STREAM: AUDIO_STREAM_WRITER

[22:38:13][I][esp_adf_pipeline:214]: [ http ] status: 14
[22:38:13][I][esp_adf_pipeline:214]: [ i2s_out ] status: 12
[22:38:13][D][esp_adf_pipeline:131]: Check element [http] status, 2
[22:38:13][D][esp-idf:000][http]: I (32306) HTTP_CLIENT: Body received in fetch header state, 0x3fcbe770, 1704

[22:38:13][D][esp-idf:000][http]: I (32310) HTTP_STREAM: total_bytes=2520

[22:38:13][I][esp_adf_pipeline:214]: [ http ] status: 12
[22:38:13][D][esp_adf_pipeline:131]: Check element [http] status, 3
[22:38:13][D][esp_adf_pipeline:131]: Check element [decoder] status, 2
[22:38:13][I][esp_adf_pipeline:214]: [ decoder ] status: 12
[22:38:13][D][esp_adf_pipeline:131]: Check element [http] status, 3
ERROR Fatal error: protocol.data_received() call failed.
protocol: <aioesphomeapi._frame_helper.noise.APINoiseFrameHelper object at 0x7f95ad901590>
[... 27 lines cut ...]
[22:38:13][D][esp_adf_pipeline:400]: Called deinit_all
[22:38:13][D][esp-idf:000]: I (32663) AUDIO_PIPELINE: audio_pipeline_unlinked

[22:38:13][D][esp-idf:000]: W (32666) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE

[22:38:13][D][esp-idf:000]: W (32668) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE

[22:38:13][D][esp-idf:000]: W (32671) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[22:38:13][D][esp-idf:000]: I (32674) I2S: DMA queue destroyed

[22:38:13][D][esp_adf_pipeline:302]: State changed from STOPPING to UNINITIALIZED
[22:38:13][I][adf_media_player:135]: got new pipeline state: 0
[22:38:16][D][sensor:094]: 'Uptime seconds': Sending state 35.54700 s with 0 decimals of accuracy
[22:38:33][D][sensor:094]: 'WiFi Signal dB': Sending state -65.00000 dBm with 0 decimals of accuracy
[22:38:33][D][sensor:094]: 'WiFi Signal': Sending state 70.00000 % with 0 decimals of accuracy
[22:38:42][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[22:38:42][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:38:42][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```
Trying to play an mp3 (like `media_player.play_media: http://IP/mp3.mp3`) still doesn't work, but I see nothng that looks wrong in its log. Same with TTS.

Setting `version: 4.4.6` hasn't improved anything for the last few versions.

I still have to stick with esphome `2024.4` to be able to use it properly.

mad-tunes, 2024-07-17:
I still find the same with 2024.7.0

mad-tunes, 2024-10-04:
After updating HAs core this morning, my assitants now useless even on its older previously working formware.
It recognised the wake word, then sits and does nothing.
I guess 2024 is the Year of SIlence for me.

k9rdu, 2024-12-13:
Still having this issue on 2024.12.2