esphome/issues #5811: Have to remove from yaml for cb3s device to work after udpate to 2024.5.0

mad-tunes, 2024-05-17:
### The problem

My cb3s based IR blaster doesn't reliably connect to my wifi and the api after updating it to 2024.5.0

### Which version of ESPHome has the issue?

2024.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2024.5.3, Supervisor 2024.05.1 ,Operating System 12.3, Frontend 20240501.1

### What platform are you using?

BK72XX

### Board

cb3s

### Component causing the issue

?

### Example YAML snippet

```yaml
substitutions:
  friendly_name: IRBlaster
  host_name: irblaster

esphome:
  name: $host_name
  friendly_name: $friendly_name

bk72xx:
  board: cb3s                           # https://docs.libretiny.eu/boards/cb3s/
# bk72xx:
#   board: generic-bk7231n-qfn32-tuya

# <<: !include common/logger.yaml
<<: !include common/api_ota.yaml
# <<: !include common/wifi.yaml
# <<: !include common/web_server.yaml
<<: !include common/sensors.yaml
# <<: !include common/binary_sensors.yaml
# <<: !include common/buttons.yaml
<<: !include common/text_sensors.yaml
<<: !include common/sync_time.yaml
<<: !include common/switch.yaml

logger:                                 # Enable logging. https://esphome.io/components/logger.html
  level: ERROR                          # NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE
  baud_rate: 0                          # Stops logging over onboard UART

wifi:                                   # WiFi connection details. Without domain:, defaults to .local. https://esphome.io/components/wifi.html
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.5.7
    gateway: 192.168.5.254
    subnet: 255.255.255.0
    dns1: 192.168.5.252
  domain: .home
  # power_save_mode: LIGHT                # NONE, LIGHT, HIGH
  power_save_mode: none
[... 558 lines cut ...]
#      - remote_transmitter.transmit_pronto:
#          data: "0000 0073 0000 0021 0060 0020 0010 0010 0010 0010 0010 0020 0010 0020 0030 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0010 0010 0010 0010 0010 0010 0020 0010 0010 0010 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0020 0020 0020 0010 0010 0010 0010 0010 0010 0020 0010 09CD"
#      - delay: 60ms
#      - light.turn_off: status_light
#  - platform: template
#    name: Kodi Mute
#    icon: mdi:undo
#    on_press:
#      - light.turn_on: status_light
#      - remote_transmitter.transmit_pronto:
#          data: "0000 0073 0000 0022 0060 0020 0010 0010 0010 0010 0010 0020 0010 0020 0030 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0010 0010 0010 0010 0010 0010 0020 0010 0010 0010 0010 0010 0010 0010 0010 0020 0020 0010 0010 0010 0010 0010 0010 0010 0010 0010 0010 0020 0010 0010 0010 0010 0020 0010 09CD"
#      - delay: 60ms
#      - light.turn_off: status_light
```

### Additional information

I found that after removing/commenting out a handful of the IR remote buttons I had defined, the device started behaving reliably again.

Is there some kind of size limit that might have reduced with the update to 2024.5.0, or maybe something updated that's using more of some space somewhere (leaving less usable for my yaml)?

The only sizes I know of are these, reported as I compile and upload to the device:
RAM:   [====      ]  37.5% (used 98304 bytes from 262144 bytes)
Flash: [========= ]  86.8% (used 940132 bytes from 1083136 bytes)
...which are both below 100%

aldabro, 2024-05-18:
Hi,
I noticed a change too with my IR blaster. In my case, I have a bunch of IR commands that get sent one after another when a button is pressed. But now with version 2024.5.0 the last command does not get sent. I compiled version 2024.4.2 for my IR blaster and it works again. So I suspect it has something to do with remote_transmitter ?

mad-tunes, 2024-09-16:
This is still a thing